fwupd/fwupd #8175: fwupd crashing: g_strsplit: assertion 'string != NULL' failed

JohnAZoidberg, 2024-11-28:
**Describe the bug**
A clear and concise description of what the bug is.

**Steps to Reproduce**
Nothing, just boot and check the daemon status.

```
> journalctl -u fwupd
-- Boot c121e824eae84dbca1c8b4441257a8a9 --
Nov 28 23:37:47 fedora systemd[1]: Starting fwupd.service - Firmware update daemon...
Nov 28 23:37:49 fedora fwupd[4018]: 15:37:49.421 FuPluginDfu          failed to parse interface data: no image idx 33 found in firmware
Nov 28 23:37:49 fedora fwupd[4018]: 15:37:49.421 FuPluginDfu          DFU version 0x0000 invalid, v1.1 assumed
Nov 28 23:37:49 fedora fwupd[4018]: 15:37:49.600 FuMain               Daemon ready for requests (locale en_US.UTF-8)
Nov 28 23:37:49 fedora systemd[1]: Started fwupd.service - Firmware update daemon.
Nov 28 23:37:49 fedora fwupd[4018]: 15:37:49.620 GLib                 g_strsplit: assertion 'string != NULL' failed
Nov 28 23:37:50 fedora systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
Nov 28 23:37:50 fedora systemd[1]: fwupd.service: Failed with result 'core-dump'.
Nov 28 23:37:50 fedora systemd[1]: fwupd.service: Consumed 1.296s CPU time, 186.9M memory peak.
```

**fwupd version information**
Please provide the version of the daemon and client.

```shell
> fwupdmgr --version
Remote peer disconnected

> fwupdtool --version
Loading…                 [************************************** ]
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.2
compile   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.2
runtime   org.freedesktop.fwupd-efi     1.6
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libjcat           0.2.2
runtime   com.hughsie.libxmlb           0.3.21
runtime   org.freedesktop.fwupd         2.0.2
runtime   org.kernel                    6.13.0-0.rc0.20241126git7eef7e306d3c.10.fc42.x86_64
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
> fwupdmgr get-devices --show-all-devices
Unable to connect to service
```

</details>

**Additional questions**

- Operating system and version: Fedora Rawhide 6.13.0-0.rc0.20241126git7eef7e306d3c.10.fc42.x86_64
- Have you tried rebooting?
- Is this a regression?

JohnAZoidberg, 2024-11-28:
Started happening after I just ran software updates in fedora rawhide.

superm1, 2024-11-28:
This should be fixed by https://github.com/fwupd/fwupd/commit/54291786967bb1df76b21c6bc30319f19a4f8cdd

superm1, 2024-12-02:
closing as duplicate of https://github.com/fwupd/fwupd/issues/8161

closed 2024-12-02 by superm1