fwupd/fwupd #8177: snap fwupd: report-history shows numeric string for the VersionNew field instead a formatted version

CragW, 2024-11-29:
**Describe the bug**
This issue happens to `snap` fwupd only, for now the version is `2.0.0` in the `latest/stable` channel. Issue is not reproducible with debian fwupd 1.9.x or 2.0.x.

After the system firmware update is applied, and rebooted to the desktop environment.  The `VersionNew` field displayed by `fwpud.fwupdmgr report-history` shows the numeric value `68096` instead a formatted version `1.10.0` according to device verfmt.

```
"Reports" : [
  {
    "Protocol" : "org.uefi.capsule",
    "UpdateState" : 2,
    "Plugin" : "uefi_capsule",
    "VersionOld" : "1.10.0",
    "VersionNew" : "68096",
    "Metadata" : {
      "CompileVersion(info.libusb)" : "1.0.27",
      "DistroName" : "Ubuntu Core",
      "DistroId" : "ubuntu-core",
      "CompileVersion(io.snapcraft.fwupd)" : "6561",
      "CompileVersion(com.hughsie.libjcat)" : "0.2.0",
      "EfivarsNvramUsed" : "76198",
      "LinuxLockdown" : "none",
      "CapsuleApplyMethod" : "nvram",
      "HostBiosVendor" : "Dell Inc.",
      "CompileVersion(com.hughsie.libxmlb)" : "0.3.19",
      "CompileVersion(org.freedesktop.fwupd)" : "2.0.0",
```

Also, `fwupd.fwupdmgr get-devices` shows the unformated version too.
```
Devices that have been updated successfully:
 • System Firmware (1.10.0 → 68096)
```

**Steps to Reproduce**
```
$ sudo apt purge fwupd
$ sudo snap install fwupd
$ fwupd.fwupdmgr refresh
$ fwupd.fwupdmgr install [System Firmware]
$ sudo reboot
$ fwupd.fwupdmgr report-history
```

**Expected behavior**
```
    "VersionOld" : "1.10.0",
    "VersionNew" : "1.10.0",
```

**fwupd version information**
```shell
$ fwupd.fwupdmgr --version
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.0
compile   com.hughsie.libxmlb           0.3.19
compile   io.snapcraft.fwupd            6561
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
runtime   com.hughsie.libjcat           0.2.0
runtime   com.hughsie.libxmlb           0.3.19
runtime   org.kernel                    6.8.0-49-generic
runtime   org.freedesktop.fwupd         2.0.0
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): snap

<details>

<summary>**fwupd device information**</summary>

[... 178 lines cut ...]
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            Dell Inc. Platform Key
      Vendor:             Dell Inc.
      GUID:               5f7cc4ea-9f7a-5c5f-b535-dfbcc73f5ba6 ← UEFI\CRT_2493B118EABD632FD3BA49C333A8329B1982E872

────────────────────────────────────────────────
Devices that have been updated successfully:
 • System Firmware (1.10.0 → 68096)
Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
Review and upload report now? (Requires internet connection) [Y|n]:

```

</details>

**Additional questions**

- Operating system and version: Ubuntu 24.04
- Have you tried rebooting? Yes
- Is this a regression? No

superm1, 2024-11-29:
How about snap track latest on channel candidate or edge?

CragW, 2024-11-29:
Issue is reproducible with `latest/edge` too.

```
"Reports" : [
  {
    "Checksum" : "23fd1f16401b7fa57c5a70c203470259e4aae4c9178373e47352f4be5c810706",
    "ReleaseId" : "99304",
    "Protocol" : "org.uefi.capsule",
    "UpdateState" : 2,
    "Guid" : [
      "3a988588-49a8-4bea-8f42-a5b04b18efc4"
    ],
    "Plugin" : "uefi_capsule",
    "VersionOld" : "1.10.0",
    "VersionNew" : "68096",
    "Flags" : 574619947,
    "Created" : 1732849106,
    "Modified" : 1732849149,
    "Metadata" : {
      "HostBiosMinorRelease" : "0a",
      "HostSku" : "C88",
      "CompileVersion(info.libusb)" : "1.0.27",
      "HostFirmwareMinorRelease" : "02",
      "CompileVersion(org.freedesktop.fwupd)" : "2.0.3",
      "DistroId" : "ubuntu-core",
```

superm1, 2024-11-29:
And you're absolutely sure no problem in 2.0 dev or venv version?

Can you share your pending.db for both?

CragW, 2024-11-29:
> And you're absolutely sure no problem in 2.0 dev or venv version?

Weirdly snap only, no idea why `version_new` saved it this way.

> Can you share your pending.db for both?

snap fwupd latest/edge 2.0.3-prelim
[snap-2.0.3-pending.db.txt](https://github.com/user-attachments/files/17954740/snap-2.0.3-pending.db.txt)

deb fwupd 1.9.24
[deb-1.9.24-pending.db.txt](https://github.com/user-attachments/files/17954735/deb-1.9.24-pending.db.txt)

hughsie, 2024-12-06:
I don't understand! :) In both those pending dbs we have `(version_old TEXT,version_new TEXT) == ('1.10.0','1.10.0')` -- @CragW do you have the same problem when using 2.0.2 as a .deb package rather than a snap? i.e. is it a 2.0.x regression rather than a _snap-specific_ issue? I'm a bit confused as `fwupdmgr get-history -> fwupd_client_get_history -> fu_engine_get_history -> fu_history_get_devices` -- i.e. this must be the value we recorded into the `pending.db`

@CragW could it have been possible you were installing an unsigned .cab file that wasn't from the LVFS? i.e. one without the `LVFS::VersionFormat` metadata key set?

CragW, 2024-12-11:
> do you have the same problem when using 2.0.2 as a .deb package rather than a snap? i.e. is it a 2.0.x regression rather than a _snap-specific_ issue? 

Nope, no idea why the unfriendly displaying `version_new` is snap only.

> @CragW could it have been possible you were installing an unsigned .cab file that wasn't from the LVFS?

The .cab file was sourced from LVFS, should be signed.

--

I think the report of this issue is notable in the backend too, which is exported to the `analysis` page.  For example, see the [top two reports in link](https://fwupd.org/lvfs/firmware/96642/analytics/reports).

The first report appears to be snap fwupd, updated from `v1.10.0 → v0x10a00`. The second report is ubuntu 22.04 with debian fwupd, updated from `v1.9.0 → v1.10.0`.

Report 1
```
Dell OptiPlex AIO Plus 7420, OptiPlex AIO 7420 65W, OptiPlex AIO 7420 35W v1.10.0 → v0x10a00 (a day ago)
Capsule Apply Method nvram
Compile Version: com.hughsie.libjcat 0.2.0
Compile Version: com.hughsie.libxmlb 0.3.21
Compile Version: info.libusb 1.0.27
Compile Version: io.snapcraft.fwupd 6855
Compile Version: org.freedesktop.fwupd 2.0.4
Compile Version: org.freedesktop.fwupd.source 2.0.3-9-g18edf497e
CPU Architecture x86_64
CPU Model Intel Core™ i9-14900
Display State connected
Distribution ID ubuntu-core
Distribution Name Ubuntu Core
Distribution Pretty Name Ubuntu Core 24
```

Report 2
```
Dell OptiPlex AIO Plus 7420, OptiPlex AIO 7420 65W, OptiPlex AIO 7420 35W v1.9.0 → v1.10.0 (a month ago)
Capsule Apply Method nvram
Compile Version: com.hughsie.libjcat 0.1.9
Compile Version: org.freedesktop.fwupd 1.7.9
Compile Version: org.freedesktop.gusb 0.3.10
CPU Architecture x86_64
Distribution ID ubuntu
Distribution Version 22.04
```

superm1, 2024-12-11:
Is this a case that the actual metadata is embargo only metadata and you're missing the embargo remote in the snap setup perhaps?

CragW, 2024-12-11:
> Is this a case that the actual metadata is embargo only metadata and you're missing the embargo remote in the snap setup perhaps?

The target version v1.10.0 has been made available in the public md, by now it is the top release at https://fwupd.org/lvfs/devices/com.dell.uefi3a988588.firmware

superm1, 2024-12-11:
Was it at the time this report made?

CragW, 2024-12-11:
> Was it at the time this report made?

Just did a quick test with snap fwupd `latest/edge` and issue still stands.

<details>

```
$ sudo fwupd.fwupdmgr install
WARNING: This package has not been validated, it may not work properly.
0.      Cancel
1.      c430a03ca2a65dfe2412ff950c79c51f6aec1317 (PC SN740 NVMe WD 256GB)
2.      f428758c51104d245255ec79a167bab59a5596b5 (PC SN740 NVMe WD 256GB)
3.      5b4a88d5893562078b07b8a3507e392642881887 (System Firmware)
4.      362301da643102b9f38477387e2193e57abaa590 (UEFI dbx)
Choose device [0-4]: 3
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall System Firmware to 1.10.0?                                         ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • This release contains security updates as disclosed in the Dell            ║
║ Security Advisory DSA2024-373.                                               ║
║                                                                              ║
║ OptiPlex AIO Plus 7420 must remain plugged into a power source for the       ║
║ duration of the update to avoid damage.                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adaptor while the update is in progress.
An update requires a reboot to complete. Restart now? [y|N]: y

Broadcast message from root@dm-OptiPlex-AIO-Plus-7420 on pts/1 (Wed 2024-12-11 12:25:27 CST):

The system will reboot now!
```
```
$ fwupd.fwupdmgr get-history
WARNING: This package has not been validated, it may not work properly.
Dell Inc. OptiPlex AIO Plus 7420
│
└─System Firmware:
  │   Device ID:          5b4a88d5893562078b07b8a3507e392642881887
  │   Previous version:   1.10.0
  │   Update State:       Success
  │   Last modified:      2024-12-11
  │   GUID:               3a988588-49a8-4bea-8f42-a5b04b18efc4
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │
  └─OptiPlex AIO Plus 7420, OptiPlex AIO 7420 65W, OptiPlex AIO 7420 35W System Update:
        New version:      0x10a00
        Remote ID:        vendor-directory
        Summary:          Firmware for the Dell OptiPlex AIO Plus 7420, OptiPlex AIO 7420 65W, OptiPlex AIO 7420 35W
        License:          Proprietary
        Urgency:          Low
        Vendor:           Dell
        Update Message:   Do not turn off your computer or remove the AC adapter while the update is in progress.
        Release Flags:    • Trusted payload
                          • Trusted metadata
        Description:
        This stable release fixes the following issues:

        • This release contains security updates as disclosed in the Dell Security Advisory DSA2024-373.
```
```
$ fwupd.fwupdmgr report-history --force
WARNING: This package has not been validated, it may not work properly.
Target:                  https://fwupd.org/lvfs/firmware/report
Payload:                 {
                           "ReportType" : "history",
                           "ReportVersion" : 2,
                           "MachineId" : "1d0674f2a6e3ac2a07ce294fada7c609f95999ab102db8ebcfd81456d678242a",
                           "Metadata" : {
                               ...                       },
                           "Reports" : [
                             {
                               "Checksum" : "633bc3533e4ddd9fa1eced9847b5ba98de4c6013",
                               "ReleaseId" : null,
                               "Protocol" : "org.uefi.capsule",
                               "UpdateState" : 2,
                               "UpdateMessage" : "Do not turn off your computer or remove the AC adapter while the update is in progress.",
                               "Guid" : [
                                 "3a988588-49a8-4bea-8f42-a5b04b18efc4"
                               ],
                               "Plugin" : "uefi_capsule",
                               "VersionOld" : "1.10.0",
                               "VersionNew" : "0x10a00",
                               "Flags" : 574619947,
                               "Created" : 1733890958,
                               "Modified" : 1733891107,
                               "Metadata" : {
                                 "HostBiosMinorRelease" : "0a",
                                 "HostSku" : "C88",
                                 "CompileVersion(info.libusb)" : "1.0.27",
                                 "HostFirmwareMinorRelease" : "02",
                                 "CompileVersion(org.freedesktop.fwupd)" : "2.0.4",
                                 "DistroId" : "ubuntu-core",
                                 ...
                                 "KernelName" : "Linux",
                                 "HostProduct" : "OptiPlex AIO Plus 7420",
                                 ...
                               }
                             }
                           ]
                         }
Proceed with upload? [Y|n]:

```

</details>

superm1, 2024-12-11:
Does get devices show same wrong version? And same thing after refresh again?

CragW, 2024-12-11:
> Does get devices show same wrong version? And same thing after refresh again?

It is okay with `get-devices`. After `refresh`, the display verison in `get-history` and `report-history` still weird. Details below.

<details>

```
├─System Firmware:
│ │   Device ID:          5b4a88d5893562078b07b8a3507e392642881887
│ │   Summary:            UEFI System Resource Table device (updated via NVRAM)
│ │   Current version:    1.10.0
│ │   Minimum Version:    1.10.0
```

```
$ fwupd.fwupdmgr refresh --force
WARNING: This package has not been validated, it may not work properly.
Updating lvfs
Downloading…             [                    \                  ]
Successfully downloaded new metadata: Updates have been published for 5 of 9 local devices

$ fwupd.fwupdmgr get-history
WARNING: This package has not been validated, it may not work properly.
Dell Inc. OptiPlex AIO Plus 7420
│
└─System Firmware:
  │   Device ID:          5b4a88d5893562078b07b8a3507e392642881887
  │   Previous version:   1.10.0
  │   Update State:       Success
  │   Last modified:      2024-12-11
  │   GUID:               3a988588-49a8-4bea-8f42-a5b04b18efc4
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │
  └─OptiPlex AIO Plus 7420, OptiPlex AIO 7420 65W, OptiPlex AIO 7420 35W System Update:
        New version:      0x10a00
        Remote ID:        vendor-directory
        Summary:          Firmware for the Dell OptiPlex AIO Plus 7420, OptiPlex AIO 7420 65W, OptiPlex AIO 7420 35W
        License:          Proprietary
        Urgency:          Low
        Vendor:           Dell
        Update Message:   Do not turn off your computer or remove the AC adapter while the update is in progress.
        Release Flags:    • Trusted payload
                          • Trusted metadata
        Description:
        This stable release fixes the following issues:

        • This release contains security updates as disclosed in the Dell Security Advisory DSA2024-373.
```
</details>

superm1, 2024-12-11:
Is it possible it was wrong before doing install and so in db it stored in wrong format?

I don't think we apply metadata version format to history database

CragW, 2024-12-16:
> Is it possible it was wrong before doing install and so in db it stored in wrong format?
> 
> I don't think we apply metadata version format to history database

I got this [link ](https://github.com/fwupd/fwupd/commit/44ea27388eff4fbe017a2a2217922d89e8823566#diff-b4d75e8ef549c69eb5d7012cfd38c89c8bbad09ae7f69477c89ed2c0fd178b33R34)from @hughsie, it seems `verfmt` has been considered in the history/database.

superm1, 2024-12-16:
In that case; can you with the snap reproduce it by:

1. clear all old stateful stuff from snap
2. install snap
3. `fwupdmgr refresh`
4. `fwupdmgr update`/`fwupdmgr install foo.cab`

If the answer is no, then I think it's because the verfmt wasn't stored into the db at the time?

CragW, 2025-01-24:
Yes, issue is reproducible with a fresh installed stable snap fwupd.

Hereunder is a system with BIOS 1.11.2 upgrade to 1.12.1, however the display version is "68609" that doesn't seem to involve verfmt at all.

```
$ fwupdmgr get-history  -vv
(pkttyagent:305747): GLib-GIO-DEBUG: 14:32:49.156: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(/snap/fwupd/6851/bin/fwupdmgr:305712): GLib-GIO-DEBUG: 14:32:49.187: _g_io_module_get_default: Found default implementation dummy (GDummyProxyResolver) for ‘gio-proxy-resolver’
(/snap/fwupd/6851/bin/fwupdmgr:305712): GLib-GIO-DEBUG: 14:32:49.189: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(/snap/fwupd/6851/bin/fwupdmgr:305712): Fwupd-DEBUG: 14:32:49.192: Emitting ::status-changed() [idle]
FuMain-INFO: 14:32:49.225: FwupdDevice:
  DeviceId:             5b4a88d5893562078b07b8a3507e392642881887
  Name:                 System Firmware
  Guid:                 3a988588-49a8-4bea-8f42-a5b04b18efc4
  Plugin:               uefi_capsule
  Flags:                internal|require-ac|supported|needs-reboot|historical|can-verify|usable-during-update|updatable-hidden
  Problems:             update-in-progress
  Version:              1.11.2
  VersionFormat:        dell-bios
  Created:              2025-01-24
  Modified:             2025-01-24
  UpdateState:          needs-reboot
  UpdateError:          An update is in progress
  FwupdRelease:
    AppstreamId:        com.dell.uefi3a988588.firmware
    ReleaseId:          104327
    RemoteId:           lvfs
    Name:               OptiPlex AIO Plus 7420, OptiPlex AIO 7420 65W, OptiPlex AIO 7420 35W
    Summary:            Firmware for the Dell OptiPlex AIO Plus 7420, OptiPlex AIO 7420 65W, OptiPlex AIO 7420 35W
    Description:        <p>This stable release fixes the following issues:</p><ul><li>This release contains security updates as disclosed in the Dell Security Advisory DSA2025-002, DSA2025-005, and DSA2025-020.</li></ul>
    Version:            68609
    Filename:           DakarMlkRpl_16.1.35.2557_QS_EVT_firmware.cab
    Protocol:           org.uefi.capsule
    Categories:         X-System
    Checksum:           SHA1(fd26a7c455fa8413d071742eb5601afca32cc3b4)
    Checksum:           SHA256(642e08ad30a082638303b1ce2d22f6055dcb5981218dbb712113cea367d4c6fc)
    License:            LicenseRef-proprietary
    Size:               27.3 MB
    Created:            2024-12-20
    Uri:                https://fwupd.org/downloads/642e08ad30a082638303b1ce2d22f6055dcb5981218dbb712113cea367d4c6fc-DakarMlkRpl_16.1.35.2557_QS_EVT_firmware.cab
    Homepage:           http://support.dell.com/
    Urgency:            low
    Vendor:             Dell
    Flags:              trusted-payload|trusted-metadata
    HostEnclosureKind:  d
    TpmFamily:          2.0
    UEFIUXCapsule:      Enabled
    HostSku:            C88
    Pcr0_SHA256:        87485168e5c36763acf5cb38f9de8762f1905ad07922b67903edd93abf8f3b29
    DistroVersion:      24
    RuntimeVersion(org.freedesktop.fwupd): 2.0.3
    RuntimeVersion(org.kernel): 6.8.0-50-generic
    EspKind:            c12a7328-f81f-11d2-ba4b-00a0c93ec93b
    CpuArchitecture:    x86_64
    SecureBoot:         Disabled
    MissingCapsuleHeader: False
    HostFirmwareMajorRelease: 01
    HostFamily:         OptiPlex
    HostBiosVersion:    1.11.2
    HostVendor:         Dell Inc.
    FwupdSupported:     True
    PowerState:         unknown
    KernelName:         Linux
    DistroPrettyName:   Ubuntu Core 24
    BatteryLevel:       101
    SystemIntegrityOld: UEFI:Boot0003=b43c8f393969c090edd4a472363f11368b315bb102e99976be879c013beed153
    UEFI:PK=a4faa022b574b834e130aee1ac1663a66290b268f08ec2dd81f0b04d3bbdd8fb
    UEFI:BootOrder=5674311bf2bf911d7314c4a1d62955ed5555d36afa843fe285268783acdac5b2
    UEFI:SignatureSupport=7579884c3681593cb81401ed6b2d28fe5e0ad4d1ea236821907f94ecace0a3b1
    UEFI:KEKDefault=b722e671ad76391d497ee4ef4fe08b4f1dfcffac2e313d6b9c53ec8eff14a0bd
[... 91 lines cut ...]
        Vendor:           Dell
        Release Flags:    • Trusted payload
                          • Trusted metadata
        Description:
        This stable release fixes the following issues:

        • This release contains security updates as disclosed in the Dell Security Advisory DSA2025-002, DSA2025-005, and DSA2025-020.
        Checksum:         642e08ad30a082638303b1ce2d22f6055dcb5981218dbb712113cea367d4c6fc

```

snap fwupd version:
```
$ fwupdmgr --version
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.3
compile   com.hughsie.libxmlb           0.3.21
compile   io.snapcraft.fwupd            6851
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
runtime   com.hughsie.libjcat           0.2.0
runtime   com.hughsie.libxmlb           0.3.21
runtime   org.kernel                    6.8.0-50-generic
runtime   org.freedesktop.fwupd         2.0.3
```

bboozzoo, 2025-01-24:
Is it possible that one side (e.g. the client) interprets the version differently? @CragW can you try the same calling `fwupd.fwupdmgr get-history` ?

CragW, 2025-02-03:
Verfiy fixed with #8393 from `channel=latest/edge`.

closed 2025-02-03 by CragW