fwupd/fwupd #8187: fwupdmgr cannot connect to daemon (Ubuntu 22.04, snap, Dell Precision 5680)

nphilbrook, 2024-12-02:
**Describe the bug**
Running any fwupdmgr commands returns this error:
```
@arlo ~ $ sudo fwupdmgr get-devices 
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
```

**Steps to Reproduce**
Install stable fwupd from snap, run any fwupd commands

**Expected behavior**
fwupdmgr is able to connect to the daemon and functions normally.

**fwupd version information**
From snap:
```shell
@arlo ~ $ sudo snap list fwupd
Name   Version  Rev   Tracking       Publisher       Notes
fwupd  2.0.0    6561  latest/stable  richard-hughes  -
```
NOTE: I cannot even run `fwupdmgr --version` as it returns the same error message as above

**Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):**
I used snap. Note that Ubuntu 22.04 ships with an older version from apt, but I was hitting a different error message with this version:
```
Failed to get releases for UEFI dbx: No releases found: Not compatible with org.freedesktop.fwupd version 1.7.9, requires >= 1.9.1
```
Hence I am trying to get a current version. I ran `apt purge fwupd` but note I did did not run the full purge the first time I tried to install from snap - just the `apt remove`. When I first encountered errors, I removed the snap, then purged the apt, then re-installed the snap.

**fwupd device information**
I cannot run the below command, same error as above.
```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**Additional questions**

NOTE: I see the daemon running:
```
@arlo ~ $ pgrep -fa fwupd        
12053 /snap/fwupd/6561/libexec/fwupd/fwupd
12568 /snap/fwupd/6561/libexec/fwupd/fwupd
```
Is this just a unit file problem? 

- Hardware:
Dell Precision 5680
- Operating system and version:
Xubuntu 22.04
- Have you tried rebooting?
Yes, several times
- Is this a regression?
Unknown, I have never run fwupd from snap before. The version installed from apt was working until a few weeks ago.

nphilbrook, 2024-12-02:
Interesting, running that command I get more detailed output, including the error that I pulled from the journal in my last message, it hangs after that error message for about 4 minutes, then returns:
```shell
@arlo ~/repos/nick.philbrook-ansible $ sudo fwupd.fwupdtool get-devices                                                                
[sudo] password for nphilbrook:                                                                                                        
Loading…                 [***************                        ]14:59:36.280 FuBackend            failed to create device from /sys/devices/platform/intel_vsc/mei/mei1: failed to probe: no parent with subsystem pci
15:03:29.551 FuPluginUefiCapsule  failed to get screen size: failed to subclass open: failed to open /dev/dri/card0: Invalid argument
Loading…                 [*****************                      ]15:03:29.584 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:07.3/0000:2c:00.0/0000:2d:02.0/0000:2e:00.0/usb5/5-2/5-2.3/5-2.3.1/5-2.3.1:1.0/host0/target0:0:0/0:0:0:0/block/sda: fa
iled to subclass open: failed to open /dev/sda: Operation not permitted                                                                
15:03:29.597 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:07.3/0000:2c:00.0/0000:2d:02.0/0000:2e:00.0/usb5/5-2/5-2.3/5-2.3.1/5-2.3.1:1.0/host0/target0:0:0/0:0:0:0/block/sda: failed to subclass open: failed to open /dev/sda: Operation not perm
itted                                                                                                                                  
Loading…                 [************************************** ]                                                                     
Dell Inc. Precision 5680                                          
│                                                                                                                                      

```
I omitted the rest of the output with my devices but can paste it if that is useful!

nphilbrook, 2024-12-02:
I was able to check for FW updates (my end goal, of course) although there is a long timeout after the first error message (same as above)
```

superm1, 2024-12-02:
OK, can you please try switching to edge channel and see if it works or still hangs on `fwupdtool`?  If it does, please attach the output of `sudo fwupd.fwupdtool get-devices -vv 2>&1 | tee fwupdtool.txt`

superm1, 2024-12-02:
It looks to me like it's hanging while probing DP outputs.  Can see nearly a two minute delay on `drm_dp_aux5:`:

```sh
16:02:59.961 FuDevice             using cd3818e2d3a168ec7e137e21fdfb21df5ad2e175 for PCI_SLOT_NAME=0000:00:02.0:drm_dp_aux4
16:02:59.983 FuBackend            failed to open device /sys/devices/pci0000:00/0000:00:02.0/drm/card1/card1-DP-4/drm_dp_aux4: failed to setup: DPCD read failed: failed to read 15: No such device or address
16:02:59.983 FuDevice             using 0c3b61d070aa5fbac8f69c093d9a370a688f2d82 for DEVNAME=drm_dp_aux5:drm_dp_aux5
16:04:52.846 FuBackend            failed to open device /sys/devices/virtual/drm_dp_aux_dev/drm_dp_aux5: failed to setup: DPCD read failed: failed to read 15: Connection timed out
16:04:52.846 FuDevice             using d7bfd19a68de57bb7de8a0cc8f7b97680677abc4 for PCI_SLOT_NAME=0000:00:02.0:drm_dp_aux6
16:04:52.852 FuBackend            failed to open device /sys/devices/pci0000:00/0000:00:02.0/drm/card1/card1-DP-6/drm_dp_aux6: failed to setup: DPCD read failed: failed to read 15: Input/output error
16:04:52.852 FuDevice             using a4d90c62862392b0f200f174dfb874c27bbbe86c for PCI_SLOT_NAME=0000:00:02.0:drm_dp_aux7
16:04:52.861 FuBackend            failed to open device /sys/devices/pci0000:00/0000:00:02.0/drm/card1/card1-DP-7/drm_dp_aux7: failed to setup: DPCD read failed: failed to read 15: Input/output error
16:04:52.861 FuDevice             using 401232f922f37f6228207f5073eedfc574417b5d for PCI_SLOT_NAME=0000:00:02.0:drm_dp_aux8
16:04:52.870 FuBackend            failed to open device /sys/devices/pci0000:00/0000:00:02.0/drm/card1/card1-DP-8/drm_dp_aux8: failed to setup: DPCD read failed: failed to read 15: Input/output error
16:04:52.874 FuBackend            failed to resolve /sys/devices/pci0000:00/0000:00:02.0/drm/card1/ddc: cannot resolve path: No such file or directory
```

Is there any matching kernel logs for that attempt?

I'm wondering where the timeout is too high, us or kernel.

superm1, 2024-12-02:
@hughsie thoughts?

nphilbrook, 2024-12-02:
That is super interesting. I am plugged in to a docking station with 2 DP outputs (one of which is in use). But back when I was still trying to get Zaphod Heads working (distinct X screens per monitor) I noticed "phantom" DP outputs that I _think_ were associated with different USB-C connectors (this type of low level hardware is not my strong suit). The laptop has 3 USB-C connectors for what it's worth, and only the connector on the _right_ side can drive monitors using the graphics card. Something like that.

nphilbrook, 2024-12-02:
> Is there any matching kernel logs for that attempt?
> 

I couldn't find anything in the journal for the string `drm_dp_aux5` and some others I tried from that block of errors. Let me know if there's anything else specifically I should search for.

superm1, 2024-12-02:
>  I tried from that block of errors. Let me know if there's anything else specifically I should search for.

Just run `dmesg --clear` and then run `sudo fwupd.fwupdtool get-devices` and then share anything that is added to `dmesg` command output.

> 
That is super interesting. I am plugged in to a docking station with 2 DP outputs (one of which is in use). But back when I was still trying to get Zaphod Heads working (distinct X screens per monitor) I noticed "phantom" DP outputs that I think were associated with different USB-C connectors (this type of low level hardware is not my strong suit). The laptop has 3 USB-C connectors for what it's worth, and only the connector on the right side can drive monitors using the graphics card. Something like that.

I'm thinking the solution to this bug is a shorter fwupd timeout on reading DPCD, no matter what the kernel timeout is.

superm1, 2024-12-02:
Hmm, looking at the code we already _do_ have a 10ms timeout, but it's polling the file descriptor for changes not for data availability AFAICT.

```c
#define FU_DPAUX_DEVICE_READ_TIMEOUT 10 /* ms */
```

Maybe we need to switch from `read()` to `select()`?

nphilbrook, 2024-12-02:
> > I tried from that block of errors. Let me know if there's anything else specifically I should search for.
> 
> Just run `dmesg --clear` and then run `sudo fwupd.fwupdtool get-devices` and then share anything that is added to `dmesg` command output.
Got it, there is a stack trace spamming my dmesg output that looks like it could be related. Attached.

[dmesg.txt](https://github.com/user-attachments/files/17982306/dmesg.txt)

superm1, 2024-12-02:
Ah yup. This isn't a fwupd big, it's a kernel bug. I'm going to close it here, please report it to the appropriate DRM trackers and mailing lists.

closed 2024-12-02 by superm1

nphilbrook, 2024-12-02:
> Ah yup. This isn't a fwupd bug, it's a kernel bug. I'm going to close it here, please report it to the appropriate DRM trackers and mailing lists.

@superm1 I really appreciate your help on this, I understand it's not a fwupd problem, but I have not reported a kernel bug in my time and would appreciate a pointer on where to go to do that.

superm1, 2024-12-02:
Should be here afaik

https://gitlab.freedesktop.org/drm/nouveau

nphilbrook, 2024-12-02:
> Should be here afaik
> 
> https://gitlab.freedesktop.org/drm/nouveau

Much appreciated. Have a great week.

nphilbrook, 2024-12-02:
For any pour souls following the trail of this one : https://gitlab.freedesktop.org/drm/nouveau/-/issues/399