fwupd/fwupd #8200: fwupd gets stuck when reading nvm_version of retimer

hughsie, 2024-12-03:
My current setup:

 * P16v AMD
 * ThinkPad Thunderbolt 4 Dock
 * fwupd 2.0.2
 * kernel 6.11.8-300.fc41.x86_64

    $ cd /sys/devices/pci0000:00/0000:00:08.3/0000:c7:00.5/domain0/0-0/0-2/usb4_port1/0-2:1.1
    [hughsie@hughsie-work-lan 0-2:1.1]$ cat uevent nvm_version 
    DEVTYPE=thunderbolt_retimer
    cat: nvm_version: Resource temporarily unavailable

I was expecting to be able to get the nvm version of the retimer, but fwupd spends multiple seconds _reading_, _waiting_, _reading_, _waiting_ **both** retimers -- and then eventually failing. Mika, Mark, any ideas? If it helps, this is:

    $ cat vendor device 
    0x1da0
    0x8830

@mrhpearson @westeri

mrhpearson, 2024-12-03:
Would this be more for @superm1 than Mika as it's AMD? (dragging you in anyway Mario...hope that's OK)

I tested on mine (with a TBT3 dock as my TBT4 dock is busy right now).
I get the same effect - unable to access nvm_version (device ID is the same).

I can check with the FW team -  but I'll wait for an ack from an expert before I do that...just in case this is expected for some reason :) 

Mark

superm1, 2024-12-03:
AFAICT from the above the retimer isn't on the board, it's in the dock.  So the issue is reading the nvm version of the retimer in the dock.

>  /sys/devices/pci0000:00/0000:00:08.3/0000:c7:00.5/domain0/0-0/0-2/usb4_port1/0-2:1.1
* 00:08.3 is the PCI root port for tunneling
* c7:00.5 should be the USB4 hub/device.

@hughsie can you turn on dynamic debugging for thunderbolt?  `thunderbolt.dyndbg=+p` on kernel command line.   Do you ever see `NVM version` [printed](https://github.com/torvalds/linux/blob/v6.13-rc1/drivers/thunderbolt/retimer.c#L101)?
Please share the kernel log with that in place.

I'm guessing this is the code block you're hitting where it returns `-EAGAIN`:
https://github.com/torvalds/linux/blob/v6.13-rc1/drivers/thunderbolt/retimer.c#L184

If that's all correct, this feels like a logic problem to me?  Why would trying to read again ever change `rt->nvm`?  Maybe it's better to instead adjust the visibility of the `nvm_version` sysfs file if it wasn't initialized.

@hughsie can you try this patch?
[0001-thunderbolt-Don-t-display-retimers-unless-nvm-was-in.patch](https://github.com/user-attachments/files/17997601/0001-thunderbolt-Don-t-display-retimers-unless-nvm-was-in.patch)

westeri, 2024-12-04:
There is one fix for this too for retimers on the other side of the cable. We should not expose those. Can you check that you have this commit:

bf791751162a ("thunderbolt: Add only on-board retimers when !CONFIG_USB4_DEBUGFS_MARGINING")

superm1, 2024-12-04:
That looks to have landed in 6.11.8 which is what @hughsie tested: https://git.kernel.org/stable/c/545f3c009c60

westeri, 2024-12-05:
It looks like it's the retimer of the upstream port of the device  so either that commit is missing or that Kconfig option is accidentally set. Third option is that the retimer is not "well known" but that would show up in the dmesg too as "unknown retimer format" or so.

superm1, 2024-12-06:
FWIW, I found a Lenovo thunderbolt dock and could reproduce this as well.  The kernel patch fixed it, so I [sent it out](https://lore.kernel.org/linux-usb/20241206183318.1701180-1-superm1@kernel.org/T/#u).

closed 2024-12-20 by vathpela