fwupd/fwupd #8208: During `fwupdmgr update`, `fwupd` crashes with SIGSEGV in strlen-avx2.s called from fu_efi_hard_drive_device_path_new_from_volume

robert-thille-cb, 2024-12-05:
**Describe the bug**
When attempting to update firmware on a Dell 7780 with no internal storage or network, booted from a USB stick which pivots to a RAM disk, fwupd crashes when attempting to update the firmware.

**Steps to Reproduce**
Sorry, but this is a bit involved...
For high security, we prepare Dell 7780 laptops by removing the NVME "drives" and boot off USB sticks with ISOs we build ourselves. Secure Boot is enabled, the Ubuntu shim is not used, and we install our own keys and the fwupdx64.efi is signed with our key.
When the USB stick boots, it creates a RAM disk, installs the OS to it, and pivots to that from the initrd.  From there we have a pretty normal Ubuntu GUI environment.  I'm attempting to add the ability to update the laptop firmware from that environment.
Without the NVME device disabled in the "BIOS" (my development laptop doesn't have the NVME device actually removed), fwupd runs properly, but updates the ESP on the NVME device despite the USB stick being the higher priority boot device.
After disabling the NVME device, the ESP on the USB stick was too small, so it would fail (report not enough space) when trying to expand the update. So the crash is after that.
After enlarging the ESP on the USB stick, `fwupdmgr update` would hang and eventually report that the peer had disconnected from the bus without replying:
```shell
root@ubuntu:~# fwupdmgr update
Devices with no available firmware updates: 
 • Internal SPI Controller
 • TPM
 • UEFI Device Firmware
 • UEFI dbx
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 1.14.0 to 1.15.0?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Fixed the issue where a Windows error message is displayed when you are    ║
║ using the system. This issue occurs when the processor runs at a high        ║
║ voltage rate.                                                                ║
║                                                                              ║
║ Precision 7780 must remain plugged into a power source for the duration of   ║
║ the update to avoid damage.                                                  ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Decompressing…           [                                       ]
Message recipient disconnected from message bus without replying
```
Looking around I found the crash log(s):
```shell
root@ubuntu:~# cat /var/log/apport.log
INFO: apport (pid 3293) 2024-11-28 01:55:03,487: called for pid 2831, signal 11, core limit 0, dump mode 1
INFO: apport (pid 3293) 2024-11-28 01:55:03,488: executable: /usr/libexec/fwupd/fwupd (command line "/usr/libexec/fwupd/fwupd")
INFO: apport (pid 3293) 2024-11-28 01:55:06,298: wrote report /var/crash/_usr_libexec_fwupd_fwupd.0.crash
INFO: apport (pid 3636) 2024-11-28 01:55:36,302: called for pid 3574, signal 11, core limit 0, dump mode 1
INFO: apport (pid 3636) 2024-11-28 01:55:36,303: executable: /usr/libexec/fwupd/fwupd (command line "/usr/libexec/fwupd/fwupd")
ERROR: apport (pid 3636) 2024-11-28 01:55:36,303: report /var/crash/_usr_libexec_fwupd_fwupd.0.crash already exists and unseen, skipping to avoid disk usage DoS
INFO: apport (pid 3687) 2024-11-28 01:55:50,137: called for pid 3650, signal 11, core limit 0, dump mode 1
INFO: apport (pid 3687) 2024-11-28 01:55:50,138: executable: /usr/libexec/fwupd/fwupd (command line "/usr/libexec/fwupd/fwupd")
ERROR: apport (pid 3687) 2024-11-28 01:55:50,138: report /var/crash/_usr_libexec_fwupd_fwupd.0.crash already exists and unseen, skipping to avoid disk usage DoS
INFO: apport (pid 4318) 2024-11-28 02:02:41,281: called for pid 4271, signal 11, core limit 0, dump mode 1
INFO: apport (pid 4318) 2024-11-28 02:02:41,282: executable: /usr/libexec/fwupd/fwupd (command line "/usr/libexec/fwupd/fwupd")
ERROR: apport (pid 4318) 2024-11-28 02:02:41,283: report /var/crash/_usr_libexec_fwupd_fwupd.0.crash already exists and unseen, skipping to avoid disk usage DoS
INFO: apport (pid 4403) 2024-11-28 02:04:48,759: called for pid 4343, signal 11, core limit 0, dump mode 1
INFO: apport (pid 4403) 2024-11-28 02:04:48,759: executable: /usr/libexec/fwupd/fwupd (command line "/usr/libexec/fwupd/fwupd")
ERROR: apport (pid 4403) 2024-11-28 02:04:48,760: report /var/crash/_usr_libexec_fwupd_fwupd.0.crash already exists and unseen, skipping to avoid disk usage DoS
root@ubuntu:~# ls -lh /var/crash/_usr_libexec_fwupd_fwupd.0.crash
-rw-r----- 1 root whoopsie 77M Nov 28 01:55 /var/crash/_usr_libexec_fwupd_fwupd.0.crash
```

Investigating other issues I found that I could run fwupd under gdb and get a backtrace:
```shell
root@ubuntu:~# gdb --args /usr/libexec/fwupd/fwupd --verbose
...
 # run the update in another terminal window...
...
Thread 1 "fwupd" received signal SIGSEGV, Segmentation fault.
Download failed: Invalid argument.  Continuing without source file ./string/../sysdeps/x86_64/multiarch/strlen-avx2.S.
__strlen_avx2 () at ../sysdeps/x86_64/multiarch/strlen-avx2.S:76
warning: 76	../sysdeps/x86_64/multiarch/strlen-avx2.S: No such file or directory
(gdb) bt
#0  __strlen_avx2 () at ../sysdeps/x86_64/multiarch/strlen-avx2.S:76
#1  0x00007ffff7c36438 in fu_efi_hard_drive_device_path_new_from_volume () from /usr/lib/x86_64-linux-gnu/fwupd-1.9.24/libfwupdplugin.so
#2  0x00007ffff7ddea49 in fu_uefi_device_build_dp_buf () from /usr/lib/x86_64-linux-gnu/fwupd-1.9.24/libfwupdengine.so
#3  0x00007ffff7ddfe02 in fu_uefi_device_write_update_info () from /usr/lib/x86_64-linux-gnu/fwupd-1.9.24/libfwupdengine.so
[... 78 lines cut ...]

```shell
efivar -l | grep fw
[efivar.txt](https://github.com/user-attachments/files/18028592/efivar.txt)

```

```shell
tree /boot
Sorry, `tree` not available, used `ls -lR ...`. Also, the ESP isn't mounted at /boot
I did `ls -laR` on the ESP mounted at `/media/root/ss-usb` before running `fwupdmgr update`, and then unmounted and ran the update. After fwupd crashed, the ESP was mounted at `/media/root/7ADB-339D` and I also did an `ls -laR` on that path.
Files attached: 
[ls-laR_USB-ESP_before.txt](https://github.com/user-attachments/files/18028577/ls-laR_USB-ESP_before.txt)
[ls-laR_USB-ESP_after.txt](https://github.com/user-attachments/files/18028580/ls-laR_USB-ESP_after.txt)

```

**Additional questions**

- Operating system and version:
- Have you tried rebooting? `Yes`
- Is this a regression? `Unknown`
- Are you using an NVMe disk? `No`
- Is secure boot enabled? `Yes`
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? `No. Dell 7780`

hughsie, 2024-12-05:
> but updates the ESP on the NVME device despite the USB stick being the higher priority boot device

Higher priority in what sense? in fwupd we order multiple ESPs using a heuristic that gives each a score: https://github.com/fwupd/fwupd/blob/main/libfwupdplugin/fu-context.c#L1664 -- using `sudo fwupdtool get-devices --plugins uefi_capsule` you'll see something like:

    20:20:20.819 FuContext            more than one ESP possible:
     - 0x30258:	/org/freedesktop/UDisks2/block_devices/nvme1n1p1
     - 0x20064:	/org/freedesktop/UDisks2/block_devices/nvme0n1p1

> 

I only see one strlen in fu_efi_hard_drive_device_path_new_from_volume on 1.9.24, and that's here:

		g_auto(GStrv) parts = g_strsplit(partition_uuid, "-", -1);
		if (!fu_firmware_strparse_uint32_safe(parts[0],
						      strlen(parts[0]),
						      0x0,
						      &value,
						      error)) {
			g_prefix_error(error, "failed to parse %s: ", parts[0]);
			return NULL;
		}

and I don't think partition_uuid can be NULL as we check it for:

	partition_uuid = fu_volume_get_partition_uuid(volume);
	if (partition_uuid == NULL) {
		g_set_error_literal(error,
				    G_IO_ERROR,
				    G_IO_ERROR_NOT_SUPPORTED,
				    "partition UUID required");
		return NULL;
	}

-- can you try installing debugging for the fwupd package and trying to get the backtrace again please? And if so, maybe also set G_DEBUG=fatal-criticals when running the daemon to see if something crazy is happening. Thanks.

superm1, 2024-12-05:
Please update to 1.9.27 too. I seem to recall we touched some of this the last few stable releases.

robert-thille-cb, 2024-12-05:
I'll try the 1.9.27 release.  I assume "install" is just untar at /?  
Also, do you publish that this is your key anywhere?

hughsie, 2024-12-05:
> I'll try the 1.9.27 release. I assume "install" is just untar at /?

No, waaay more involved than that. For debian I'll let @superm1 provide instructions.

> Also, do you publish that [this](https://keyserver.ubuntu.com/pks/lookup?search=163EB50119225DB3DF8F49EA17ACBA8DFA970E17&fingerprint=on&op=index) is your key anywhere?

I think I've just pushed the key there too. Does that work? It should also be on the opengpg keyserver too.

robert-thille-cb, 2024-12-05:
Yeah, just wanted you to confirm that was your key and not someone pretending to be you :-)
I'll work through the CONTRIBUTING.md and see how far that gets me...
Thanks!

superm1, 2024-12-05:
Just follow 

https://fwupd.github.io/libfwupdplugin/building.html

This will set it all up in a venv (except build deps obviously install in your host).

If you can't reproduce on main branch you can check out 1_9_X branch and rebuild and try it again there.

closed 2025-05-03 by hughsie