fwupd/fwupd #8209: fwupd is failing to start on Arch KDE (AMD hardware)

motedon, 2024-12-06:
fwupd can't and won't run after updating to the latest version?

from discover:
Failed to download metadata for lvfs: user agent unsuitable; fwupd version required

after googling a bit to understand where is the problem, I tried starting the service using:
systemctl start fwupd.service
and then checked the status, here is the output:

× fwupd.service - Firmware update daemon
     Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static)
     Active: failed (Result: core-dump) since Fri 2024-12-06 10:50:49 +07; 4s ago
   Duration: 441ms
 Invocation: 9d53aaad1dc0489ebc500da5ac9d00ac
       Docs: https://fwupd.org/
    Process: 6022 ExecStart=/usr/lib/fwupd/fwupd (code=dumped, signal=SEGV)
   Main PID: 6022 (code=dumped, signal=SEGV)
   Mem peak: 13.9M
        CPU: 1.164s

Dec 06 10:50:47 thinkpad systemd[1]: Starting Firmware update daemon...
Dec 06 10:50:49 thinkpad fwupd[6022]: 03:50:49.135 FuPluginDfu          failed to parse interface data: no image idx 33 found in firmware
Dec 06 10:50:49 thinkpad fwupd[6022]: 03:50:49.135 FuPluginDfu          DFU version 0x0000 invalid, v1.1 assumed
Dec 06 10:50:49 thinkpad fwupd[6022]: 03:50:49.223 FuMain               Daemon ready for requests (locale en_US.UTF-8)
Dec 06 10:50:49 thinkpad systemd[1]: Started Firmware update daemon.
Dec 06 10:50:49 thinkpad fwupd[6022]: 03:50:49.249 GLib                 g_strsplit: assertion 'string != NULL' failed
Dec 06 10:50:49 thinkpad systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
Dec 06 10:50:49 thinkpad systemd[1]: fwupd.service: Failed with result 'core-dump'.
Dec 06 10:50:49 thinkpad systemd[1]: fwupd.service: Consumed 1.164s CPU time, 13.9M memory peak.

fwupd version information
2.0.2 updated using pacman

fwupdmgr --version outputs:
remote peer disconnected

OS: Arch Linux x86_64 
ThinkPad T14 Gen 5
Kernel: 6.12.1-arch1-1 
DE: Plasma 6.2.4 
CPU: AMD Ryzen 7 PRO 8840U w/ Radeon 780M Graphics (16) @ 3.301GHz

superm1, 2024-12-06:
We fixed this in main already, but you're the 3rd report on it.  @hughsie how would you feel about a pipe cleaner quick release of 2.0.3 with the fixes the last week or two?

superm1, 2024-12-06:
Here is the patch BTW: https://github.com/fwupd/fwupd/commit/54291786967bb1df76b21c6bc30319f19a4f8cdd

motedon, 2024-12-06:
> We fixed this in main already, but you're the 3rd report on it.  @hughsie how would you feel about a pipe cleaner quick release of 2.0.3 with the fixes the last week or two?

Oh, I'm glad to know it's fixed. I'll just wait then, thanks for the quick reply!

hughsie, 2024-12-06:
> how would you feel about a pipe cleaner quick release of 2.0.3

I think it's probably a good idea. I'll do a quick sweep of the open PRs and issues and do the usual covscan type stuff.

closed 2024-12-06 by motedon