fwupd/fwupd #8217: CET Platform is supported, but CET OS support is not

64-bitman, 2024-12-07:
**Describe the question**
As the title says. However, CET OS support used to be supported but for some reason recently it was changed to unsupported? I tried checking DNF's history, but there wasn't any system upgrades or possibly related actions on that date. Any way to possibly resupport it again? Using Framework 13 laptop with ryzen 7640u cpu.

Distro: Fedora 41 Workstation
```shell
Host Security ID: HSI:3! (v1.9.27)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ Fused platform:                Locked
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled

HSI-2
✔ SPI write protection:          Enabled
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✔ SPI replay protection:         Enabled
✔ CET Platform:                  Supported
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled

HSI-4
✔ SMAP:                          Enabled
✘ Processor rollback protection: Disabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ Linux kernel:                  Untainted
✘ CET OS Support:                Not supported

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix

Host Security Events
  2024-11-22 23:39:42:  ✘ CET OS Support changed: Supported → Not supported
  2024-11-22 23:33:45:  ✔ Kernel is no longer tainted
  2024-11-16 04:43:18:  ✘ Kernel is tainted
  2024-11-15 19:48:19:  ✔ Kernel lockdown enabled
  2024-11-15 19:48:19:  ✔ Secure Boot enabled
```
 

**fwupd version information**
Please provide the version of the daemon and client if applicable to your current installation of `fwupd`.

```shell
compile   org.freedesktop.fwupd         1.9.27
compile   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.2
runtime   org.freedesktop.fwupd-efi     1.6
compile   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.2
runtime   org.freedesktop.Passim        0.1.8
runtime   org.freedesktop.gusb          0.4.9
runtime   org.kernel                    6.11.10-300.fc41.x86_64
runtime   org.freedesktop.fwupd         1.9.27
```

64-bitman, 2024-12-10:
> What does this print?
> 
> ```
> /usr/libexec/fwupd/fwupd-detect-cet
> echo $?
> ```

It prints 1

hughsie, 2024-12-10:
> It prints 1

So that means that the CET protection is not enabled. @superm1 I get the same on Fedora 41; do you remember how we debugged this last time? For my the helper is compiled with:

	ccache cc -Iplugins/cpu/libfwupdcethelper.a.p -Iplugins/cpu -I../../plugins/cpu -I. -I../.. -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c11 -O0 -g -Wfatal-errors -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wclobbered -Wdeclaration-after-statement -Wdiscarded-qualifiers -Wduplicated-branches -Wduplicated-cond -Wempty-body -Wfloat-equal -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Wimplicit-int -Wincompatible-pointer-types -Winit-self -Wint-conversion -Wlogical-op -Wmaybe-uninitialized -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-cast-function-type -Wno-address-of-packed-member -Wno-unknown-pragmas -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wno-nonnull-compare -Wno-analyzer-use-of-uninitialized-value -Wno-analyzer-fd-double-close -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wvla -Wwrite-strings -fstack-protector-strong -DFWUPD_COMPILATION -D_DEFAULT_SOURCE -D_BSD_SOURCE -D__BSD_VISIBLE -D_XOPEN_SOURCE=700 -D_GNU_SOURCE -D_DARWIN_C_SOURCE=900000 -fPIC -fcf-protection=none -MD -MQ plugins/cpu/libfwupdcethelper.a.p/fu-cpu-helper-cet-common.S.o -MF plugins/cpu/libfwupdcethelper.a.p/fu-cpu-helper-cet-common.S.o.d -o plugins/cpu/libfwupdcethelper.a.p/fu-cpu-helper-cet-common.S.o -c ../../plugins/cpu/fu-cpu-helper-cet-common.S
	ccache cc -Iplugins/cpu/fwupd-detect-cet.p -Iplugins/cpu -I../../plugins/cpu -I. -I../.. -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c11 -O0 -g -Wfatal-errors -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wclobbered -Wdeclaration-after-statement -Wdiscarded-qualifiers -Wduplicated-branches -Wduplicated-cond -Wempty-body -Wfloat-equal -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Wimplicit-int -Wincompatible-pointer-types -Winit-self -Wint-conversion -Wlogical-op -Wmaybe-uninitialized -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-cast-function-type -Wno-address-of-packed-member -Wno-unknown-pragmas -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wno-nonnull-compare -Wno-analyzer-use-of-uninitialized-value -Wno-analyzer-fd-double-close -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wvla -Wwrite-strings -fstack-protector-strong -DFWUPD_COMPILATION -D_DEFAULT_SOURCE -D_BSD_SOURCE -D__BSD_VISIBLE -D_XOPEN_SOURCE=700 -D_GNU_SOURCE -D_DARWIN_C_SOURCE=900000 -fcf-protection=full -MD -MQ plugins/cpu/fwupd-detect-cet.p/fu-cpu-helper-cet.c.o -MF plugins/cpu/fwupd-detect-cet.p/fu-cpu-helper-cet.c.o.d -o plugins/cpu/fwupd-detect-cet.p/fu-cpu-helper-cet.c.o -c ../../plugins/cpu/fu-cpu-helper-cet.c
	rm -f plugins/cpu/libfwupdcethelper.a && gcc-ar csrDT plugins/cpu/libfwupdcethelper.a plugins/cpu/libfwupdcethelper.a.p/fu-cpu-helper-cet-common.S.o
	cc  -o plugins/cpu/fwupd-detect-cet plugins/cpu/fwupd-detect-cet.p/fu-cpu-helper-cet.c.o -Wl,--as-needed -Wl,--no-undefined -Wl,-z,relro -Wl,-z,defs -Wl,-z,now -Wl,-z,ibt,-z,shstk plugins/cpu/libfwupdcethelper.a

superm1, 2024-12-10:
Is the glibc env variable set?

hughsie, 2024-12-10:
I forgot all about cc3e67b8bbb522344cfe0a3e3ff0ac940d9e537c -- you're correct:

    $ GLIBC_TUNABLES=glibc.cpu.hwcaps=SHSTK ./plugins/cpu/fwupd-detect-cet && echo $?
    0

And using fwupdmgr (on my work AMD machine):

    Runtime Suffix -!
    ✔ CET OS Support:                Supported

@64-bitman what do you get when using `GLIBC_TUNABLES`?

64-bitman, 2024-12-10:
> I forgot all about [cc3e67b](https://github.com/fwupd/fwupd/commit/cc3e67b8bbb522344cfe0a3e3ff0ac940d9e537c) -- you're correct:
> 
> ```
> $ GLIBC_TUNABLES=glibc.cpu.hwcaps=SHSTK ./plugins/cpu/fwupd-detect-cet && echo $?
> 0
> ```
> 
> And using fwupdmgr (on my work AMD machine):
> 
> ```
> Runtime Suffix -!
> ✔ CET OS Support:                Supported
> ```
> 
> @64-bitman what do you get when using `GLIBC_TUNABLES`?

running `GLIBC_TUNABLES=glibc.cpu.hwcaps=SHSTK /usr/libexec/fwupd/fwupd-detect-cet` I still get an error code of 1, or should I being using a git build of fwupd?

superm1, 2024-12-10:
> running GLIBC_TUNABLES=glibc.cpu.hwcaps=SHSTK /usr/libexec/fwupd/fwupd-detect-cet I still get an error code of 1, or should I being using a git build of fwupd

You don't need to be running a git build.  We've had this in for a while.  I'm suspect that there has been a regression in glibc.  Can you look at your package update logs and see if those correlate?

superm1, 2024-12-10:
CC @hjl-tools, any regressions you're aware in glibc in this area?

64-bitman, 2024-12-11:
> You don't need to be running a git build. We've had this in for a while. I'm suspect that there has been a regression in glibc. Can you look at your package update logs and see if those correlate?

Looked over my package logs again and there were no system upgrades on that list or anything I could see that would be related.

```shell
166 dnf install mpv                                                                                    2024-11-22 23:57:08                 7
165 dnf install totem-video-thumbnailer                                                                2024-11-22 23:41:51                 1
164 dnf install mokutil                                                                                2024-11-22 22:19:57                 1
163 dnf install dkms                                                                                   2024-11-22 22:08:49                 2
162 dnf install kernel-devel                                                                           2024-11-22 22:06:23                 4
161 dnf remove gnome-shell-extension-screenshot-window-sizer                                           2024-11-22 22:03:36                 2
```

hughsie, 2025-05-03:
Fixed in cc3e67b8bbb522344cfe0a3e3ff0ac940d9e537c I think.

closed 2025-05-03 by hughsie

zearp, 2026-08-21:
Can any of the maintainers (@hughsie or @superm1) please chime in and share their thoughts on my last two comments?

https://github.com/fwupd/fwupd/issues/8217#issuecomment-5286166636
https://github.com/fwupd/fwupd/issues/8217#issuecomment-5288184698

Do you share my conclusion that this issue will fix itself once upstream fixes it so the note is added back? Auto enable won't work until thats added back. It is the root cause and until thats is fixed not a single binary will run with shadow stacking enabled unless `GLIBC_TUNABLES=glibc.cpu.hwcaps=SHSTK` is set, which is not a proper solution. I'm sorry if I posted the wrong info in the wrong place. I hoped it would be helpful but if unwanted please let me know.

wirl, 2026-09-02:
As mentioned in the post above due to shadow stacks auto enable being broken in the kernel until [this](https://git.kernel.org/torvalds/c/a48acbaf99d239e60a09a9e2b7d0f7e9feb62769) is fixed the check currently only shows shadow stack state for the `fwupdmgr` process and not the system itself. As that is the only process running with `GLIBC_TUNABLES=glibc.cpu.hwcaps=SHSTK` forced into its environment. Which is done by cc3e67b8bbb522344cfe0a3e3ff0ac940d9e537c. Without that environment variable the check fails as @ValdikSS showed earlier. This would imply that shadow stacking is disabled system wide else it would pass when run with `fwupdtool` as well.

My question to @superm1 / @hughsie: Is it expected behaviour that the check does not indicate any system wide protection but only checks if shadow stacking can work for a single process when force enabled by an environment setting? Maybe a new issue is warranted for this? I know Fedora is planning to [implement](https://forge.fedoraproject.org/fesco/tickets/issues/3636) shadow stacks in F46. They do not seem aware auto enable is currently broken.

As end user I expected passing this check indicates global/system wide protection is working but as it stands now this is not the case. This should resolve itself once the kernel fixes the missing vdso note but until then maybe the docs or security report can clarify that passing the check does not indicate global shadow stack protection? I also think that once the kernel fixes this that `fwupdmgr` should no longer force that environment variable as it should be auto enabled when the process starts and not need to be force enabled.

hughsie, 2026-09-02:
I must admit I was surprised the GLIBC_TUNABLES was required. When the kernel is fixed we can certainly remove that.