fwupd/fwupd #8219: Available firmware updates not shown when using get-updates subcommand

Vringe, 2024-12-09:
**Describe the bug**
According to #6853 when running `fwupdmgr get-updates` I should see available firmware updates even if the affected device has "problems" like "lid is closed" or "device is not connected to AC".  But I don't. Have there been any changes that I'm not aware of regarding this?

If that is actually not possible anymore: Is there any way to get this information? This is a requirement for our use-case as we have a custom updater.

**Steps to Reproduce**
Output of `fwupdmgr get-updates`
```shell
Devices with no available firmware updates:
 • Embedded Controller
 • Intel Management Engine
 • Intel Management Engine
 • System Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • Integrated Camera
 • USB4 Retimer
Devices with the latest available firmware version:
 • Micron MTFDKBA512TFK
No updates available
```

Output of `fwupdmgr get-devices` (only the relevant part)
```shell
├─System Firmware:
│ │   Device ID:          758cece0e40a0a7f90d90e26ecd1f7774c3f9852
│ │   Summary:            UEFI System Resource Table device (updated via NVRAM)
│ │   Current version:    0.1.20
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   Problems:           • Device requires AC power to be connected
│ │   GUID:               30fbf24a-e1da-4840-84f6-6d7deb630e1d
│ │   Device Flags:       • Internal device
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │                       • Updatable
│ │   Device Requests:    • Message
│ │
```

**Expected behavior**
I expect `fwupdmgr get-updates`, (also with the `--json` formatter) to return available updates regardless of whether there are problems or not.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.3
compile   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.2
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libxmlb           0.3.21
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libjcat           0.2.2
runtime   org.kernel                    6.12.3-arch1-1
runtime   org.freedesktop.fwupd         2.0.3
[... 369 lines cut ...]
│     Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
│     Summary:            REDACTED
│     GUID:               REDACTED
│
└─USB4 Retimer:
      Device ID:          fa7ebe94a3e7265f44a1d3346dffe26f29f1be42
      Summary:            A physical layer protocol-aware, software-transparent extension device that forms two separate electrical link segments
      Current version:    10.00
      Vendor:             Intel (THUNDERBOLT:0x8087)
      GUIDs:              e6f2e4ac-f76a-5fc4-b36a-6f9cb548689e ← THUNDERBOLT\VEN_8087&DEV_15EE
                          dfeacc0a-7af4-5460-ab40-22ed630568de ← TBT-808715ee-retimer1-0:1.1
      Device Flags:       • Internal device
                          • Updatable
                          • Device stages updates
                          • Unsigned Payload
                          • Can tag for emulation
```

</details>

**Additional questions**

- Operating system and version: Arch Linux (up to date)
- Have you tried rebooting? Yes
- Is this a regression? Maybe

hughsie, 2024-12-09:
> Have there been any changes that I'm not aware of regarding this?

It seems to have broken; not sure why. Could you bisect and try an identify when it broke please?

hughsie, 2024-12-09:
I think it might be bfa5fe696c3ae5e5cc5e9455c922f80a5793093d

closed 2024-12-09 by superm1

Vringe, 2024-12-10:
Many thanks for the quick fix ❤️ Can confirm this works now.