fwupd/fwupd #8224: *fwupdmgr update* failed to write-firmware: /boot/efi/EFI/debian: Is a directory

cgzones, 2024-12-10:
**Describe the bug**
Updating the UEFI database fails with:

```
failed to write-firmware: Blocked executable in the ESP, ensure grub and shim are up to date: failed to load /boot/efi/EFI/debian/: Error opening file /boot/efi/EFI/debian: Is a directory
```

**Steps to Reproduce**
```
fwupdmgr update
```

**Expected behavior**
Success.

**fwupd version information**

```shell
$ fwupdmgr --version
compile   info.libusb                   1.0.27
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.0
compile   org.freedesktop.fwupd         2.0.3
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.0
runtime   org.kernel                    6.12.3-amd64
runtime   org.freedesktop.fwupd         2.0.3
```

Installed via `apt`.

<details>

<summary>**fwupd device information**</summary>

```shell
$ fwupdmgr get-devices --show-all-devices
[...]
UEFI Platform Key:
│     Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
│     Summary:            GIGABYTE
│     GUID:               0ad2e789-96e4-5f02-a40e-73e3f75ac138 ← UEFI\CRT_9CB3EA9CC064A0E6A2FF1F5803717B842E524B68
│   
└─UEFI dbx:
      Device ID:          362301da643102b9f38477387e2193e57abaa590
      Summary:            UEFI revocation database
      Current version:    77
      Minimum Version:    77
      Vendor:             UEFI:Linux Foundation
      Install Duration:   1 second
      Update State:       Failed
      Update Error:       failed to write-firmware: Blocked executable in the ESP, ensure grub and shim are up to date: failed to load /boot/efi/EFI/debian/: Error opening file /boot/efi/EFI/debian: Is a directory
      Last modified:      2024-12-10
      GUIDs:              d07ff664-b0e1-5f4e-a723-d7fbcbfcb94f ← UEFI\CRT_3CD3F0309EDAE228767A976DD40D9F4AFFC4FBD5218F2E8CC3C9DD97E8AC6F9D&ARCH_X64
                          f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
                          b621f4ef-160a-5139-9fb1-6e89d03cc8db ← UEFI\CRT_81CAE10B24C40F687D7E695BBCFA02AF635148588B35712A25C2FCDA16DC776C&ARCH_X64
      Device Flags:       • Internal device
                          • Updatable
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Only version upgrades are allowed
                          • Signed Payload
```

</details>

**System UEFI configuration**
Please provide the output of the following commands:
[... 558 lines cut ...]
│       ├── zfs.mod
│       └── zstd.mod
├── initrd.img-6.12.3-amd64
├── initrd.img-6.12.3-custom
├── initrd.img-6.12.4-custom
├── lost+found
├── System.map-6.12.3-amd64
├── System.map-6.12.3-custom
├── System.map-6.12.4-custom
├── vmlinuz-6.12.3-amd64
├── vmlinuz-6.12.3-custom
└── vmlinuz-6.12.4-custom

54 directories, 489 files

```

**Additional questions**

- Operating system and version: *Debian sid*
- Have you tried rebooting? *yes*
- Is this a regression? *not sure*
- Are you using an NVMe disk? *yes*
- Is secure boot enabled? *yes*
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? *no*

superm1, 2024-12-11:
Can you please reproduce using `sudo fwupdtool update -vv 2>&1 | tee fwupd.log` and share `fwupd.log`?

cgzones, 2024-12-11:
<details>

<summary>update log</summary>

```
[...]
14:33:27.247 FuEngine             resetting update motd timeout
14:33:27.248 FuEngine             d07ff664-b0e1-5f4e-a723-d7fbcbfcb94f was not found: indexed string 'd07ff664-b0e1-5f4e-a723-d7fbcbfcb94f' was unfound
14:33:27.249 FuEngine             f8ba2887-9411-5c36-9cee-88995bb39731 matched 1 components
14:33:27.249 FuRelease            remote lvfs has kind=download and so marking as trusted
14:33:27.250 FuEngine             requirement 1.9.1 ge 2.0.3 -> org.freedesktop.fwupd passed
14:33:27.250 FuRelease            remote lvfs has kind=download and so marking as trusted
14:33:27.250 FuEngine             requirement 1.9.1 ge 2.0.3 -> org.freedesktop.fwupd passed
14:33:27.250 FuRelease            remote lvfs has kind=download and so marking as trusted
14:33:27.250 FuEngine             requirement 1.9.1 ge 2.0.3 -> org.freedesktop.fwupd passed
14:33:27.250 FuRelease            remote lvfs has kind=download and so marking as trusted
14:33:27.250 FuEngine             requirement 1.9.1 ge 2.0.3 -> org.freedesktop.fwupd passed
14:33:27.250 FuRelease            remote lvfs has kind=download and so marking as trusted
14:33:27.250 FuEngine             requirement 1.9.1 ge 2.0.3 -> org.freedesktop.fwupd passed
14:33:27.250 FuEngine             f8ba2887-9411-5c36-9cee-88995bb39731 matched 5 releases
14:33:27.250 FuEngine             b621f4ef-160a-5139-9fb1-6e89d03cc8db was not found: indexed string 'b621f4ef-160a-5139-9fb1-6e89d03cc8db' was unfound
14:33:27.250 FuEngine             ignoring 77 == 77
14:33:27.250 FuProgress           raw timing data was { 0.001, 0.000, 0.003, 0.000 } -- steps were set as [ 1 1 3 95 ] but should have been [ 24 3 73 1 ] at ../libfwupdplugin/fu-context.c:1058
14:33:27.251 FuProgress           raw timing data was { 0.000, 0.011, 0.001, 0.000, 0.000 } -- steps were set as [ 1 8 64 26 1 ] but should have been [ 1 94 4 0 1 ] at ../plugins/uefi-capsule/fu-uefi-capsule-plugin.c:993
Loading…: 0%
Writing…: 0%
Writing…: 50%
14:33:27.252 FuCommon             creating path /root/.cache/fwupd
14:33:27.252 Fwupd                downloading https://fwupd.org/downloads/fc3feb015df2710fcfa07583d31b5975ee398357016699cfff067f422ab91e13-DBXUpdate-20230509-x64.cab
14:33:27.252 pxbackend            px_manager_get_proxies_sync: url=https://fwupd.org/downloads/fc3feb015df2710fcfa07583d31b5975ee398357016699cfff067f422ab91e13-DBXUpdate-20230509-x64.cab online=1
14:33:27.252 pxbackend            px_manager_get_proxies_sync: Proxy[0] = direct://
14:33:27.252 Fwupd                Emitting ::status-changed() [downloading]
Downloading…: 0%
14:33:28.104 Fwupd                download progress: 100%
Downloading…: 100%
14:33:28.452 Fwupd                download progress: 9%
14:33:28.452 Fwupd                download progress: 14%
Downloading…: 14%
14:33:28.453 Fwupd                download progress: 19%
Downloading…: 19%
14:33:28.453 Fwupd                download progress: 24%
Downloading…: 24%
14:33:28.453 Fwupd                download progress: 29%
Downloading…: 29%
14:33:28.453 Fwupd                download progress: 33%
Downloading…: 33%
14:33:28.453 Fwupd                download progress: 38%
Downloading…: 38%
14:33:28.453 Fwupd                download progress: 43%
Downloading…: 43%
14:33:28.454 Fwupd                download progress: 48%
Downloading…: 48%
14:33:28.454 Fwupd                download progress: 53%
Downloading…: 53%
14:33:28.454 Fwupd                download progress: 58%
Downloading…: 58%
14:33:28.454 Fwupd                download progress: 62%
Downloading…: 62%
14:33:28.454 Fwupd                download progress: 67%
Downloading…: 67%
14:33:28.455 Fwupd                download progress: 72%
Downloading…: 72%
14:33:28.455 Fwupd                download progress: 77%
Downloading…: 77%
14:33:28.455 Fwupd                download progress: 82%
Downloading…: 82%
14:33:28.456 Fwupd                download progress: 87%
Downloading…: 87%
14:33:28.456 Fwupd                download progress: 92%
Downloading…: 92%
[... 588 lines cut ...]
14:33:29.028 FuPlugin             cleanup(jabra)
14:33:29.028 FuDeviceList         no replug or re-enumerate required
14:33:29.028 FuHistory            modifying device UEFI dbx [362301da643102b9f38477387e2193e57abaa590]
14:33:29.030 FuPlugin             composite_cleanup(dell_dock)
14:33:29.030 FuPlugin             composite_cleanup(dell_kestrel)
14:33:29.030 FuPlugin             composite_cleanup(logind)
14:33:29.030 FuPluginLogind       closed logind fd 19
14:33:29.030 FuPlugin             composite_cleanup(logitech_tap)
14:33:29.030 FuPlugin             composite_cleanup(wacom_usb)
14:33:29.030 FuPlugin             composite_cleanup(thunderbolt)
14:33:29.030 FuDeviceList         no replug or re-enumerate required
14:33:29.030 FuIdle               uninhibiting: timeout,signals by update
14:33:29.030 FuIdle               now inhibited: none
14:33:29.030 FuEngine             clearing device-changed allowlist as transaction done
failed to write-firmware: Blocked executable in the ESP, ensure grub and shim are up to date: failed to load /boot/efi/EFI/debian/: Error opening file /boot/efi/EFI/debian: Is a directory
Devices with no available firmware updates: 
 • KINGSTON SKC3000S1024G
 • SBAT
 • SSD 840 Series
 • SSD 850 EVO 250GB
14:33:29.030 FuMain               skipping reboot check
```
</details>

Full log (due to size constraints) available at https://brei.seltendoof.de/rCMj+RHUawo

superm1, 2024-12-11:
So from this entry:
```
Boot0002* debian        HD(1,GPT,f15f0b69-1c4d-4662-8653-28ef1cc602a0,0x800,0x100000)/File(\EFI\debian\shimx64.efi) File(.)
      dp: 04 01 2a 00 01 00 00 00 00 08 00 00 00 00 00 00 00 00 10 00 00 00 00 00 69 0b 5f f1 4d 1c 62 46 86 53 28 ef 1c c6 02 a0 02 02 / 04 04 34 00 5c 00 45 00 46 00 49 00 5c 00 64 00 65 00 62 00 69 00 61 00 6e 00 5c 00 73 00 68 00 69 00 6d 00 78 00 36 00 34 00 2e 00 65 00 66 00 69 00 00 00 / 7f ff 04 00
    data: 00 00 42 4f
```

It appears it's causing us to look at a directory.
```
14:33:29.024 FuContext            check for 2nd stage bootloader: /boot/efi/EFI/debian/
```

Try this: https://github.com/fwupd/fwupd/pull/8226

commit e41f1b78 (2024-12-11): trivial: Exclude directories from second stage bootloader checks

Fixes: #8224
PR #8226 (2024-12-11): trivial: Exclude directories from second stage bootloader checks
Fixes: #8224

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [ ] Code fix
- [ ] Feature
- [ ] Documentation
commit e948717d (2024-12-11): trivial: Exclude directories from second stage bootloader checks

Fixes: #8224
closed 2024-12-11 by superm1