fwupd/fwupd #8227: uefi-capsule: insufficient permission to write the capsule files in ESP (confined snap fwupd)

CragW, 2024-12-13:
**Describe the bug**
The confined snap fwupd cannot install capsule image files onto ESP.

```
$ sudo fwupd.fwupdmgr install
...
Perform operation? [Y|n]: y
Writing…                 [                                       ] Less than one minute remaining…
failed to write-firmware: Failed to create file “/boot/efi/EFI/dell/bios/recovery/BIOS_TRS.rcv.UJ18X2”: Permission denied
```

**Steps to Reproduce**
```
$ sudo fwupd.fwupdmgr install
```

**Expected behavior**
Install firmware updates on specified location successfully.

**fwupd version information**

```shell
$ fwupd.fwupdmgr --version
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.3
compile   com.hughsie.libxmlb           0.3.21
compile   io.snapcraft.fwupd            6851
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
runtime   com.hughsie.libjcat           0.2.0
runtime   com.hughsie.libxmlb           0.3.21
runtime   org.kernel                    6.8.0-1018-oem
runtime   org.freedesktop.fwupd         2.0.3
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): snap

<details>

<summary>**fwupd device information**</summary>

```shell
├─System Firmware:
│ │   Device ID:          47680ebdf8d366c510736f7c2581cd1b7085e776
│ │   Summary:            UEFI System Resource Table device (Updated via capsule-on-disk)
│ │   Current version:    1.6.1
│ │   Vendor:             Dell (DMI:Dell Inc.)
│ │   Update State:       Success
│ │   GUID:               1a99170e-ad92-46c1-a8bb-d1e221fd7537
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │   Device Requests:    • Message

```

</details>

**Additional questions**

- Operating system and version: Ubuntu 24.04
- Have you tried rebooting? Yes
- Is this a regression? No

CragW, 2024-12-13:
@valentindavid do you think we need below interface in snapd?
```
/boot/efi/EFI/dell/bios/recovery/
```

bboozzoo, 2025-01-13:
Thanks for the snapd upstream PR. I had a quick look at it looks good to me. Requested a review from other folks as well and hopefully this should be in snapd `edge` soon.

closed 2025-01-20 by CragW