fwupd/fwupd #8229: Ubuntu 22.04.5: Since today service fwupd-refresh.service fails with Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 17f6e9f089531f434a6cd3f0b46e603656971fc4

TheReptile, 2024-12-16:
**Describe the bug**
Since today the service fwupd-refresh.service fails on multiple Ubuntu 22.04.5 VMs.
When I manually try to execute `/usr/bin/fwupdmgr refresh` I get this error: 
`Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 17f6e9f089531f434a6cd3f0b46e603656971fc4`

**Steps to Reproduce**
Execute `sudo /usr/bin/fwupdmgr refresh` in a shell and get the error message.

**Expected behavior**
No error message after the refresh.

**fwupd version information**

```shell
fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.kernel                    5.15.0-126-generic
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Normal Ubuntu server install installed via apt.

<summary>**fwupd device information**</summary>

```shell
fwupdmgr get-devices --show-all-devices
WARNING: UEFI firmware can not be updated in legacy BIOS mode
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:legacy-bios for more information.
vServer
│
├─DVD-ROM:
│     Device ID:          2428d7c55669781587dd33ed050bdc45c04bc97e
│     Summary:            SCSI device
│     Current version:    2.5+
│     Vendor:             QEMU (SCSI:QEMU)
│     GUIDs:              94d1701f-a76b-5047-8617-5421edfa80af ← SCSI\VEN_QEMU&DEV_QEMU-DVD-ROM&REV_2.5+
│                         97cdce03-65bc-5402-991a-eda44f60dac3 ← SCSI\VEN_QEMU&DEV_QEMU-DVD-ROM
│   
├─Display controller:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    02
│     Vendor:             PCI:0x1234
│     GUIDs:              e358a53d-98bc-5565-b55e-7df8e0d06c5e ← PCI\VEN_1234&DEV_1111&SUBSYS_1AF41100&REV_02
│                         7365091f-756a-5c83-878c-edd1120ca718 ← PCI\VEN_1234&DEV_1111&SUBSYS_1AF41100
│                         06208e9f-1dd0-5857-b700-3d77525793aa ← PCI\VEN_1234&DEV_1111&REV_02
│                         af9ff5a0-c613-5da3-bab8-5d411adebbca ← PCI\VEN_1234&DEV_1111
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─EPYC Processor:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             AMD
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         997104da-ec11-5e04-9d94-ac5501fbf609 ← CPUID\PRO_0&FAM_17
│                         2b9136a8-1b96-5f53-9a69-57baf6f94c9b ← CPUID\PRO_0&FAM_17&MOD_31
│                         a477c4dd-63da-5027-afd3-43ae87333563 ← CPUID\PRO_0&FAM_17&MOD_31&STP_0
│     Device Flags:       • Internal device
│   
├─HARDDISK:
│     Device ID:          82359aa31a320a1f8d340ca6f775a0e942a6e5b0
│     Summary:            SCSI device
│     Current version:    2.5+
│     Vendor:             QEMU (SCSI:QEMU)
│     GUIDs:              6f0e893a-b5e8-5bd1-8459-b8de43e2b6aa ← SCSI\VEN_QEMU&DEV_QEMU-HARDDISK&REV_2.5+
│                         006a78e7-3340-5b40-9776-70db07abe578 ← SCSI\VEN_QEMU&DEV_QEMU-HARDDISK
│     Device Flags:       • Internal device
│   
└─Virtio network device:
      Device ID:          8bc8891dabab35ae76292428f63359adabdb99de
      Vendor:             Red Hat, Inc. (PCI:0x1AF4)
      GUIDs:              b99062d8-3ff1-57a1-83cc-1f7259aaa31b ← PCI\VEN_1AF4&DEV_1000&SUBSYS_1AF40001&REV_00
                          b93ef629-0df1-5505-9fee-6992b8b9abd8 ← PCI\VEN_1AF4&DEV_1000&SUBSYS_1AF40001
                          62deba3e-b70b-51b5-b66a-919b6727f5cf ← PCI\VEN_1AF4&DEV_1000&REV_00
                          21c85fac-5270-576f-a84e-04969f8cf75a ← PCI\VEN_1AF4&DEV_1000
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available
```

</details>

**Additional questions**

- Operating system and version: Ubuntu 22.04.5
- Have you tried rebooting? YES
- Is this a regression? YES

TheReptile, 2024-12-16:
It seems the firmware package has changed.
On some machines I see this:
```
(fwupdmgr:1880005): Fwupd-DEBUG: 11:53:34.108: changing metadata URI from https://cdn.fwupd.org/downloads/firmware.xml.gz to https://cdn.fwupd.org/downloads/firmware-07651-stable.xml.gz
(fwupdmgr:1880005): Fwupd-DEBUG: 11:53:34.108: downloading https://cdn.fwupd.org/downloads/firmware-07651-stable.xml.gz
```
With now a valid signature in 
```
(fwupdmgr:1022161): Fwupd-DEBUG: 11:58:19.246: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz.jcat
```

And on some other machines I see this:
```
(fwupdmgr:1022161): Fwupd-DEBUG: 11:58:19.359: changing metadata URI from https://cdn.fwupd.org/downloads/firmware.xml.gz to https://cdn.fwupd.org/downloads/firmware-07652-stable.xml.gz
```
The joys of cdns :)

But since the valid signature is live now, I guess this can be closed.

closed 2024-12-16 by TheReptile

hughsie, 2024-12-16:
I manually regenerated the metadata after one of the workers crashed with `psycopg2.errors.DeadlockDetected: deadlock detected` -- I'll debug that today.