fwupd/fwupd #8232: Regression -- UEFI update failed (Thinkpad t14s) -- Failed to find 'HD(1,GPT...)...cap' DevicePath

gdonval, 2024-12-16:
**Describe the bug**

Re #6023 (maybe not the same cause though). Back then, we figured assuming 512b block sizes was wrong and caused the problem.

Here fwupd-efi 1.7 fails to update:

    WARNING: Failed to find 'HD(1,GPT,0C8FC2F7-EF45-472E-BCAA-D5691BD44E83)/\EFI\systemd\fw\fwupd-453a2dae-ad3f-4f54-b1c0-37bf3bcbe827.cap' DevicePath
    WARNING: Could not locate device handle: Unsupported

**Steps to Reproduce**

* fwupdmgr refresh
* fwupdmgr get-updates
* fwupdmgr update
* Reboot when prompted.

**Expected behavior**

The cap files should be found and the system should be updated.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version

compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.3
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.2
runtime   org.freedesktop.fwupd-efi     1.7
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.2
runtime   org.kernel                    6.12.4-arch1-1
runtime   org.freedesktop.fwupd         2.0.3
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `pacman`

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices

LENOVO 21F9S1JQ00
│
├─AMD Ryzen 7 PRO 7840U w/ Radeon 780M Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a704104
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              2a1df9de-fe02-59c8-8af1-32c8bb011f94 ← CPUID\PRO_0&FAM_19&MOD_74
│ │                       5a4bca32-6547-584f-bb36-920d9adede25 ← CPUID\PRO_0&FAM_19&MOD_74&STP_1
│ │   Device Flags:       • Internal device
│ │ 
│ ├─AMD Radeon 780M:
│ │ │   Device ID:        e7509615d330e48a229d2012ea768cb805694e19
│ │ │   Summary:          AMD AMD_PHOENIX_GENERIC
│ │ │   Current version:  022.012.000.027.000001
│ │ │   Vendor:           Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│ │ │   GUID:             4a1501b7-b500-5255-9d9c-41d652a4d5bc ← AMD\113-PHXGEN
│ │ │   Device Flags:     • Internal device
│ │ │                     • Can tag for emulation
│ │ │ 
│ │ └─ATNA40YK15-0:
[... 282 lines cut ...]
│   │   └── arch-linux-lts-fallback.efi
│   └── systemd
│       ├── fw
│       │   └── fwupd-453a2dae-ad3f-4f54-b1c0-37bf3bcbe827.cap
│       ├── fwupdx64.efi
│       └── systemd-bootx64.efi
└── loader
    ├── entries
    │   └── fwupd.conf
    ├── entries.srel
    ├── loader.conf
    └── random-seed

10 directories, 13 files

```

**Additional questions**

- Operating system and version: Arch (rolling), up-to-date
- Have you tried rebooting? Yep
- Is this a regression? Yep
- Are you using an NVMe disk? Yep
- Is secure boot enabled? Yep
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? Nope

superm1, 2024-12-16:
Please pick up https://github.com/fwupd/fwupd/commit/11be42d1a79bb2285df91b5d1af0e37c14342910.  Does it still happen with that?

ZJaume, 2024-12-16:
I'm getting the same error on Arch also and a Thinkpad L14. My ssd is using 4k block sizes, so I guess it's affected by the bug that commit tries to fix. Unfortunately, I've tried to compile the AUR package `fwupd-git` and tests fail:
```
4/38 fwupd-self-test                           FAIL             0.18s   killed by signal 6 SIGABRT
>>> G_TEST_BUILDDIR=/home/user/work/fwupd-git/src/build/libfwupd MSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 LD_LIBRARY_PATH=/home/user/work/fwupd-git/src/build/libfwupd MALLOC_PERTURB_=140 MESON_TEST_ITERATION=1 ASAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1 G_TEST_SRCDIR=/home/user/work/fwupd-git/src/fwupd-git/libfwupd UBSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 /home/user/work/fwupd-git/src/build/libfwupd/fwupd-self-test

33/38 PciPspNoHsi.test_pci_psp_hsi              FAIL             0.47s   exit status 1
>>> MSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 MALLOC_PERTURB_=109 DAEMON_BUILDDIR=/home/user/work/fwupd-git/src/build/src MESON_TEST_ITERATION=1 CACHE_DIRECTORY=/home/user/work/fwupd-git/src/build/cache STATE_DIRECTORY=/home/user/work/fwupd-git/src/build/state PYTHONPATH=/home/user/work/fwupd-git/src/fwupd-git/data/tests ASAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1 FWUPD_DATADIR_QUIRKS=/home/user/work/fwupd-git/src/build LIBFWUPD_BUILD_DIR=/home/user/work/fwupd-git/src/build/libfwupd UBSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 /usr/bin/python3 /home/user/work/fwupd-git/src/build/../fwupd-git/plugins/pci-psp/pci_psp_test.py PciPspNoHsi.test_pci_psp_hsi

35/38 PciPspWithHsi.test_pci_psp_hsi            FAIL             0.44s   exit status 1
>>> MSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 DAEMON_BUILDDIR=/home/user/work/fwupd-git/src/build/src MESON_TEST_ITERATION=1 MALLOC_PERTURB_=65 STATE_DIRECTORY=/home/user/work/fwupd-git/src/build/state CACHE_DIRECTORY=/home/user/work/fwupd-git/src/build/cache PYTHONPATH=/home/user/work/fwupd-git/src/fwupd-git/data/tests ASAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1 FWUPD_DATADIR_QUIRKS=/home/user/work/fwupd-git/src/build LIBFWUPD_BUILD_DIR=/home/user/work/fwupd-git/src/build/libfwupd UBSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 /usr/bin/python3 /home/user/work/fwupd-git/src/build/../fwupd-git/plugins/pci-psp/pci_psp_test.py PciPspWithHsi.test_pci_psp_hsi
```

adopauco, 2024-12-23:
I have the same issue on Lenovo laptop with nvme disk using 4096 logical and physical block sizes. 
Also can not test https://github.com/fwupd/fwupd/commit/11be42d1a79bb2285df91b5d1af0e37c14342910 but due to different test fail:
```
4/38 fwupd-self-test                           FAIL             0.16s   killed by signal 6 SIGABRT
 G_TEST_BUILDDIR=/home/andrej/.cache/paru/clone/fwupd-git/src/build/libfwupd MSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 LD_LIBRARY_PATH=/home/andrej/.cache/paru/clone/fwupd-git/src/build/libfwupd MESON_TEST_ITERATION=1 ASAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1 MALLOC_PERTURB_=194 G_TEST_SRCDIR=/home/andrej/.cache/paru/clone/fwupd-git/src/fwupd-git/libfwupd UBSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 /home/andrej/.cache/paru/clone/fwupd-git/src/build/libfwupd/fwupd-self-test

ex3mpted, 2024-12-30:
Experiencing same issue on Arch Linux with GRUB and 4096 logical and physical block size NVMe

opcod3, 2024-12-30:
Experienced the same issue on Arch Linux with systemd-boot and 4096 physical block size NVMe. Lenovo ThinkPad P16s Gen 2 AMD. 

Update is working when using fwupd-git.

superm1, 2025-01-06:
closing as dup of https://github.com/fwupd/fwupd/issues/7967

closed 2025-01-06 by superm1