fwupd/fwupd #8238: fwupd crash given error message  "flag no-hwid-guids is unknown" on i2c deivce

CragW, 2024-12-17:
**Describe the bug**
fwupd crash when adding the ddc i2c devices.

```

14:56:54.320 FuEngine             /sys/devices/pci0000:00/0000:00:02.0/drm/card1/card1-DP-2 added FuDrmDevice:
  Guid:                 f541f1eb-fd32-5d91-ada4-45ff3f10f8de <- DRM\VEN_DEL&DEV_4340 ?
  Serial:               1145581640
  Flags:                can-emulation-tag
  VendorId:             PNP:DEL
  Guid[quirk]:          d07d4a5f-22c0-5c05-8739-e3377d6aabb1 <- DRM\TYPE_DRM_CONNECTOR
  PhysicalId:           card1-DP-2
  BackendId:            /sys/devices/pci0000:00/0000:00:02.0/drm/card1/card1-DP-2
  AcquiesceDelay:       2500
  CustomFlags:          bank2-only
  PossiblePlugin:       mediatek_scaler
  PossiblePlugin:       linux_display
  ParentBackendIds:     /sys/devices/pci0000:00/0000:00:02.0
  PrivateFlags:         refcounted-proxy
  Proxy:
    Guid[quirk]:        6d11c57a-00c6-584c-82be-d05e8ecfaca8 <- I2C\NAME_AUX-USBC4-DDI-TC4-PHY-E
    Guid[quirk]:        22757966-1568-51cd-bd2f-3d9185c28f83 <- I2C\NAME_AUX-USBC4-DDI-TC4-PHY-E&HWID_acda370f-dd82-5954-b8e9-ac1d795947f1
    Guid[quirk]:        e7d890a2-ba69-5c70-a9f1-15a6c9fe4ba8 <- I2C\NAME_AUX-USBC4-DDI-TC4-PHY-E&HWID_54e22044-cf53-50db-b5e9-393de351503e
    Guid[quirk]:        84d10915-b5f3-57d8-a0bb-9198aa76efde <- I2C\NAME_AUX-USBC4-DDI-TC4-PHY-E&HWID_01b8da60-9e40-50e9-8578-7bd2d7bfb10c
    Guid[quirk]:        d2304f4c-5270-55b3-86e2-9f3260310920 <- I2C\NAME_AUX-USBC4-DDI-TC4-PHY-E&HWID_26f8072e-7686-5392-9d7e-8562c30e0c30
    Guid[quirk]:        4f81497d-0f35-5daa-8de1-c49b859e0e51 <- I2C\NAME_AUX-USBC4-DDI-TC4-PHY-E&HWID_4f0fc4d5-98e9-5d0b-b5a3-05e711242379
    Guid[quirk]:        12ca8d55-0e31-5be3-ae48-58f8d4d645d6 <- I2C\NAME_AUX-USBC4-DDI-TC4-PHY-E&HWID_bacc210c-4b38-5b3e-9713-c2613699d64d
    Guid[quirk]:        53a25721-213c-565c-b7c4-2f1449defc44 <- I2C\NAME_AUX-USBC4-DDI-TC4-PHY-E&HWID_8c9edac6-839d-5973-9654-f6fa1c6f57d7
    Guid[quirk]:        c10a0fb9-254d-5934-956e-ef35f3107a73 <- I2C\NAME_AUX-USBC4-DDI-TC4-PHY-E&HWID_e9d9375d-755c-548f-9098-f187d7c78449
    Guid[quirk]:        0400cb1d-906a-5080-ad57-bda3822bc460 <- I2C\NAME_AUX-USBC4-DDI-TC4-PHY-E&HWID_b1882640-5c9b-5a83-b5b9-e28e6dc09511
    Guid[quirk]:        d5c13e0e-a3c9-5e06-b7b3-86c509a908ac <- I2C\NAME_AUX-USBC4-DDI-TC4-PHY-E&HWID_8d2a237c-a7e8-5698-8078-d897eb1f4eed
    Guid[quirk]:        4784b972-2ad0-57c3-b252-b2f3b0fe97dc <- I2C\NAME_AUX-USBC4-DDI-TC4-PHY-E&HWID_b9948c18-4a44-5243-bda4-84f53898de04
    Guid[quirk]:        fdfe9e1b-66a5-542f-910f-2baa67a18cee <- I2C\NAME_AUX-USBC4-DDI-TC4-PHY-E&HWID_91251071-a01b-5067-9dca-81de39f555f9
    Guid[quirk]:        1e473419-e2f2-550f-b54d-b4f980015804 <- I2C\NAME_AUX-USBC4-DDI-TC4-PHY-E&HWID_a7578aa1-3485-52b8-a49f-db18b9a162cb
    Guid[quirk]:        07d18bdf-6561-502d-b7e5-4589016c529c <- I2C\NAME_AUX-USBC4-DDI-TC4-PHY-E&HWID_3b5ebb93-9c4e-57f0-9d19-fdae4e853348
    Guid[quirk]:        4f3a005f-2526-50bf-8de1-d244639b3f00 <- I2C\NAME_AUX-USBC4-DDI-TC4-PHY-E&HWID_85d38fda-fc0e-5c6f-808f-076984ae7978
    Guid[quirk]:        006d15ef-bb97-5cd2-a3d4-9592ecee884f <- I2C\NAME_AUX-USBC4-DDI-TC4-PHY-E&HWID_01ade365-1cf0-5847-ad33-18941b87441f
    Guid[quirk]:        2a0d6d4f-389c-52b4-954d-4f550b836f1c <- I2C\NAME_AUX-USBC4-DDI-TC4-PHY-E&HWID_10e925fa-54d2-5fd3-92ab-b20095ad7698
    Guid[quirk]:        1069f498-9d49-5054-9a59-bb3e13a3df9b <- I2C\NAME_AUX-USBC4-DDI-TC4-PHY-E&HWID_36cb5c6e-fb91-55e9-8077-e004f2b1ddad
    PhysicalId:         DEVPATH=/devices/pci0000:00/0000:00:02.0/drm/card1/card1-DP-2/i2c-14
    BackendId:          /sys/devices/pci0000:00/0000:00:02.0/drm/card1/card1-DP-2/i2c-14
    AcquiesceDelay:     2500
    PrivateFlags:       no-hwid-guids
  Number:               0x2
  Subsystem:            drm
  Devtype:              drm_connector
  OpenFlags:            read
  ConnectorId:          259
  Enabled:              true
  State:                connected

14:56:54.320 FuDevice             using 7ac62f1047ad254a03c22dd8b69ec7065d47563a for DEVPATH=/devices/pci0000:00/0000:00:02.0/drm/card1/card1-DP-2/i2c-14
14:56:54.320 FuDevice             FuMediatekScalerDevice flag no-hwid-guids is unknown -- use fu_device_register_private_flag()
/home/dm/mywork/fwupd/venv/bin/fwupdtool: line 66: 15574 Trace/breakpoint trap   ${SUDO} ${ENV} ${DEBUG} ${EXE} "$@"

```

[mediatek-scaler-chip-i2c-fwupd-crash.txt](https://github.com/user-attachments/files/18167358/mediatek-scaler-chip-i2c-fwupd-crash.txt)

**Steps to Reproduce**
$ fwupdtool get-devices

**Expected behavior**
Devices are probed and added to fwupd daemon successfully.

**fwupd version information**

```shell
$ fwupdtool --version
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.4
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
runtime   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.fwupd.source  2.0.3-20-ga2c32ea27
runtime   com.hughsie.libjcat           0.2.0
runtime   org.kernel                    6.8.0-50-generic
runtime   org.freedesktop.fwupd         2.0.4

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): source

**Additional questions**

- Operating system and version: Ubuntu 24.04
- Have you tried rebooting? Yes
- Is this a regression? Yes

hughsie, 2024-12-17:
Can you get a backtrace? I don't see why we'd be checking that flag -- it's supposed to be on the proxy from memory. Also; getting access to a physical machine would be great tomorrow; this is the kind of thing we **need** emulation data for.

CragW, 2024-12-17:
> Can you get a backtrace? 

does this help?
[mediatek-scaler-chip-i2c-fwupd-crash.txt](https://github.com/user-attachments/files/18167358/mediatek-scaler-chip-i2c-fwupd-crash.txt)

> we **need** emulation data for.

let's try tomorrow

superm1, 2024-12-17:
> > does this help?
> 
> Nah, something from gdb, with the callchain.

I wonder if we should put a wrapper/trap on the venv to automatically get us backtraces.

CragW, 2024-12-18:
Added gdb traces and the sha based off the source code at top.

```

02:18:50.497 FuDevice             using 7ac62f1047ad254a03c22dd8b69ec7065d47563a for DEVPATH=/devices/pci0000:00/0000:00:02.0/drm/card1/card1-DP-2/i2c-14
02:18:50.497 FuDevice             FuMediatekScalerDevice flag no-hwid-guids is unknown -- use fu_device_register_private_flag()

Thread 1 "fwupdtool" received signal SIGTRAP, Trace/breakpoint trap.
0x00007ffff759a6a1 in g_logv () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
(gdb) bt
#0  0x00007ffff759a6a1 in g_logv () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#1  0x00007ffff759a953 in g_log () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff78f8358 in fu_device_add_private_flag (self=0x555555af71e0, flag=0x555555793a60 "no-hwid-guids") at ../../libfwupdplugin/fu-device.c:375
#3  0x00007ffff7907299 in fu_device_incorporate (self=0x555555af71e0, donor=0x55555598e3f0, flag=FU_DEVICE_INCORPORATE_FLAG_ALL) at ../../libfwupdplugin/fu-device.c:6348
#4  0x00007ffff795e31c in fu_plugin_backend_device_added (self=0x555555682c00, device=0x55555598c360, progress=0x555555ae9620, error=0x7fffffffd9a8)
    at ../../libfwupdplugin/fu-plugin.c:1748
#5  0x00007ffff795e5bb in fu_plugin_runner_backend_device_added (self=0x555555682c00, device=0x55555598c360, progress=0x555555ae9620, error=0x7fffffffd9a8)
    at ../../libfwupdplugin/fu-plugin.c:1798
#6  0x00007ffff7abce3d in fu_engine_backend_device_added_run_plugin
    (self=0x55555560c1e0, device=0x55555598c360, plugin_name=0x555555990360 "mediatek_scaler", progress=0x555555ae9620, error=0x7fffffffd9a8) at ../../src/fu-engine.c:7243
#7  0x00007ffff7abcf08 in fu_engine_backend_device_added_run_plugins (self=0x55555560c1e0, device=0x55555598c360, progress=0x555555ae92c0) at ../../src/fu-engine.c:7273
#8  0x00007ffff7abd302 in fu_engine_backend_device_added (self=0x55555560c1e0, device=0x55555598c360, progress=0x5555557f49f0) at ../../src/fu-engine.c:7341
#9  0x00007ffff7abe8f2 in fu_engine_backends_coldplug_backend_add_devices (self=0x55555560c1e0, backend=0x555555610a00, progress=0x5555557a5af0, error=0x7fffffffdb00)
    at ../../src/fu-engine.c:7751
#10 0x00007ffff7abeab1 in fu_engine_backends_coldplug_backend (self=0x55555560c1e0, backend=0x555555610a00, progress=0x55555579cca0, error=0x7fffffffdb00) at ../../src/fu-engine.c:7786
#11 0x00007ffff7abec56 in fu_engine_backends_coldplug (self=0x55555560c1e0, progress=0x555555616460) at ../../src/fu-engine.c:7822
#12 0x00007ffff7ac0494 in fu_engine_load (self=0x55555560c1e0, flags=366, progress=0x555555601020, error=0x7fffffffde10) at ../../src/fu-engine.c:8193
#13 0x0000555555561d9e in fu_util_start_engine (priv=0x5555555b1ad0, flags=366, progress=0x555555601020, error=0x7fffffffde10) at ../../src/fu-tool.c:250
#14 0x0000555555564227 in fu_util_get_devices (priv=0x5555555b1ad0, values=0x555555613c90, error=0x7fffffffde10) at ../../src/fu-tool.c:875
#15 0x00007ffff7faa8a6 in fu_util_cmd_array_run (array=0x5555555b1ba0, priv=0x5555555b1ad0, command=0x7fffffffe65f "get-devices", values=0x7fffffffe3e8, error=0x7fffffffde10)
    at ../../src/fu-util-common.c:532
#16 0x0000555555571372 in main (argc=2, argv=0x7fffffffe3d8) at ../../src/fu-tool.c:5277
(gdb) quit
A debugging session is active.

        Inferior 1 [process 21154] will be killed.

Quit anyway? (y or n) y

(fwupd) dm@dm-OptiPlex-AIO-Plus-7420:~/mywork/fwupd$ git rev-parse HEAD
a2c32ea273bc367849a472c269b300823d59ed7d
(fwupd) dm@dm-OptiPlex-AIO-Plus-7420:~/mywork/fwupd$

```

closed 2024-12-18 by superm1