fwupd/fwupd #8243: fwupdmgr doesn't show new updates until restarted

JohnAZoidberg, 2024-12-19:
**Describe the bug**
`fwupdmgr get-updates` doesn't show new updates until restarted.
GNOME Firmware does show them, but fwupdmgr CLI does not.

**Steps to Reproduce**
Updating my remotes:
```
> fwupdmgr refresh --force && fwupdmgr get-updates
Updating framework-embargo
Downloading…             [***************************************]
Updating lvfs-testing
Downloading…             [***************************************]
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: Updates have been published for 2 of 2 local devices
```

No new updates?
```
> fwupdmgr get-updates
Devices with no available firmware updates:
 • System Firmware
 • WD PC SN740 SDDPNQD-1T00
Devices with the latest available firmware version:
 • Laptop Webcam Module (2nd Gen)
 • UEFI dbx
No updates available

Let's try restarting fwupd
```
> sudo systemctl restart fwupd
```

Now there are updates?
```
> fwupdmgr get-updates
Devices with no available firmware updates:
 • Fingerprint Sensor
 • WD PC SN740 SDDPNQD-1T00
Devices with the latest available firmware version:
 • Laptop Webcam Module (2nd Gen)
 • UEFI dbx
Framework
│
└─System Firmware:
[...]
```

**Expected behavior**
I don't expect having to restart the daemon to see the new updates.

**fwupd version information**

```shell
> fwupdmgr --version
compile   info.libusb                   1.0.27
compile   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.2
compile   org.freedesktop.fwupd         2.0.3
runtime   org.freedesktop.fwupd-efi     1.6
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.2
runtime   org.kernel                    6.12.0-0.rc5.20241101git6c52d4da1c74.48.fc42.x86_64
runtime   org.freedesktop.fwupd         2.0.3
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): dnf

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?

I've seen this a couple times in the last few weeks.
I  think I realized that restarting the daemon would help is because `sudo fwupdtool get-updates` immediately works after refreshing the remotes.

hughsie, 2024-12-19:
Boggle; unexpected. Can you add `-vv` to the fwupd.service file, than reboot and then grab the logs (which show no devices) from the fresh boot please?

houseofleaf, 2024-12-20:
I am also affected by this issue. fwupd version 2.0.3-2 on Arch Linux. Restarting the service allowed me to update as well.

JohnAZoidberg, 2024-12-23:
> which show no devices

Ok, I'll try that on the next firmware update.

JohnAZoidberg, 2024-12-31:
I just ran into this problem again.

[image]

I have a feeling that it might be related to the problem checks (#8258).
I noticed some weird behavior.
I think what is expected that:
- When no AC, it shows "Problems" and no available updates.
- When AC it does not show problems and it shows all updates

However sometimes, as you can see below, it shows problem AND updates.

[image]

> grab the logs (which show no devices) from the fresh boot please?

It does show some information that I cannot share yet, I will send you an email.

superm1, 2025-01-01:
Could it be related to the quirk we did for framework I wonder?

https://github.com/fwupd/fwupd/commit/6ab157f1911c9b9b980a73e6ebe665e21d7ee519

What battery level did you have in both cases?

superm1, 2025-01-01:
And also does that hwid match this hardware? If not maybe that could help explain it.

warwickmm, 2025-01-01:
I also experienced this on a Thinkpad T14s Gen 4 AMD with `fwupd 2.0.3-2` on Arch.

superm1, 2025-01-01:
> I also experienced this on a Thinkpad T14s Gen 4 AMD with `fwupd 2.0.3-2` on Arch. 

Ok. Then it shouldn't be related to that quirk, thanks.

JohnAZoidberg, 2025-01-02:
> Could it be related to the quirk we did for framework I wonder?
> 
> [6ab157f](https://github.com/fwupd/fwupd/commit/6ab157f1911c9b9b980a73e6ebe665e21d7ee519)
> 
> What battery level did you have in both cases?

Actually the opposite, I booted with 2% battery and then let it charge past the 20% minimum limit.
@hughsie found that in my log I sent him - restarting the service makes it pick up the new charging level, apparently fixed by https://github.com/fwupd/fwupd/commit/67efd92fd6021acc3f51dfc2b05ac9741d65ea5f

superm1, 2025-01-02:
Ah great, this should be fixed when we tag 2.0.4 then.

JohnAZoidberg, 2025-01-10:
I just tried from latest master and I don't see this issue anymore.
I started fwupd with 15% battery and checked `fwupdmgr get-updates`. It can show it and tells me I need to charge and connect AC.
After meeting the conditions I can normally update.

No need to restart the daemon anymore.
Thanks guys :D

closed 2025-01-10 by JohnAZoidberg