fwupd/fwupd #8251: fwupd 2.0.3: fwupd.service fails to be active on Arch GNU/Linux

ShabazzKing, 2024-12-23:
**Describe the bug**
The fwupd service fails to start. Here is the output of systemctl status fwupd.service

× fwupd.service - Firmware update daemon
     Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static)
     Active: failed (Result: exit-code) since Mon 2024-12-23 23:27:34 CET; 28s ago
 Invocation: 3fb2a3846d3b452db416b4df8f22735c
       Docs: https://fwupd.org/
    Process: 25003 ExecStart=/usr/lib/fwupd/fwupd (code=exited, status=1/FAILURE)
   Main PID: 25003 (code=exited, status=1/FAILURE)
   Mem peak: 6.4M
        CPU: 57ms

gru 23 23:27:34 Darter-Pro systemd[1]: Starting Firmware update daemon...
gru 23 23:27:34 Darter-Pro fwupd[25003]: 22:27:34.131 FuPluginFlashrom     Cannot open file stream for /dev/mtd0
gru 23 23:27:34 Darter-Pro fwupd[25003]: pcilib: Cannot map ecam region: Operacja niedozwolona.
gru 23 23:27:34 Darter-Pro systemd[1]: fwupd.service: Main process exited, code=exited, status=1/FAILURE
gru 23 23:27:34 Darter-Pro systemd[1]: fwupd.service: Failed with result 'exit-code'.
gru 23 23:27:34 Darter-Pro systemd[1]: Failed to start Firmware update daemon.

The /dev/mtd0 file:

[panlukasz@Darter-Pro ~]$ ls -l /dev/mtd0
crw------- 1 root root 90, 0 12-23 23:09 /dev/mtd0

**Steps to Reproduce**
1. Try to run fwupdmgr get-devices
2. Observe the result:
Połączenie z usługą się nie powiodło: GDBus.Error:org.freedesktop.DBus.Error.NameHasNoOwner: Could not activate remote peer 'org.freedesktop.fwupd': startup job failed
3. Run systemctl status fwupd.service and observe the result.
4. Try to sudo systemctl restart fwupd.service:
Job for fwupd.service failed because the control process exited with error code.

**Expected behavior**
fwupd.service works as expected, the deamon works, the systemd service is fully loaded.

**fwupd version information**
Version 2.0.3-2
I couldn't run fwupdmgr --version, because it outputs GDBus.Error:org.freedesktop.DBus.Error.NameHasNoOwner: Could not activate remote peer 'org.freedesktop.fwupd': startup job failed

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
pacman

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

I couldn't do that because of the bug. My computer is System76 Darter Pro.

**Additional questions**

- Operating system and version: Arch GNU/Linux
- Have you tried rebooting? Yes
- Is this a regression? I don't know. I have just started to deal with this problem.

hughsie, 2024-12-24:
Does https://github.com/fwupd/fwupd/pull/8249 fix things?

ShabazzKing, 2024-12-24:
I can check, when a new release arrives in Arch. I can't compile and swap such a low-level system component (not enough knowledge).

superm1, 2024-12-24:
> I can check, when a new release arrives in Arch. I can't compile and swap such a low-level system component (not enough knowledge).

You can add those changes to the systemd unit (fwupd.service) and then reload systemd units (systemctl reload).

No need to compile on this case.

ShabazzKing, 2024-12-24:
It doesn't fix the issue. I added fadvise64 to SystemCallFilter in /usr/lib/systemd/system/fwupd.service, then did syudo systemctl reload fwupd.service. I also did sudo systemctl daemon-reload, and the problem still exists.

Integral-Tech, 2024-12-31:
Same problem for me.

hughsie, 2025-01-07:
Someone is going to need to follow the steps in https://github.com/fwupd/fwupd/blob/main/data/meson.build#L151 -- it seems Arch specific for some reason.

superm1, 2025-01-07:
Just to confirm when you tested you tested with daemon launched from the systemd unit, not from the venv, right?

tbocek, 2025-02-03:
I have the same problem with the most recent version 2.0.5. However, only on certain devices. I am also using arch. Not sure what I should be looking for to help get this fixed...

hughsie, 2025-02-03:
> pcilib: Cannot map ecam region: Operacja niedozwolona

Maybe compile fwupd without flashrom support and try that?

tbocek, 2025-02-03:
Its not a dealbreaker, its just convenience for me. I could update my bios on my Starlite MkV with:

```sudo flashrom -p internal -w 25.01.rom -i bios --ifd -n -N```

as described here: https://github.com/StarLabsLtd/firmware/issues/184

bkfirmen, 2025-02-04:
I can confirm that; I have problems on my starlite but no problems on my other laptop.

hughsie, 2025-02-04:
> I can confirm that; I have problems on my starlite but no problems on my other laptop.

What's the difference in the `sudo fwupdtool get-devices -vv --no-timestamp` output please?

bkfirmen, 2025-02-12:
Sorry for the late answer ...
Am I getting this wrong, but shouldn't the output be different, because they are completely different devices? Although I did a `diff <(cat output_laptop.txt) <(cat output_starlite.txt) > differences.txt`

Here is the output file:
[difference.txt](https://github.com/user-attachments/files/18764746/difference.txt)

I can also send you the output of the commands on the different devices without the diff command if this helps further.

bR3iN, 2025-02-22:
Setting the `iomem` kernel parameter to `relaxed` (as suggested [here](https://github.com/StarLabsLtd/firmware/issues/179#issuecomment-2243006814)) fixed the problem for me. 

Had the problem also on a Starlite MkV but running OpenSuse Tumbleweed.

superm1, 2025-02-22:
@Sean-StarLabs should the flashrom plugin check for this and inhibit updates or avoid probing in this case perhaps?

tbocek, 2025-02-22:
nice, iomem=releaxed solves the problem. But this loosens memory access permissions, and I am not sure if this is a good solution... For upgrading I can switch, and then switch back.

Sean-StarLabs, 2025-02-22:
Probing is fine (it's onyl write ops that wont work), but I think the ideal would to check and throw some form of warning/info as to why it's inhibited.

I guess proper solution is to switch from flashrom to mtd though

superm1, 2025-02-22:
> Probing is fine (it's onyl write ops that wont work), but I think the ideal would to check and throw some form of warning/info as to why it's inhibited.
> 
> I guess proper solution is to switch from flashrom to mtd though

It **seems** that probing is failing (daemon doesn't start?)

Sean-StarLabs, 2025-02-23:
> It seems that probing is failing (daemon doesn't start?)

It sounds like bisect time - we use 1.9.25, so I can confirm that's happy. I can take a look later this week?

stalkerg, 2025-03-05:
I can confirm with Arch (CachyOS) on StarLabs device. Also, I can debug/bisect it if needed.

dsager, 2025-03-06:
FYI, this error still happens after an update to `fwupd` version `2.0.6-1` on Manjaro Linux:
```
Mar 06 12:43:44 dani-starbook systemd[1]: Starting Firmware update daemon...
Mar 06 12:43:44 dani-starbook fwupd[3017]: 11:43:44.988 FuPluginFlashrom     Cannot open file stream for /dev/mtd0
Mar 06 12:43:44 dani-starbook fwupd[3017]: pcilib: Cannot map ecam region: Operation not permitted.
Mar 06 12:43:44 dani-starbook systemd[1]: fwupd.service: Main process exited, code=exited, status=1/FAILURE
Mar 06 12:43:44 dani-starbook systemd[1]: fwupd.service: Failed with result 'exit-code'.
Mar 06 12:43:44 dani-starbook systemd[1]: Failed to start Firmware update daemon.
```

stalkerg, 2025-03-26:
@Sean-StarLabs do you have any update?

Sean-StarLabs, 2025-03-26:
No, haven't had a chance to look yet

soc7, 2025-03-27:
fwupd 2.0.7-1 here running arch and the problem persists

superm1, 2025-03-27:
I think for now as a workaround you can add the flashrom plugin into `DisabledPlugins`.  See [documentation](https://fwupd.github.io/libfwupdplugin/fwupd.conf.html#daemon-parameters) for guidance on how to do this.

soc7, 2025-03-27:
> I think for now as a workaround you can add the flashrom plugin into `DisabledPlugins`. See [documentation](https://fwupd.github.io/libfwupdplugin/fwupd.conf.html#daemon-parameters) for guidance on how to do this.

Just did it and it works, but without flashrom can I flash new versions of coreboot when they arrive?

superm1, 2025-03-27:
> Just did it and it works, but without flashrom can I flash new versions of coreboot when they arrive?

It's a workaround for fwupd failing to start.  You can at least flash other devices.  We need @Sean-StarLabs to help here with the issues with flashrom plugin and those will have to come in a future release.

superm1, 2025-03-27:
BTW there was another W/A suggested in the thread you can try with flashrom plugin enabled `iomem=releaxed `.

soc7, 2025-04-09:
Just upgraded to fwupd 2.0.8-1 and the problem persists on Arch Linux

hughsie, 2025-05-03:
Can anyone get a backtrace of the failed fwupd service? Thanks.

w3irdrobot, 2025-07-01:
it's july 1, 2025 and this is still an issue when running arch on a star labs starbook.

bkfirmen, 2025-07-22:
I can report that with the recent fwupd version (2.0.13-1) this is somehow fixed for me on EndeavourOS :D

tbocek, 2025-07-29:
For me its fixed as well, I removed iomem=releaxed, and fwupd was working (Arch/StarLite MK V). So, I guess this issue can be closed?

closed 2025-07-29 by hughsie

Sean-StarLabs, 2025-07-30:
> It's still broken on Fedora 42 (6.15) StarBook Mk VII.

No, that's lockdown working correctly - flashrom shouldn't be able to do anything when Secure Boot is enabled

stalkerg, 2025-07-30:
I can confirm for CachyOS it's also fixed.

espindola, 2025-08-31:
I just hit this on alpine 3.22.1

fwupd-2.0.13-r0
linux-lts-6.12.44-r0

iomem=releaxed did not help

espindola, 2025-08-31:
user error , iomem=releaxed did not help, but iomem=relaxed did :-)