fwupd/fwupd #8252: fwupd 2.0.3 update fails with "[Invalid UTF-8]" amongst other things

andyhhp, 2024-12-26:
**Describe the bug**

Attempting to install updates fails reliably:

```shell
$ fwupd.fwupdmgr update
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 1.27.0 to 1.32.0?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • This release contains security updates as disclosed in the Dell            ║
║ Security Advisory.                                                           ║
║ • Fixed the issue where the system cannot turn on. This issue occurs when    ║
║ you disconnect the AC adapter and the system battery during the BIOS         ║
║ update.                                                                      ║
║                                                                              ║
║ XPS 15 7590 must remain plugged into a power source for the duration of the  ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Writing…                 [                                       ]
[Invalid UTF-8]
$
```
Very occasionally, there is an error other than `[Invalid UTF-8]`.  I've encountered:
```
failed to write-firmware: sbatlevel entry shim for shim /snap/fwupd/6851/libexec/fwupd/efi/shimx64.efi.signed: CSV value not found
```
or
```
Message recipient disconnected from message bus without replying
```
too.

**Expected behavior**
The firmware update to be successful.

**fwupd version information**

```shell
$ fwupdmgr --version
compile   info.libusb                   1.0.27
compile   com.hughsie.libxmlb           0.3.21
compile   io.snapcraft.fwupd            6851
compile   com.hughsie.libjcat           0.2.0
compile   org.freedesktop.fwupd         2.0.3
runtime   org.freedesktop.fwupd-efi     1.4
runtime   com.hughsie.libjcat           0.2.0
runtime   com.hughsie.libxmlb           0.3.21
runtime   org.kernel                    5.15.0-130-generic
runtime   org.freedesktop.fwupd         2.0.3
```

This is installed on Ubuntu 22.04 which uses the snap.

```shell
$ snap info fwupd
name:      fwupd
summary:   Firmware updates for Linux
publisher: Richard Hughes
store-url: https://snapcraft.io/fwupd
contact:   https://github.com/fwupd/fwupd
license:   LGPL-2.1-or-later
description: |
  The full fwupd firmware updating stack for Linux.  Can be used to perform updates on all supported
  devices via https://fwupd.org
commands:
  - fwupd.fwupdmgr
  - fwupd.fwupdtool
services:
  fwupd:         dbus, enabled, active
  fwupd.refresh: oneshot, enabled, inactive
snap-id:      HpOj37PuyuaMUZY0NQhtwnp7oS5P8u5R
tracking:     latest/stable
refresh-date: today at 20:57 GMT
channels:
  latest/stable:    2.0.3                2024-12-11 (6851) 26MB -
  latest/candidate: 2.0.3                2024-12-06 (6851) 26MB -
  latest/beta:      ↑                                           
  latest/edge:      2.0.3-33-g576e918b3  2024-12-23 (6874) 26MB -
  1.9.x/stable:     1.9.25               2024-10-08 (6500) 24MB -
  1.9.x/candidate:  1.9.27               2024-12-05 (6844) 24MB -
  1.9.x/beta:       ↑                                           
  1.9.x/edge:       1.9.26-65-gde4905d87 2024-12-05 (6843) 24MB -
installed:          2.0.3                           (6851) 26MB -
```

Any suggestions for how to debug further?  I get the feeling this is collateral damage from not getting a sane error out of another process.

superm1, 2024-12-26:
Can you reproduce with `fwupd.fwupdtool -vv update`?

If so share the output of that.

andyhhp, 2024-12-26:
```shell
$ fwupd.fwupdmgr -vv update
(pkttyagent:11337): GLib-GIO-DEBUG: 21:28:28.413: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(/snap/fwupd/6851/bin/fwupdmgr:11309): GLib-GIO-DEBUG: 21:28:28.428: _g_io_module_get_default: Found default implementation dummy (GDummyProxyResolver) for ‘gio-proxy-resolver’
(/snap/fwupd/6851/bin/fwupdmgr:11309): GLib-GIO-DEBUG: 21:28:28.428: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:28.511: Emitting ::status-changed() [idle]
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 1.27.0 to 1.32.0?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • This release contains security updates as disclosed in the Dell            ║
║ Security Advisory.                                                           ║
║ • Fixed the issue where the system cannot turn on. This issue occurs when    ║
║ you disconnect the AC adapter and the system battery during the BIOS         ║
║ update.                                                                      ║
║                                                                              ║
║ XPS 15 7590 must remain plugged into a power source for the duration of the  ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Fwupd-INFO: 21:28:33.364: downloading https://fwupd.org/downloads/febf332e179dcfe18fe49578003295c6a0469ca78bb18886e080fe603f0324dc-firmware.cab
(/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:33.364: Emitting ::status-changed() [downloading]
Downloading…             [               /                       ]Fwupd-INFO: 21:28:33.956: download progress: 100%
Downloading…             [***************************************]Fwupd-INFO: 21:28:34.335: download progress: 0%
Downloading…             [  -                                    ]Fwupd-INFO: 21:28:34.388: download progress: 1%
<snip>
Downloading…             [************************************** ]Fwupd-INFO: 21:28:37.337: download progress: 100%
(/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:37.337: Emitting ::status-changed() [idle]
Fwupd-INFO: 21:28:37.337: status-code was 200
(/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:37.457: Emitting ::status-changed() [decompressing]
Decompressing…           [***************************************](/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:37.947: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [    \                                  ](/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:38.086: Emitting ::status-changed() [device-busy]
Waiting…                 [    \                                  ](/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:38.087: Emitting ::device-changed(dabb27f5fa6578b8243f9effee71bd33520193bc)
(/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:38.088: Emitting ::device-changed(dabb27f5fa6578b8243f9effee71bd33520193bc)
(/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:38.096: Emitting ::device-changed(dabb27f5fa6578b8243f9effee71bd33520193bc)
(/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:38.098: Emitting ::device-changed(dabb27f5fa6578b8243f9effee71bd33520193bc)
(/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:38.104: Emitting ::status-changed() [device-write]
Writing…                 [                                       ](/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:38.105: Emitting ::status-changed() [device-restart]
Restarting device…       [                                       ](/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:38.106: Emitting ::status-changed() [device-write]
Writing…                 [                                       ](/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:38.106: Emitting ::status-changed() [scheduling]
Scheduling…              [                                       ](/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:38.107: Emitting ::status-changed() [decompressing]
Decompressing…           [                                       ](/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:38.107: Emitting ::status-changed() [device-write]
Writing…                 [                                       ](/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:38.223: Emitting ::device-changed(dabb27f5fa6578b8243f9effee71bd33520193bc)
(/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:38.224: Emitting ::device-changed(dabb27f5fa6578b8243f9effee71bd33520193bc)
(/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:38.224: Emitting ::device-changed(dabb27f5fa6578b8243f9effee71bd33520193bc)
(/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:38.227: Emitting ::device-changed(dabb27f5fa6578b8243f9effee71bd33520193bc)
(/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:38.228: Emitting ::device-changed(dabb27f5fa6578b8243f9effee71bd33520193bc)
(/snap/fwupd/6851/bin/fwupdmgr:11309): Fwupd-DEBUG: 21:28:38.230: Emitting ::device-changed(dabb27f5fa6578b8243f9effee71bd33520193bc)

[Invalid UTF-8]
$ 
```

superm1, 2024-12-26:
Look closer at the command I suggested. It will be a lot of output, attach a text file.

andyhhp, 2024-12-26:
Oh, that's quite different.  Here you go
[fwupdtool-vv-update.log](https://github.com/user-attachments/files/18255539/fwupdtool-vv-update.log)

superm1, 2025-01-16:
@bboozzoo any ideas here?

bboozzoo, 2025-01-16:
I doesn't look like snap sandbox should be directly affecting this. However, I'm confused why it's trying to parse `,/snap/fwupd/6851/libexec/fwupd/efi/shimx64.efi.signed`, but since it's shim and signed, I'm assuming that this is correct behavior. Looking at snapcraft.yaml the file is copied directly from the build container, and:
```
root@u2404:~# dpkg -S /usr/lib/shim/shimx64.efi
shim-signed: /usr/lib/shim/shimx64.efi
root@u2404:~# ll /usr/lib/shim/shimx64.efi.signed
lrwxrwxrwx 1 root root 36 Apr  4  2024 /usr/lib/shim/shimx64.efi.signed -> /etc/alternatives/shimx64.efi.signed
root@u2404:~# ll /etc/alternatives/shimx64.efi.signed
lrwxrwxrwx 1 root root 39 Apr  4  2024 /etc/alternatives/shimx64.efi.signed -> /usr/lib/shim/shimx64.efi.signed.latest
root@u2404:~# dpkg -S /usr/lib/shim/shimx64.efi.signed.latest
shim-signed: /usr/lib/shim/shimx64.efi.signed.latest
```

Is there a way to extract sbatlevel?

cc @valentindavid any ideas?

superm1, 2025-01-17:
I installed the snap and pulled the EFI binary out.

[shimx64.efi.signed.gz](https://github.com/user-attachments/files/18447728/shimx64.efi.signed.gz)

Trying to parse it, to me it looks like it's missing `component_generation` for `shim` in `previous` entry.

I used `fwupdtool firmware-parse shimx64.efi.signed pefile` and this is the relevant snippet for the `previous` entry.

```
    <firmware gtype="FuCsvFirmware">
      <flags>no-auto-detection</flags>
      <id>previous</id>
      <offset>0xc</offset>
      <data size="0x2d">sbat,1,2023012900.shim,2.grub,3.grub.debian,4</data>
      <write_column_ids>false</write_column_ids>
      <firmware gtype="FuCsvEntry">
        <id>sbat</id>
        <data size="0x11">sbat,1,2023012900</data>
        <values>
          <component_generation>1</component_generation>
        </values>
      </firmware>
      <firmware gtype="FuCsvEntry">
        <id>shim</id>
        <idx>0x1</idx>
        <data size="0x6">shim,2</data>
        <values>
        </values>
      </firmware>
      <firmware gtype="FuCsvEntry">
        <id>grub</id>
        <idx>0x2</idx>
        <data size="0x6">grub,3</data>
        <values>
        </values>
      </firmware>
    </firmware>
```

So if `component_generation` is mandatory to me this seems like a real problem?

bboozzoo, 2025-01-17:
contents of `.sbat` seem ok:
```
$ objcopy -O binary -j .sbat squashfs-root/libexec/fwupd/efi/shimx64.efi.signed sbat
$ cat sbat
sbat,1,SBAT Version,sbat,1,https://github.com/rhboot/shim/blob/main/SBAT.md
shim,4,UEFI shim,shim,1,https://github.com/rhboot/shim
shim.ubuntu,1,Ubuntu,shim,15.8-0ubuntu1,https://www.ubuntu.com/
```

and `.sbatlevel`:

```
$xxd sbatlevel
00000000: 0000 0000 0800 0000 3700 0000 7362 6174  ........7...sbat
00000010: 2c31 2c32 3032 3330 3132 3930 300a 7368  ,1,2023012900.sh
00000020: 696d 2c32 0a67 7275 622c 330a 6772 7562  im,2.grub,3.grub
00000030: 2e64 6562 6961 6e2c 340a 0073 6261 742c  .debian,4..sbat,
00000040: 312c 3230 3234 3031 3039 3030 0a73 6869  1,2024010900.shi
00000050: 6d2c 340a 6772 7562 2c33 0a67 7275 622e  m,4.grub,3.grub.
00000060: 6465 6269 616e 2c34 0a00                 debian,4..
```

aside from efi header, the rest seems to match the content outlined here https://github.com/rhboot/shim/blob/main/SbatLevel_Variable.txt or am I missing something?

bboozzoo, 2025-01-17:
Added some debugs to FuCsvFirmware and FuCsvEntry, it's clear that the token callback is not called for the last entry:
```
08:24:43.540 FIXME                parse csv line: sbat,1,2023012900
08:24:43.540 FIXME                parse csv entry: sbat, column id $id   <-- printed from fu_csv_entry_parse_token_cb()
08:24:43.540 FIXME                parse csv entry: 1, column id component_generation
08:24:43.540 FIXME                key: $id, value: (null)
08:24:43.540 FIXME                key: component_generation, value: 1
08:24:43.540 FIXME                parse csv line: shim,2
08:24:43.540 FIXME                parse csv entry: shim, column id $id
08:24:43.540 FIXME                key: $id, value: (null)
08:24:43.540 FIXME                parse csv line: grub,3
08:24:43.540 FIXME                parse csv entry: grub, column id $id
08:24:43.540 FIXME                key: $id, value: (null)
08:24:43.540 FIXME                parse csv line: sbat,1,2024010900
```

ended up withthis ugly diff (there's definitely a better way to fix it I think):
```
diff --git a/libfwupdplugin/fu-csv-firmware.c b/libfwupdplugin/fu-csv-firmware.c
index 39536d07f..d59f2abfb 100644
--- a/libfwupdplugin/fu-csv-firmware.c
+++ b/libfwupdplugin/fu-csv-firmware.c
@@ -148,12 +148,14 @@ fu_csv_firmware_parse_line_cb(GString *token, guint token_idx, gpointer user_dat
 {
        FuCsvFirmware *self = FU_CSV_FIRMWARE(user_data);
        g_autoptr(FuFirmware) entry = fu_csv_entry_new();
+       g_autoptr(GByteArray) tmp = g_byte_array_new();
        g_autoptr(GBytes) fw = NULL;

        /* ignore blank lines */
        if (token->len == 0)
                return TRUE;

+       g_info("parse csv line: %s", token->str);
        /* title */
        if (g_str_has_prefix(token->str, "#")) {
                return fu_strsplit_full(token->str + 1,
@@ -166,7 +168,10 @@ fu_csv_firmware_parse_line_cb(GString *token, guint token_idx, gpointer user_dat
        }

        /* parse entry */
-       fw = g_bytes_new(token->str, token->len);
+       g_byte_array_append(tmp, token->str, token->len);
+       /* add terminating null */
+       g_byte_array_append(tmp, "\0", 1);
+       fw = g_bytes_new(tmp->data, tmp->len);
        fu_firmware_set_idx(entry, token_idx);
        if (!fu_firmware_add_image_full(FU_FIRMWARE(self), entry, error))
                return FALSE;
```

and the output seems correct now:

```
    <id>.sbatlevel</id>
    <offset>0x7ec00</offset>
    <data size="0x200">[GInputStream]</data>
    <firmware gtype="FuCsvFirmware">
      <flags>no-auto-detection</flags>
      <id>previous</id>
      <offset>0xc</offset>
      <data size="0x2d">sbat,1,2023012900.shim,2.grub,3.grub.debian,4</data>
      <write_column_ids>false</write_column_ids>
      <firmware gtype="FuCsvEntry">
        <id>sbat</id>
        <data size="0x12">sbat,1,2023012900</data>
        <values>
          <component_generation>1</component_generation>
          <date_stamp>2023012900</date_stamp>
        </values>
      </firmware>
      <firmware gtype="FuCsvEntry">
        <id>shim</id>
        <idx>0x1</idx>
        <data size="0x7">shim,2</data>
        <values>
          <component_generation>2</component_generation>
        </values>
      </firmware>
      <firmware gtype="FuCsvEntry">
        <id>grub</id>
        <idx>0x2</idx>
        <data size="0x7">grub,3</data>
        <values>
          <component_generation>3</component_generation>
        </values>
      </firmware>
    </firmware>
```

closed 2025-01-17 by superm1