fwupd/fwupd #8257: fwupdmgr update displays the lesser firmware version instead of the greater firmware version

mjekdi, 2024-12-30:
**Describe the bug**
I have a firmware package that contains two firmware binaries. One is a bridge firmware with a lesser version string (EIFK31T6) and the other is a release firmware with a greater version string (EIFK31.7). So the upgrade path is EIFK31.6 -> EIFK31T6 -> EIFK31.7. When running `fwupdmgr update` it displays the lesser version string of the bridge firmware. I think it would be less confusing to display the greater version string of the release firmware (EIFK31.7) since that is what end users are expecting to update to.

**Steps to Reproduce**
Run `fwupmgr update` and it shows the lesser firmware version:
Upgrade KINGSTON SFYRSK500G from EIFK31.6 to EIFK31T6?

**Expected behavior**
Run `fwupdmgr update` and it shows the greater firmware version:
Upgrade KINGSTON SFYRSK500G from EIFK31.6 to EIFK31.7?

**fwupd version information**
[fwupdmgr-version.txt](https://github.com/user-attachments/files/18278351/fwupdmgr-version.txt)

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
apt

**fwupd device information**
[fwupdmgr-get-devices.txt](https://github.com/user-attachments/files/18278348/fwupdmgr-get-devices.txt)
[fwupdmgr-get-updates.txt](https://github.com/user-attachments/files/18278349/fwupdmgr-get-updates.txt)
[fwupdmgr-update.txt](https://github.com/user-attachments/files/18278350/fwupdmgr-update.txt)

**Additional questions**

- Operating system and version:
ubuntu 24.04.1
- Have you tried rebooting?
NA
- Is this a regression?
No

superm1, 2025-01-02:
Hmm, if you need to go to the bridge firmware **first** isn't this totally expected behavior?  You have to run the update twice to go normal -> bridge -> update.

mjekdi, 2025-01-02:
I made a single firmware package that contains both EIFK31T6 and EIFK31.7 firmware binaries with the goal being to apply both firmwares in a single update instead of two separate updates.

So **normal -> bridge -> update** becomes **normal -> bridge + update** if that makes sense?

superm1, 2025-01-02:
I think what you're trying to do makes a lot of sense, but I also think that we need a little bit of other work in the daemon to do a complicated update like this. 

I don't believe we have a sense of a multi stage update on "the same device".

IoW It's going to need some design work in the daemon.

@hughsie  💭?

mjekdi, 2025-01-03:
To be clear the update package I have is working and does apply both updates in a single update step successfully.

My only issue is with the prompt when running `fwupdmgr update` where it shows the **from** and **to** firmware versions.

So instead of showing "Upgrade KINGSTON SFYRSK500G from EIFK31.6 to **EIFK31T6**?" it would show "Upgrade KINGSTON SFYRSK500G from EIFK31.6 to **EIFK31.7**?"

superm1, 2025-01-03:
> To be clear the update package I have is working and does apply both updates in a single update step successfully.
>
> My only issue is with the prompt when running `fwupdmgr update` where it shows the [...]

Huh, how did you get it to do both steps in one go? To me this feels like it's potentially abusing unintended daemon behavior for how it handles composite updates.

Can you share a daemon log that demonstrates the whole thing end to end working this way?

mjekdi, 2025-01-03:
Sure I can do that. But can you give me the command(s) to run to get the verbose log?

superm1, 2025-01-03:
Modify the systemd unit for fwupd.service to add `-vv` to the ExecStart command. Then reload the service and restart daemon.

Then run your update sequence and save the journal output to a text file.

mjekdi, 2025-01-03:
Attaching fwupd log:
[fwupd.txt](https://github.com/user-attachments/files/18304588/fwupd.txt)

superm1, 2025-01-04:
Ah I see.  It did treat it as a composite two step update, but I think it's ONLY because you have both releases in a single cab file.

```c
Jan 03 15:05:23 matt-sped fwupd[31935]: 23:05:23.549 FuEngine             composite update 1: 71b677ca0f1bc2c5b804fa1d59e52064ce589293 EIFK31.6->EIFK31T6 (n/a, order:0: priority:1)
Jan 03 15:05:23 matt-sped fwupd[31935]: 23:05:23.549 FuEngine             composite update 2: 71b677ca0f1bc2c5b804fa1d59e52064ce589293 EIFK31.6->EIFK31.7 (n/a, order:0: priority:0)
```

Let me (attempt) to explain it.

* Basically when the tool calls `fwupdmgr update` this will call out to `fu_util_update()` which gets a list of valid upgrades from the daemon via `fwupd_client_get_upgrades()`.  In this list I'm (expecting) that you only see the release `EIFK31T6` as valid because that's the only hop the daemon _knows_ it can make at this time.
* At this time the daemon only has **METADATA**, so it won't know the  _structure_ of the cab.  So it really thinks it can only do one hop.
* Then the code calls `fu_util_update_device_with_release()` which attempts to update to `EIFK31T6`.  But since it's a CAB file with a bunch of stuff in it, the daemon processes the CAB file as a "composite" update.  That means that it will install everything in the CAB that "it can".

So in short; I feel you're getting lucky because of how the CAB file is crafted that the composite handling does what you expect.

To prove I'm right; you can split this up to two CAB files each with their own release and I would expect the daemon doesn't know what to do.

I feel that if we're going to support your explicit ask we need to come up with a way to indicate in the client that we want to upgrade ALL possible releases.  Maybe we put something in the metadata too?

This is just a back of the napkin idea:

1) Add a new metadata flag to all the applicable releases for the daemon to upgrade all possible releases.  Maybe `LVFS::UseBridgeReleases`?  You could add this to both of your releases.
2) When providing output for `fwupd_client_get_upgrades()` the daemon can do a more complicated pass of possible upgrades by using that flag.  The client should pick the latest then.
3) When the client passes the latest to the daemon, the daemon will need to look at metadata to find all the necessary bridge releases and put them into an explicit composite update flow.

Does that make sense?  @hughsie thoughts?

hughsie, 2025-01-04:
> Add a new metadata flag to all the applicable releases

Isn't this already `Flags=install-all-releases`? i.e.

	<custom>
	  <value key="LVFS::DeviceFlags">install-all-releases</value>
	</custom>

I certainly do think we need some self tests for this scenario.

superm1, 2025-01-05:
Oh yeah!  That certainly makes this clearer.  @mjekdi can you please add that to both of your releases and then test the branch I pushed?

mjekdi, 2025-01-06:
Yes I will try to test this but I will need some time to downgrade the firmware on my test samples and attempt to build fwupd from scratch.

superm1, 2025-01-06:
I slept on it some and I think we'll go with @hughsie's approach instead as it lets us do single CAB releases with bridges.  You can test with his linked branch instead when you're ready (or main if it's merged by then).

mjekdi, 2025-01-06:
I've added install-all-releases to my metainfo files like below, but it's getting stripped when uploaded to LVFS.
```
  <custom>
    <value key="LVFS::VersionFormat">plain</value>
    <value key="LVFS::UpdateProtocol">org.nvmexpress</value>
    <value key="LVFS::DeviceFlags">commit-ca3</value>
    <value key="LVFS::DeviceFlags">install-all-releases</value>
    <value key="LVFS::DeviceIntegrity">signed</value>
    <value key="LVFS::InhibitDownload"/>
  </custom>
```

superm1, 2025-01-06:
With the approach Richard did I think we can do it without that flag. Just try main now.

mjekdi, 2025-01-06:
Okay I'm now able to test this and I'm getting the same behavior:
```
fwupdmgr update
Upgrade KINGSTON SKC3000S512G from EIFK31.6 to EIFK31T6?
```
```
fwupmgr --version
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.4
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.fwupd.source  2.0.3-56-g2e7d70865
runtime   com.hughsie.libjcat           0.2.0
runtime   com.hughsie.libxmlb           0.3.21
runtime   org.kernel                    6.8.0-51-generic
runtime   org.freedesktop.fwupd         2.0.4
```

superm1, 2025-01-07:
I think we need the verbose log again to see what else stands out.

mjekdi, 2025-01-07:
[fwupd2.txt](https://github.com/user-attachments/files/18336942/fwupd2.txt)

closed 2025-01-30 by d4s

mjekdi, 2025-02-04:
I tested git again today and the issue remains unresolved.