fwupd/fwupd #8258: It's hard to know when and why available firmware updates are not able to be applied

JohnAZoidberg, 2024-12-31:
**Describe the bug**
Checking which updates are available:

```
> fwupdmgr get-updates
Devices with no available firmware updates: 
 • Fingerprint Sensor
 • UEFI dbx
 • WD PC SN740 SDDPNQD-1T00
Devices with the latest available firmware version:
 • Laptop Webcam Module (2nd Gen)
 • System Firmware
No updates available
```

it says none are available.

But when I check the devices it shows me that "System power is too low to perform the update".

```
> fwupdmgr get-devices ce9c74e0-6c5e-5f98-821f-566c1a5c3048
Selected device: Laptop Webcam Module (2nd Gen)
Framework Computer
│
└─Laptop Webcam Module (2nd Gen):
      Device ID:          ff75e69eba7f7bc73bf2e2524de4c41d32d61506
      Current version:    1.11
      Vendor:             Framework (USB:0x32AC)
      Serial Number:      FRANJBCH814175000W
      Problems:           • System power is too low to perform the update
      GUID:               ce9c74e0-6c5e-5f98-821f-566c1a5c3048 ← USB\VID_32AC&PID_001C
      Device Flags:       • Supported on remote server
                          • Updatable
                          • Unsigned Payload
                          • Can tag for emulation
```

"the update"? What does that mean? I think the wording is misleading, making me believe that there is an update.
I know that for this device specificially there is no newer update.
But what if there is a newer firmware available?

**Expected behavior**
I expect `get-updates` to show me all "available" updates. Everything that I could possibly update to, even if temporarily I can't (battery, AC).
And then it should show me that some updates are not able to be applied temporarily because of battery, AC, ...
Otherwise how do I know that I can update? Only if I check at the right time when AC is plugged in and my battery is high enough?

My system firmware has a new version, but when AC is detached, I have no way of knowing that:

```
> fwupdmgr get-updates 6ac3a633-35d3-4049-9642-816e8dd515bf
Selected device: System Firmware
Devices with the latest available firmware version:
 • System Firmware
No updates available
```

At least it should list the "Problems", letting me know that I need to connect AC to get accurate information.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): dnf

JohnAZoidberg, 2024-12-31:
Maybe `fwupdmgr get-updates` could have an extra flag for debugging that could show extra information (other than temporary failures like battery).
For example it could show that the minimum version is not matched or that the system GUID(CHID) is not matched.

For the fingerprint reader case above, I did expect that it says "latest version" available.
Only after digging through 4k lines of -vv log I found:

```
06:59:32.656 FuRelease            remote lvfs-testing has kind=download and so marking as trusted                                                                                 
06:59:32.656 FuEngine             requirement 1.0.8 ge 2.0.3 -> org.freedesktop.fwupd passed                                                                                      
06:59:32.656 FuEngine             failed to set release for component: no HWIDs matched 2185e781-ed4b-5fc2-8839-b5ddb1e7d649                                                      
06:59:32.656 FuEngine             1e8c8470-a49c-571a-82fd-19c9fa32b8c3 matched 0 releases 
06:59:32.656 FuDeviceList         ::added d787669ee4a103fe0b361fe31c10ea037c72f27c [Fingerprint Sensor]
```

Could we make this more easy to figure out?

hughsie, 2025-01-01:
What fwupd version please?

superm1, 2025-01-02:
```
Loading…                 [*******************                    ]01:09:47.895 FuPluginDfu          failed to parse interface data: no image idx 33 found in firmware
01:09:47.895 FuPluginDfu          DFU version 0x0000 invalid, v1.1 assumed
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
Devices with no available firmware updates: 
 • System Firmware
 • WD PC SN740 SDDPNQD-512G
 • Fingerprint Sensor
 • HDMI Expansion Card
 • Laptop Webcam Module (2nd Gen)
 • UEFI dbx
Framework Laptop 13 (AMD Ryzen 7040Series)
│
└─Integrated Webcam™:
  │   Device ID:          08d460be0f1f9f128413f816022a6439e0078018
  │   Summary:            Fake webcam
  │   Current version:    1.2.2
  │   Minimum Version:    1.2.0
  │   Bootloader Version: 0.1.2
  │   Vendor:             ACME Corp. (USB:0x046D)
  │   Problems:           • Device requires AC power to be connected
  │   GUID:               b585990a-003e-5270-89d5-3705a17f9a43
  │   Device Flags:       • System requires external power source
  │                       • Supported on remote server
  │                       • Cryptographic hash verification is available
  │                       • Updatable
  │                       • Unsigned Payload
  │                       • Can tag for emulation
  │   Device Requests:    • Message
  │ 
  ├─FakeDevice Firmware Update:
  │     New version:      1.2.4
  │     Remote ID:        fwupd-tests
  │     Summary:          Firmware for the ACME Corp Integrated Webcam
  │     License:          GPL-2.0+
  │     Size:             17 bytes
  │     Created:          2017-06-15
  │     Urgency:          Medium
  │     Vendor:           ACME Corp
  │     Release Flags:    • Trusted payload
  │                       • Trusted metadata
  │                       • Is upgrade
  │     Description:      
  │     Fixes another bug with the flux capacitor to prevent time going backwards.
  │   
  └─FakeDevice Firmware Update:
        New version:      1.2.3
        Remote ID:        fwupd-tests
        Summary:          Firmware for the ACME Corp Integrated Webcam
        License:          GPL-2.0+
        Size:             17 bytes
        Created:          2017-06-15
        Urgency:          Medium
        Vendor:           ACME Corp
        Release Flags:    • Trusted payload
                          • Trusted metadata
                          • Is upgrade
        Description:      
        Fixes a bug with the flux capacitor to avoid year 2038 overflow.   
```

With `fwupdtool` and `fwupdmgr` main I'm seeing problems indicated.  See above.  I think _what you're looking for_ is actually not a case of a problem though because hwid matching is a lot more fundamental.  We don't show the updates for things that don't match hwids when the metadata requires a match.

Think of it like this; if you have two vendors with the exact same component but different firmware streams (and thus hwids) and then you introduce a 3rd vendor with the same component but it's a new hwid, what would you show?  I don't think it's logical for one vendor's hw stream to EVER show up on another vendor's machines.

hughsie, 2025-01-02:
@superm1 what about if we change the wording from `System power is too low to perform the update` to just `System power is too low`?

superm1, 2025-01-02:
Sounds good

closed 2025-01-04 by hughsie

JohnAZoidberg, 2025-01-05:
Awesome, thanks!
I think my main issue was #8219 but the better wording is great, too!