fwupd/fwupd #8260: Aerox 3 Wireless USB Receiver Checksum Invalid

Flat, 2024-12-31:
**Describe the bug**
When trying to update the Aerox 3 Wireless receiver fwupd aborts with a checksum mismatch.

`checksum invalid, expected 9881ab0f42c7dbb6b131149ec48b1a744836b003321be1840db7ba405ba2908b got 596d1032e7c916822ffae84a48d5f84d641808be761723536c0d82bb83ac077f`

**Steps to Reproduce**
With an Aerox 3 Wireless USB Received connected
`fwupdmgr get-updates && fwupdatemgr update`

**Expected behavior**
Downloaded firmware should match the expected checksum

**fwupd version information**
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.3
compile   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.2
runtime   org.freedesktop.fwupd-efi     1.7
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.2
runtime   org.freedesktop.fwupd         2.0.3
runtime   org.kernel                    6.13.0-rc5-1-mainline

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
pacman

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

Gigabyte Technology Co., Ltd. X870I AORUS PRO ICE
│
├─Aerox 3 Wireless USB Receiver:
│ │   Device ID:          22393fc023ab027cc856e804c7ad83af5bb1a8ca
│ │   Current version:    1.3.1
│ │   Vendor:             SteelSeries ApS (USB:0x1038)
│ │   Install Duration:   13 seconds
│ │   GUID:               291b0582-8d08-5714-88db-986911c744d9 ← USB\VID_1038&PID_1838
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Unsigned Payload
│ │                       • Can tag for emulation
│ │ 
│ └─Aerox 3 Wireless Mouse via USB Receiver:
│       Device ID:        ca544071868a8601ce472b4cc9f12546f92f687f
│       Current version:  1.11.4
│       Vendor:           SteelSeries ApS (USB:0x1038)
│       Install Duration: 37 seconds
│       Battery:          100% (threshold 20%)
│       GUID:             541b2713-d367-5240-a443-bed07f09ffbf ← STEELSERIES\VID_1038&PID_1838&PROTOCOL_FIZZ_TUNNEL
│       Device Flags:     • Updatable
│                         • Cryptographic hash verification is available
│                         • Unsigned Payload
│     
├─AMD Radeon RX 6900 XT:
│ │   Device ID:          594baa45f6742120d71dd82166e762f1529c2a8a
│ │   Summary:            NAVI21 Gaming XTX D412
│ │   Current version:    020.001.000.043.015929
│ │   Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│ │   GUID:               36ebe767-62d9-5820-96cc-164b10059a0e ← AMD\113-D41201
│ │   Device Flags:       • Cryptographic hash verification is available
│ │                       • Can tag for emulation
│ │ 
│ ├─Unknown Device:
│ │     Device ID:        57e35fbedc32c4e5e69c55e3d7810f7ae6f1427a
[... 142 lines cut ...]
│     Summary:            Platform Key
│     GUID:               2c29036b-dd68-5a09-8a23-f3dac4144696 ← UEFI\CRT_9E5741778BF7B17F1E5C9635D9D4C35BAC829F9A
│   
└─USB4 host controller:
      Device ID:          93cc0f725649091898248865b19ddccbb674c2a4
      Summary:            Unmatched performance for high-speed I/O
      Current version:    200011.240521
      Vendor:             ASMedia Technology Inc. (THUNDERBOLT:0x00B8, TBT:0x00B8)
      GUIDs:              7d6cf48a-b678-52cb-aba9-5131ff029eff ← THUNDERBOLT\VEN_00B8&DEV_2428
                          37a14b17-74bc-502e-b0b5-90c0f1ab03b8 ← TBT-00b82428
                          4dca1ce3-3aab-5c5d-beb5-70ca4047c894 ← TBT-00b82428-controller0-0
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Device stages updates
                          • Can tag for emulation
    

</details>

**Additional questions**

- Operating system and version: Arch Linux 6.13.0-rc5
- Have you tried rebooting? Yes
- Is this a regression? N/A

d4s, 2025-01-07:
@hughsie confirmed.

The issue is here: https://github.com/fwupd/fwupd/blob/main/libfwupd/fwupd-client.c#L3387

Blob size returned by `fwupd_client_download_bytes_finish()` is `149919` bytes, while the actual size:
```
$ stat --printf="%s\n" 9881ab0f42c7dbb6b131149ec48b1a744836b003321be1840db7ba405ba2908b-aerox-3-wireless-dongle.cab
150137
```

d4s, 2025-01-07:
@hughsie tbh, I don't think the problem is on LVFS side:
```
Fwupd-INFO: 14:59:07.740: downloading https://fwupd.org/downloads/9881ab0f42c7dbb6b131149ec48b1a744836b003321be1840db7ba405ba2908b-aerox-3-wireless-dongle.cab
```
```
$ wget https://fwupd.org/downloads/9881ab0f42c7dbb6b131149ec48b1a744836b003321be1840db7ba405ba2908b-aerox-3-wireless-dongle.cab
--2025-01-07 14:59:44--  https://fwupd.org/downloads/9881ab0f42c7dbb6b131149ec48b1a744836b003321be1840db7ba405ba2908b-aerox-3-wireless-dongle.cab
Resolving fwupd.org (fwupd.org)... 54.212.102.150, 52.38.164.39, 2600:1f14:414:5602::9c36, ...
Connecting to fwupd.org (fwupd.org)|54.212.102.150|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 150137 (147K) [application/vnd.ms-cab-compressed]
Saving to: '9881ab0f42c7dbb6b131149ec48b1a744836b003321be1840db7ba405ba2908b-aerox-3-wireless-dongle.cab.1'

9881ab0f42c7dbb6b131149ec48b1a74483 100%[===================================================================>] 146.62K   254KB/s    in 0.6s    

2025-01-07 14:59:45 (254 KB/s) - '9881ab0f42c7dbb6b131149ec48b1a744836b003321be1840db7ba405ba2908b-aerox-3-wireless-dongle.cab.1' saved [150137/150137]
```

hughsie, 2025-01-08:
The LVFS fsck found quite a few problems, e.g.  `FirmwareRevision #384997 has invalid size, fixing` -- can you do "fwupdmgr refresh --force" and see if the download works correctly this time.

d4s, 2025-01-08:
Nothing changed:
```
$ LANGUAGE=C fwupdmgr refresh --force
WARNING: This package has not been validated, it may not work properly.
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: Updates have been published for 3 of 14 local devices

$ LANGUAGE=C fwupdmgr update
WARNING: This package has not been validated, it may not work properly.
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Aerox 3 Wireless USB Receiver from 1.4.2 to 1.11.4?                  ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issue:                               ║
║                                                                              ║
║ • Fix issue with wireless connection dropping during use                     ║
║                                                                              ║
║ Aerox 3 Wireless USB Receiver and all connected devices may not be usable    ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Downloading…             [*************************************  ]
checksum invalid, expected 9881ab0f42c7dbb6b131149ec48b1a744836b003321be1840db7ba405ba2908b got 596d1032e7c916822ffae84a48d5f84d641808be761723536c0d82bb83ac077f
```

d4s, 2025-01-08:
@hughsie I checked the downloaded file in HEX viewer, the full size is `150137` and looks correct, since the end of the CAB file contains `firmware.jcat`, so I don't think the problem is in LVFS. See https://github.com/fwupd/fwupd/issues/8260#issuecomment-2575121731

d4s, 2025-02-03:
@swarming1905 OOC do you still on the same  `fwupd` version?
It would be great if you provide the fwupd version information for the working variant, thanks.

d4s, 2025-02-03:
@swarming1905 Please provide the version of the daemon and client:

```shell
fwupdtool get-report-metadata
```
**or**
```shell
fwupdtool --version
```

Flat, 2025-02-03:
Looks like fwupdmgr upgrade fails with the error, while fwupdtool upgrade succeeds.

```
sudo fwupdmgr upgrade
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Aerox 3 Wireless USB Receiver from 1.3.1 to 1.11.4?                  ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issue:                               ║
║                                                                              ║
║ • Fix issue with wireless connection dropping during use                     ║
║                                                                              ║
║ Aerox 3 Wireless USB Receiver and all connected devices may not be usable    ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Downloading…             [***************************************]
checksum invalid, expected 9881ab0f42c7dbb6b131149ec48b1a744836b003321be1840db7ba405ba2908b got 596d1032e7c916822ffae84a48d5f84d641808be761723536c0d82bb83ac077f
```

```
 sudo fwupdtool upgrade
Loading…                 [*****                                  ]13:35:58.024 FuPluginUefiCapsule  skipping device that failed coldplug: ESRT GUID '00000000-0000-0000-0000-000000000000' was not valid
Loading…                 [*******************                    ]
Loading…                 [************************************** ]
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Aerox 3 Wireless USB Receiver from 1.3.1 to 1.11.4?                  ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issue:                               ║
║                                                                              ║
║ • Fix issue with wireless connection dropping during use                     ║
║                                                                              ║
║ Aerox 3 Wireless USB Receiver and all connected devices may not be usable    ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Restarting device…       [***********************************    ]13:36:21.473 FuPluginSteelSeries  failed to reset: failed to do control transfer: USB error: Pipe error [-9]
13:36:21.500 FuDevice             disabling polling: failed to ping: failed to get connection status: failed to do control transfer: USB error: Pipe error [-9]
Waiting…                 [***************************************]
Devices with the latest available firmware version:
 • Aerox 3 Wireless Mouse via USB Receiver
```

d4s, 2025-02-03:
@Flat good catch, thank you!

@hughsie indeed -- different behavior for downloading via `fwupdtool` and `fwupdmgr`, tested with the current main as well.

hughsie, 2025-02-13:
Can you try with 2.0.5 (or git main branch) please? There was an input stream issue it might have been.

Flat, 2025-02-14:
Unfortunately issue still persists.

```
❯ sudo fwupdmgr --version
compile   info.libusb                   1.0.27
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.2
compile   org.freedesktop.fwupd         2.0.6
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.fwupd.source  2.0.5-75-g2dc288e4c
runtime   com.hughsie.libjcat           0.2.2
runtime   org.kernel                    6.13.2-arch1-1
runtime   org.freedesktop.fwupd         2.0.6

❯ sudo fwupdmgr upgrade
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Aerox 3 Wireless USB Receiver from 1.3.1 to 1.11.4?                  ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issue:                               ║
║                                                                              ║
║ • Fix issue with wireless connection dropping during use                     ║
║                                                                              ║
║ Aerox 3 Wireless USB Receiver and all connected devices may not be usable    ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Downloading…             [***************************************]
Successfully installed firmware
checksum invalid, expected 9881ab0f42c7dbb6b131149ec48b1a744836b003321be1840db7ba405ba2908b got 596d1032e7c916822ffae84a48d5f84d641808be761723536c0d82bb83ac077f
```