fwupd/fwupd #8265: rp2040 devices not found after restart: permission issue?

zeha, 2025-01-03:
**Describe the bug**

fwupd 2.0.3 as installed from distribution fails to update MNT Pocket Reform Sysctl devices. They are left in the bootrom state:

```
13:12 ch@zeppelin:~ % sudo fwupdmgr install  --allow-reinstall --allow-older ~/sysctl-usbpd2.cab
Restarting device…       [                                       ]
failed to detach: failed to wait for detach replug: device fa2a92cb1a4694892659d98578eac2e13efa01a6 did not come back
```

```
Jän 03 13:08:07 zeppelin fwupd[1081]: 12:08:07.866 FuEngine             failed to add device /sys/devices/platform/fc000000.usb/xhci-hcd.1.auto/usb4/4-1/4-1:1.0/host0/target0:0:0/0:0:0:0/block/sda: failed to subclass open: failed to open /dev/sda: Operation not permitted
Jän 03 13:08:07 zeppelin fwupd[1081]: 12:08:07.874 FuEngine             failed to add device /sys/devices/platform/fc000000.usb/xhci-hcd.1.auto/usb4/4-1/4-1:1.0/host0/target0:0:0/0:0:0:0/block/sda: failed to subclass open: failed to open /dev/sda: Operation not permitted
Jän 03 13:08:07 zeppelin fwupd[1081]: 12:08:07.892 FuEngine             failed to add device /sys/devices/platform/fc000000.usb/xhci-hcd.1.auto/usb4/4-1/4-1:1.0/host0/target0:0:0/0:0:0:0/block/sda: failed to subclass open: failed to open /dev/sda: Operation not permitted
Jän 03 13:08:08 zeppelin fwupd[1081]: 12:08:08.869 FuEngine             failed to add device /sys/devices/platform/fc000000.usb/xhci-hcd.1.auto/usb4/4-1/4-1:1.0/host0/target0:0:0/0:0:0:0/block/sda/sda1: failed to subclass open: failed to open /dev/sda1: Operation not permitted
```

```
13:13 ch@zeppelin:~ % lsusb | grep Boot
Bus 004 Device 007: ID 2e8a:0003 Raspberry Pi RP2 Boot
```

```
13:13 ch@zeppelin:~ % ls -la /sys/devices/platform/fc000000.usb/xhci-hcd.1.auto/usb4/4-1/4-1:1.0/host0/target0:0:0/0:0:0:0/block/sda/sda1
total 0
drwxr-xr-x  5 root root    0  3. Jän 13:13 ./
drwxr-xr-x 10 root root    0  3. Jän 13:12 ../
-r--r--r--  1 root root 4096  3. Jän 13:13 alignment_offset
-r--r--r--  1 root root 4096  3. Jän 13:13 dev
-r--r--r--  1 root root 4096  3. Jän 13:13 discard_alignment
drwxr-xr-x  2 root root    0  3. Jän 13:13 holders/
-r--r--r--  1 root root 4096  3. Jän 13:13 inflight
-r--r--r--  1 root root 4096  3. Jän 13:12 partition
drwxr-xr-x  2 root root    0  3. Jän 13:13 power/
-r--r--r--  1 root root 4096  3. Jän 13:12 ro
-r--r--r--  1 root root 4096  3. Jän 13:12 size
-r--r--r--  1 root root 4096  3. Jän 13:12 start
-r--r--r--  1 root root 4096  3. Jän 13:13 stat
lrwxrwxrwx  1 root root    0  3. Jän 13:12 subsystem -> ../../../../../../../../../../../../../class/block/
drwxr-xr-x  2 root root    0  3. Jän 13:13 trace/
-rw-r--r--  1 root root 4096  3. Jän 13:12 uevent
13:13 ch@zeppelin:~ % ls -la /dev/sda1
brw-rw---- 1 root disk 8, 1  3. Jän 13:12 /dev/sda1
```

**Steps to Reproduce**

See above. Try to update a sysctl device while fwupd is running as started by systemd

**Expected behavior**

Updating should work. It does work when using `fwupdtool` directly:

```
13:14 ch@zeppelin:~ % sudo fwupdtool install  --allow-reinstall --allow-older ~/sysctl-usbpd2.cab
Loading…                 [*******                                ]12:14:28.832 FuDevice             20241224b is not a valid number (guessed plain)
Restarting device…       [*******************                    ]12:14:31.488 FuEngine             failed to add device /sys/devices/platform/fc000000.usb/xhci-hcd.1.auto/usb4/4-1/4-1:1.0/host0/target0:0:0/0:0:0:0/block/sda: failed to setup: failed to read version: command fail with status 2, senseKey 0x05, asc 0x20, ascq 0x00
12:14:36.430 FuDevice             20241224b is not a valid number (guessed plain)
Installing on RP2…       [**********************************     ] Less than one minute remaining…
Waiting…                 [***************************************]
13:14 ch@zeppelin:~ % lsusb | grep System
Bus 004 Device 010: ID 1209:6d07 Generic Pocket Reform System Controller 1.0
```

**fwupd version information**

```shell
13:15 ch@zeppelin:~ % fwupdmgr --version

runtime   org.freedesktop.fwupd         2.0.3
runtime   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.0
compile   com.hughsie.libjcat           0.2.0
runtime   org.kernel                    6.12.6-mnt-reform-arm64
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.3
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
from apt:

```
13:15 ch@zeppelin:~ % dpkg -l | grep fwupd
ii  fwupd                                  2.0.3-3                          arm64        Firmware update daemon
ii  fwupd-unsigned                         1:1.7-1                          arm64        Firmware update daemon (Unsigned EFI binaries)
ii  fwupd-unsigned-dev                     1:1.7-1                          arm64        Firmware update daemon (Development information)
ii  gir1.2-fwupd-2.0:arm64                 2.0.3-3                          arm64        GObject introspection data for libfwupd
ii  gnome-firmware                         47.0-1+b1                        arm64        GTK front end for fwupd
ii  libfwupd2:arm64                        1.9.26-2                         arm64        Firmware update daemon library
ii  libfwupd3:arm64                        2.0.3-3                          arm64        Firmware update daemon library
```

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices

mntre MNT Pocket Reform with RCORE RK3588 Module
│
├─Pocket Reform Input 1.0:
│     Device ID:          5fe66092f161ca848cbb6a442915cea6ce915e70
│     Current version:    0102
│     Vendor:             MNT Research (USB:0x1209)
│     Serial Number:      DE625492032E7A32
│     GUID:               a515e8b3-d950-5228-a136-ebaed3e376e8 ← USB\VID_1209&PID_6D06
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Unsigned Payload
│                         • Can tag for emulation
│
├─Pocket Reform System Controller 1.0:
│     Device ID:          fa2a92cb1a4694892659d98578eac2e13efa01a6
│     Current version:    0103
│     Vendor:             MNT Research (USB:0x1209)
│     Serial Number:      DE63544193253234
│     GUID:               df0c9fc8-88d6-5100-ba66-0f5c436fc522 ← USB\VID_1209&PID_6D07
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Supported on remote server
│                         • Unsigned Payload
│                         • Can tag for emulation
│
├─RK3588:
│     Device ID:          da4607f4b7c5f9872ac5e3a43daf584012004ac9
│     Current version:    01
│     Vendor:             Rockchip Electronics Co., Ltd (PCI:0x1D87)
│     GUID:               b50aadac-0d82-5c2c-99bb-d6bd2fb638d6 ← PCI\VEN_1D87&DEV_3588
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│                         • Can tag for emulation
│
└─SSD 1TB:
      Device ID:          0cc6c5eeda11013abde3b5c9ef46de5839146a42
      Summary:            NVM Express solid state drive
      Current version:    W0613A3
      Vendor:             Silicon Motion, Inc. (PCI:0x126F)
      Serial Number:      ALJK2023112100000520
      GUIDs:              eb4c6074-9dc2-57ae-bd43-1119ad6080f5 ← NVME\VEN_126F&DEV_2263
                          165a89d1-acc2-51c7-92b9-300207a5409c ← NVME\VEN_126F&DEV_2263&SUBSYS_126F2263
                          ea9ce501-0633-5767-8264-0c3a51b59f72 ← SSD 1TB
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Can tag for emulation

```

</details>

**Additional questions**

- Operating system and version: Debian unstable
- Have you tried rebooting? yes
- Is this a regression? I believe with 2.0.2 it worked, but cannot exactly confirm

zeha, 2025-01-03:
I'm not sure whats going on, after retrying a few times without rebooting it starts to work.
Maybe this is once again a race with udev somehow?

hughsie, 2025-01-04:
> failed to open /dev/sda: Operation not permitted

I wonder if this is the same as https://github.com/fwupd/fwupd/commit/5caa472b9de98b022c85ab0db6d26c3f0a767d20 -- i.e. can you make that `fadvise64` change in the installed `fwupd.service` and see if that fixes things. Also, I'd wonder if running `sudo /usr/libexec/fwupd/fwupd` allows it to succeed? i.e. without the systemd sandbox.

zeha, 2025-01-04:
>  i.e. can you make that `fadvise64` change in the installed `fwupd.service` and see if that fixes things.

That didn't help unfortunately.

> I'd wonder if running `sudo /usr/libexec/fwupd/fwupd` allows it to succeed? i.e. without the systemd sandbox.

Yeah that works.

Do you have any suggestions on how to debug this? Bit of a bummer that for a clean repro rebooting is necessary.

zeha, 2025-01-04:
I tried sticking strace into the fwupd.service unit, which apparently causes it to succeed. Will try again. For now here is the strace output, just in case anything obvious stands out.

[strace-succeeded.log.gz](https://github.com/user-attachments/files/18306603/strace-succeeded.log.gz)

zeha, 2025-01-04:
Here's an strace with open returning EPERM:

```
Jän 04 12:27:10 zeppelin strace[1037]: openat(AT_FDCWD, "/sys/devices/platform/fc000000.usb/xhci-hcd.1.auto/usb1/1-1/1-1:1.0/uevent", O_RDONLY) = 41
Jän 04 12:27:10 zeppelin strace[1037]: ppoll([{fd=41, events=POLLIN|POLLPRI|POLLERR}], 1, {tv_sec=0, tv_nsec=50000000}, NULL, 0) = 1 ([{fd=41, revents=POLLIN|POLLPRI|POLLERR}], left {tv_sec=0, tv_nsec=49995333})
Jän 04 12:27:10 zeppelin strace[1037]: read(41, "DEVTYPE=usb_interface\nDRIVER=usb"..., 1024) = 147
Jän 04 12:27:10 zeppelin strace[1037]: ppoll([{fd=41, events=POLLIN|POLLPRI|POLLERR}], 1, {tv_sec=0, tv_nsec=50000000}, NULL, 0) = 1 ([{fd=41, revents=POLLIN}], left {tv_sec=0, tv_nsec=49999125})
Jän 04 12:27:10 zeppelin strace[1037]: read(41, "", 1024)                      = 0
Jän 04 12:27:10 zeppelin strace[1037]: close(41)                               = 0
Jän 04 12:27:10 zeppelin strace[1037]: openat(AT_FDCWD, "/sys/devices/platform/fc000000.usb/xhci-hcd.1.auto/usb1/1-1/1-1:1.0/vendor", O_RDONLY) = -1 ENOENT (No such file or directory)
Jän 04 12:27:10 zeppelin strace[1037]: openat(AT_FDCWD, "/sys/devices/platform/fc000000.usb/xhci-hcd.1.auto/usb1/1-1/1-1:1.0/device", O_RDONLY) = -1 ENOENT (No such file or directory)
Jän 04 12:27:10 zeppelin strace[1037]: openat(AT_FDCWD, "/sys/devices/platform/fc000000.usb/xhci-hcd.1.auto/usb1/1-1/1-1:1.0/class", O_RDONLY) = -1 ENOENT (No such file or directory)
Jän 04 12:27:10 zeppelin strace[1037]: fcntl(7, F_SETLK, {l_type=F_RDLCK, l_whence=SEEK_SET, l_start=1073741824, l_len=1}) = 0
Jän 04 12:27:10 zeppelin strace[1037]: fcntl(7, F_SETLK, {l_type=F_RDLCK, l_whence=SEEK_SET, l_start=1073741826, l_len=510}) = 0
Jän 04 12:27:10 zeppelin strace[1037]: fcntl(7, F_SETLK, {l_type=F_UNLCK, l_whence=SEEK_SET, l_start=1073741824, l_len=1}) = 0
Jän 04 12:27:10 zeppelin strace[1037]: newfstatat(AT_FDCWD, "/var/cache/fwupd/quirks.db-journal", 0xffffc1f17968, 0) = -1 ENOENT (No such file or directory)
Jän 04 12:27:10 zeppelin strace[1037]: pread64(7, "\0\0\0\4\0\0\7\341\0\0\0\0\0\0\0\0", 16, 24) = 16
Jän 04 12:27:10 zeppelin strace[1037]: newfstatat(AT_FDCWD, "/var/cache/fwupd/quirks.db-wal", 0xffffc1f17968, 0) = -1 ENOENT (No such file or directory)
Jän 04 12:27:10 zeppelin strace[1037]: fstat(7, {st_mode=S_IFREG|0644, st_size=8261632, ...}) = 0
Jän 04 12:27:10 zeppelin strace[1037]: fcntl(7, F_SETLK, {l_type=F_UNLCK, l_whence=SEEK_SET, l_start=0, l_len=0}) = 0
Jän 04 12:27:10 zeppelin strace[1037]: fcntl(7, F_SETLK, {l_type=F_RDLCK, l_whence=SEEK_SET, l_start=1073741824, l_len=1}) = 0
Jän 04 12:27:10 zeppelin strace[1037]: fcntl(7, F_SETLK, {l_type=F_RDLCK, l_whence=SEEK_SET, l_start=1073741826, l_len=510}) = 0
Jän 04 12:27:10 zeppelin strace[1037]: fcntl(7, F_SETLK, {l_type=F_UNLCK, l_whence=SEEK_SET, l_start=1073741824, l_len=1}) = 0
Jän 04 12:27:10 zeppelin strace[1037]: newfstatat(AT_FDCWD, "/var/cache/fwupd/quirks.db-journal", 0xffffc1f17968, 0) = -1 ENOENT (No such file or directory)
Jän 04 12:27:10 zeppelin strace[1037]: pread64(7, "\0\0\0\4\0\0\7\341\0\0\0\0\0\0\0\0", 16, 24) = 16
Jän 04 12:27:10 zeppelin strace[1037]: newfstatat(AT_FDCWD, "/var/cache/fwupd/quirks.db-wal", 0xffffc1f17968, 0) = -1 ENOENT (No such file or directory)
Jän 04 12:27:10 zeppelin strace[1037]: fstat(7, {st_mode=S_IFREG|0644, st_size=8261632, ...}) = 0
Jän 04 12:27:10 zeppelin strace[1037]: fcntl(7, F_SETLK, {l_type=F_UNLCK, l_whence=SEEK_SET, l_start=0, l_len=0}) = 0
Jän 04 12:27:10 zeppelin strace[1037]: openat(AT_FDCWD, "/dev/sda", O_RDONLY|O_SYNC) = -1 EPERM (Operation not permitted)
Jän 04 12:27:10 zeppelin strace[1037]: write(2, "11:27:10.636 ", 13
Jän 04 12:27:10 zeppelin strace[1043]: 11:27:10.636
Jän 04 12:27:10 zeppelin strace[1037]: )           = 13
Jän 04 12:27:10 zeppelin strace[1037]: write(2, "FuEngine             ", 21
Jän 04 12:27:10 zeppelin strace[1043]: FuEngine
Jän 04 12:27:10 zeppelin strace[1037]: )   = 21
Jän 04 12:27:10 zeppelin strace[1037]: write(2, "failed to add device /sys/device"..., 200
Jän 04 12:27:10 zeppelin strace[1043]: failed to add device /sys/devices/platform/fc000000.usb/xhci-hcd.1.auto/usb1/1-1/1-1:1.0/host0/target0:0:0/0:0:0:0/block/sda: failed to subclass open: failed to open /dev/sda: Operation not permitted
```
[strace-broken.log.gz](https://github.com/user-attachments/files/18306618/strace-broken.log.gz)

zeha, 2025-01-04:
broken:

```
strace: openat(AT_FDCWD, "/dev/sda", O_RDONLY|O_SYNC) = -1 EPERM (Operation not permitted)
strace: write(2, "11:27:10.636 ", 13
strace[1043]: 11:27:10.636
strace: )           = 13
strace: write(2, "FuEngine             ", 21
strace[1043]: FuEngine
strace: )   = 21
strace: write(2, "failed to add device /sys/device"..., 200
strace[1043]: failed to add device /sys/devices/platform/fc000000.usb/xhci-hcd.1.auto/usb1/1-1/1-1:1.0/host0/target0:0:0/0:0:0:0/block/sda: failed to subclass open: failed to open /dev/sda: Operation not permitted
strace: ) = 200
```

good:
```
strace: openat(AT_FDCWD, "/dev/sda", O_RDONLY|O_SYNC) = 41
strace: ioctl(41, SG_IO, {interface_id='S', dxfer_direction=SG_DXFER_FROM_DEV, cmd_len=16, cmdp="\xc3\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00", mx_sb_len=32, iovec_count=0, dxfer_len=512, timeout=20000, flags=SG_FLAG_DIRECT_IO, dxferp="", status=0x2, masked_status=0x1, msg_status=0, sb_len_wr=18, sbp="\x70\x00\x05\x00\x00\x00\x00\x0a\x00\x00\x00\x00\x20\x00\x00\x00\x00\x00", host_status=0, driver_status=0x8, resid=512, duration=4, info=SG_INFO_CHECK}) = 0
strace: close(41)                               = 0
strace: write(2, "11:29:27.017 ", 13
strace[1150]: 11:29:27.017
strace: )           = 13
strace: write(2, "FuEngine             ", 21
strace[1150]: FuEngine
strace: )   = 21
strace: write(2, "failed to add device /sys/device"..., 230
strace[1150]: failed to add device /sys/devices/platform/fc000000.usb/xhci-hcd.1.auto/usb1/1-1/1-1:1.0/host0/target0:0:0/0:0:0:0/block/sda: failed to setup: failed to read version: command fail with status 2, senseKey 0x05, asc 0x20, ascq 0x00
strace: ) = 230
```

hughsie, 2025-01-04:
Does https://github.com/fwupd/fwupd/blob/main/data/meson.build#L151 help? `systemd-analyze syscall-filter` gives you the list.

zeha, 2025-01-04:
No, it doesn't end up crashing with that. 

What makes it work, is commenting out these from the .service file:

```
; DeviceAllow=block-sd rw
; DeviceAllow=char-aux rw
; DeviceAllow=char-cpu/* rw
; DeviceAllow=char-drm rw
; DeviceAllow=char-gpiochip rw
; DeviceAllow=char-hidraw rw
; DeviceAllow=char-usb rw
; DeviceAllow=char-usb_device rw
; DeviceAllow=char-mem rw
; DeviceAllow=char-nvme rw
; DeviceAllow=char-tpm rw
; DeviceAllow=char-mei rw
```

zeha, 2025-01-04:
Okay, I know what the problem here is.

`DeviceAllow=` has this in its documentation:

> Note that allow lists defined this way should only reference device groups which are resolvable at the time the unit is started. Any device groups not resolvable then are not added to the device allow list. In order to work around this limitation, consider extending service units with a pair of After=modprobe@xyz.service and Wants=modprobe@xyz.service lines that load the necessary kernel module implementing the device group if missing. [..]

And that is indeed our problem. The laptop I'm testing this on does not have any SCSI Disk devices. Until the rp2040 bootrom USB-Storage shows up. At this point uas, usb_storage and sd_mod get loaded. Restarting `fwupd.service` after this also works consistently.

zeha, 2025-01-04:
PR is in #8270