fwupd/fwupd #8267: Why is community firmware no longer offerred for the PinePhone modem?

hamishmb, 2025-01-03:
**Describe the question**
As of recently, I've noticed that no community firmware is available for the PinePhone modem any more. I'm testing on a PinePhone Pro with gnome-firmware, by the way. Is this a bug that no firmware is showing up, or is it intentional?

**fwupd version information**
Please provide the version of the daemon and client if applicable to your current installation of `fwupd`.

```shell
fwupdmgr --version
```

```shell
compile   info.libusb                   1.0.27
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.2
compile   org.freedesktop.fwupd         2.0.3
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.2
runtime   org.kernel                    6.11.11
runtime   org.freedesktop.fwupd         2.0.3
```

hughsie, 2025-01-04:
What version of the firmware have you got right now, and what were you expecting? Thanks.

hamishmb, 2025-01-04:
Hi there,

I've currently got 0.7.2, but the latest release is 0.8.0 (https://github.com/the-modem-distro/pinephone_modem_sdk/releases). I'm not seeing the option to downgrade to older versions either, which I was previously seeing at some point (not exactly sure when, probably over a year ago).

hamishmb, 2025-01-04:
Oh, and I'm running PostmarketOS by the way, in case that's relevant.

d4s, 2025-01-06:
@hamishmb have you checked https://github.com/the-modem-distro/pinephone_modem_sdk/issues/199 ?

@hughsie jfyi: I have PinePhone on hands to test the update if needed.

d4s, 2025-01-07:
urgh!  Looks like I was wrong. Quick check with the self-test by adding  `"USB\\VID_2C7C&PID_0125"` and `"USB\\VID_2C7C&PID_0125&REV_0318"` generates the same GUIDs for 2.x and 1.9.9.

Not sure where from `db379a33-254f-5140-b37e-d36ae7e5c039` comes but looks like not from [those instance ID](https://github.com/fwupd/fwupd/blob/2836d32db223590b0f5b5c4c6f0423dc7281abcf/plugins/modem-manager/fu-mm-device.c#L174) in `fu_mm_device_add_instance_id()`:

I have to find and use the [proper image](https://images.postmarketos.org/bpo/v23.12/pine64-pinephone/phosh/20240731-0446/) for PinePhone with 1.9.x:
```
# fwupdmgr --version
WARNING: The system clock has not been set correctly and downloading files may fail.
runtime   org.freedesktop.fwupd         1.9.9
compile   com.hughsie.libxmlb           0.3.14
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.1.14
runtime   org.kernel                    6.4.7
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.fwupd         1.9.9
runtime   org.freedesktop.gusb          0.4.9
```

```
└─QUECTEL Mobile Broadband Module:
      Device ID:          976c4a39e87f61e6940ea6a8d39c583cfa99615f
      Summary:            Quectel EG25-G modem
      Current version:    0.7.2
      Vendor:             Qualcomm (USB:0x2C7C)
      Release Branch:     community
      GUIDs:              db379a33-254f-5140-b37e-d36ae7e5c039
                          1a2996cb-f86e-5583-a464-e1b96e1c6ae9 ← USB\VID_2C7C&PID_0125&REV_0318
      Device Flags:       • Updatable
                          • System requires external power source
                          • Unsigned Payload
```
Please see the full `fwupdtool` query log attached if that helps : 
[fwupdtool_get-devices_1.9.9.log](https://github.com/user-attachments/files/18337695/fwupdtool_get-devices_1.9.9.log)

I hope that info should be enough, since I have no build environment for ARM-based devices to test with PinePhone and I'd be happy to avoid adding it. But let me know if that issue is not so obvious.

lvoegl, 2025-01-13:
`db379a33-254f-5140-b37e-d36ae7e5c039` is  `USB\VID_18D1&PID_D00D` which is the corresponding fastboot device. Unfortunately I don't know when or where this was added in 1.9.X versions.

hughsie, 2025-01-13:
> db379a33-254f-5140-b37e-d36ae7e5c039

This is a "counterpart" GUID -- i.e. what we're expecting to switch to. fwupd 2.0.x got a bit more strict, and didn't allow firmware to "match" this ID when in runtime mode. I think that's what the pinephone modem is doing -- which is probably pretty dangerous -- and is why there's a HWID on each firmware. We can work around it and restore the fwupd 1.9.x (unsafer!) behaviour using the `counterpart-visible` flag added in 57f39b96adb52428df52a8985c9aa39f6bb620a2.

e.g. does https://github.com/fwupd/fwupd/pull/8305 fix this?

hamishmb, 2025-01-14:
fwupdmgr get-devices from my PinePhone Pro:

`WARNING: This package has not been validated, it may not work properly.
PINE64 PinePhone Pro
│
├─A3A442:
│     Device ID:          b819c01fb3573d90b6110f6e56699c0282925217
│     Current version:    0x1000000000000000
│     Vendor:             EMMC:0x0000d6
│     GUIDs:              6c528c16-67bd-5f93-9859-db787c5e10f9 ← EMMC\NAME_A3A442
│                         42e489fc-0597-528f-b2f2-c05a8d90c10c ← EMMC\MAN_00D6&OEM_0103&NAME_A3A442
│                         7060db51-5f9d-5f53-a268-1cda05556454 ← EMMC\MAN_00D6&NAME_A3A442&REV_0x1000000000000000
│                         173e3fcb-036b-55a4-9145-6914d4c2e940 ← EMMC\MAN_00D6&OEM_0103&NAME_A3A442&REV_0x1000000000000000
│                         4ad1c540-521f-5fc8-86a0-b77fdc0ce206 ← EMMC\0214&0259&A3A442
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Can tag for emulation
│   
├─QUECTEL Mobile Broadband Module:
│     Device ID:          27730920c2119d128e431bbd20a8f225a2013d3a
│     Summary:            Quectel EG25-G/EC25 modem
│     Current version:    0.7.2
│     Vendor:             Qualcomm (USB:0x2C7C)
│     Release Branch:     community
│     GUIDs:              1a2996cb-f86e-5583-a464-e1b96e1c6ae9 ← USB\VID_2C7C&PID_0125&REV_0318
│                         587bf468-6859-5522-93a7-6cce552a0aa3 ← USB\VID_2C7C&PID_0125
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Unsigned Payload
│   
`

hamishmb, 2025-01-14:
Oh. that didn't format nicely at all, here's a file with it:
[fwupd.txt](https://github.com/user-attachments/files/18411848/fwupd.txt)

hughsie, 2025-05-04:
Okay; should work with 3c196af01f7e4750883589aaa6a4c7e86d0432d4 now.

hamishmb, 2025-05-10:
Cheers!