fwupd/fwupd #8268: tried to run a fwupd update, recieved this error failed to write-firmware: Blocked executable in the ESP

azizn03, 2025-01-03:
**Describe the bug**
Error I received shown below

**Steps to Reproduce**

```shell
`fwupdmgr update
WARNING: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 83 to 371?                                             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Insecure versions of the Microsoft Windows boot manager affected by Black    ║
║ Lotus were added to the list of forbidden signatures due to a discovered     ║
║ security problem.This updates the dbx to the latest release from Microsoft.  ║
║                                                                              ║
║ Before installing the update, fwupd will check for any affected executables  ║
║ in the ESP and will refuse to update if it finds any boot binaries signed    ║
║ with any of the forbidden signatures.Applying this update may also cause     ║
║ some Windows install media to not start correctly.                           ║
║                                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
╔══════════════════════════════════════════════════════════════════════════════╗
║ Full Disk Encryption Detected                                                ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Some of the platform secrets may be invalidated when updating this           ║
║ firmware. Please ensure you have the volume recovery key before continuing.  ║
║                                                                              ║
║ See https://github.com/fwupd/fwupd/wiki/Full-Disk-Encryption-Detected for    ║
║ more details.                                                                ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Verifying…               [                                       ]
failed to write-firmware: Blocked executable in the ESP, ensure grub and shim are up to date: failed to call org.freedesktop.UDisks2.Manager.GetBlockDevices(): The name is not activatable`
```
```shell
**Expected behavior**
A clear and concise description of what you expected to happen.

**fwupd version information**
`fwupdmgr --version
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.3
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.2
runtime   org.freedesktop.fwupd-efi     1.7
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.2
runtime   org.kernel                    6.6.65-1-lts
runtime   org.freedesktop.fwupd         2.0.3`
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Pacman

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
freedesktop fwupdmgr get-devices --show-all-devices
WARNING: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
LENOVO 20QD00L1UK
│
├─Unknown Device:
[... 168 lines cut ...]
│     GUID:               71599d14-9b31-5270-b3bd-74c494585820 ← UEFI\CRT_9AEF2123F4DE7C19AFABD909BB2C8CAC4411E07E
│
└─WhiskeyLake-U GT2 [UHD Graphics 620]:
      Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
      Current version:    02
      Vendor:             Intel Corporation (PCI:0x8086)
      GUID:               52aa8ac7-c837-561f-af32-b7979cd2f6ed ← PCI\VEN_8086&DEV_3EA0
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available
                          • Can tag for emulation

────────────────────────────────────────────────
Devices that were not updated correctly:
 • UEFI dbx (83 → 371)
```

</details>

**Additional questions**

- Operating system and version:
Arch Linux Linux n 6.6.65-1-lts #1 SMP PREEMPT_DYNAMIC
- Have you tried rebooting?
Yes
- Is this a regression?

superm1, 2025-01-03:
Install udisks2

azizn03, 2025-01-03:
that fixed it thanks. Curious how you came to that conclusion? @superm1

superm1, 2025-01-03:
From this

```
failed to call org.freedesktop.UDisks2.Manager.GetBlockDevices()
```

closed 2025-01-03 by superm1