fwupd/fwupd #8282: Update requires reboot for unkown reason

JohnAZoidberg, 2025-01-08:
**Describe the bug**
I update my USB camera and fwupd wants me to reboot. I don't know why.

**Steps to Reproduce**
I first checked whether the update was available:

```
> sudo /home/zoid/clone/fwupd/venv/bin/fwupdtool get-updates
Loading…                 [*****************                      ]13:25:08.386 FuPluginDfu          failed to parse interface data: no image idx 33 found in firmware
13:25:08.386 FuPluginDfu          DFU version 0x0000 invalid, v1.1 assumed
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
Devices with no available firmware updates:
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • System Firmware
Framework Laptop
│
└─Laptop Webcam Module:
  │   Device ID:          371581745d6f93adaeae648c7fdf18e7b3c5ccaf
  │   Current version:    0.5
  │   Vendor:             Framework (USB:0x32AC)
  │   Serial Number:      0001
  │   GUID:               cc6e80f4-6a4b-5d4e-bb4a-c1bfb66ce518 ← USB\VID_32AC&PID_001C
  │   Device Flags:       • Updatable
  │                       • Supported on remote server
  │                       • Unsigned Payload
  │                       • Can tag for emulation
  │
  └─Laptop Webcam Module Camera Update:
        New version:      0.11
        Remote ID:        framework-embargo
        Release ID:       105064
        Summary:          Firmware for the Framework Webcam Module
        License:          Proprietary
        Size:             229.4 kB
        Created:          2025-01-02
        Urgency:          Low
        Vendor:           Framework
        Duration:         2 minutes
        Release Flags:    • Trusted metadata
                          • Is upgrade
        Description:
        This update includes the following changes:
   
        • XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
        Checksum:         1bef4d44d17c78ff9b936356106704691fe7758fb17f8434e37f788693540e0e
```

Then I launched the update like this:

```
> sudo /home/zoid/clone/fwupd/venv/bin/fwupdtool update
Loading…                 [*****************                      ]13:26:13.795 FuPluginDfu          failed to parse interface data: no image idx 33 found in firmware
13:26:13.795 FuPluginDfu          DFU version 0x0000 invalid, v1.1 assumed
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Laptop Webcam Module from 0.5 to 0.11?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This update includes the following changes:                                  ║
║                                                                              ║
║ • XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX                    ║
║                                                                              ║
║ Laptop Webcam Module and all connected devices may not be usable while       ║
║ updating.                                                                    ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]:
Decompressing…           [ -                                     ]13:26:43.595 FuDevice             0.0.9.39 is not a valid bcd (guessed quad)
13:26:43.595 FuDevice             0.0.9.39 is not a valid bcd (guessed quad)
13:26:43.595 FuDevice             0.0.3.148 is not a valid bcd (guessed quad)
Installing on Laptop Webcam Module…*******                    ]
Restarting device…       [*******************                    ]13:26:44.226 FuPluginDfu          failed to parse interface data: no image idx 33 found in firmware
13:26:44.226 FuPluginDfu          DFU version 0x0000 invalid, v1.1 assumed
Restarting device…       [************************************   ]13:27:16.559 FuPluginDfu          failed to parse interface data: no image idx 33 found in firmware
13:27:16.559 FuPluginDfu          DFU version 0x0000 invalid, v1.1 assumed
Waiting…                 [***************************************]
An update requires a reboot to complete. Restart now? [y|N]:
Devices with the latest available firmware version:
 • System Firmware
Devices with no available firmware updates:
 • Micron XXX
 • Camera
An update requires a reboot to complete. Restart now? [y|N]:
```

Why would a DFU update of a USB camera need reboot? It doesn't and I'm not sure why fwupd added the flag.
My metainfo XML does not include that flag.

How can I debug this?

Both the System Firmware and the Micron firmware require reboot after update. But I didn't update either of them.

**Expected behavior**
A clear and concise description of what you expected to happen.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
> /home/zoid/clone/fwupd/venv/bin/fwupdtool --version
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.4
compile   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.2
runtime   org.freedesktop.fwupd-efi     1.6
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.fwupd.source  2.0.3-64-ga696579ea
runtime   com.hughsie.libjcat           0.2.2
runtime   org.kernel                    6.13.0-0.rc3.20241218gitaef25be35d23.31.fc42.x86_64
runtime   org.freedesktop.fwupd         2.0.4
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

built from source: a696579ea990068e2e4640cc91336ececc811384

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**Additional questions**

- Operating system and version:
  - Fedora Rawhide with kernel 6.13.0-0.rc3.20241218gitaef25be35d23.31.fc42.x86_64
- Have you tried rebooting?
  - Not yet
- Is this a regression?
  - I don't remember seeing it before

JohnAZoidberg, 2025-01-08:
I rebooted and tried again.

If use the daemon `sudo venv/bin/fwupd` and then `sudo venv/bin/fwupdmgr install` and select the camera and the version it does not ask me to reboot.

Then I downgraded and use `sudo venv/bin/fwupdtool update` again and it does ask me to reboot.
Then I downgraded again and use `sudo venv/bin/fwupdmgr update`  and it does NOT ask me to reboot, which is expected.

So the problem is only with `fwupdtool update`.

JohnAZoidberg, 2025-01-08:
Oh i guess it detects that I'm piping and doesn't ask for confirmation about anything?
It also doesn't ask for confirmation that I do want to run the update or not.

superm1, 2025-01-08:
Can you run with a debugger and see if `fu_util_update_device_changed_cb` gets called multiple times for some reason?  Or add some prints there to see what devices it hits.

My guess is you're hitting this block somehow:

```
	if (fwupd_device_has_flag(device, FWUPD_DEVICE_FLAG_NEEDS_REBOOT))
		priv->completion_flags |= FWUPD_DEVICE_FLAG_NEEDS_REBOOT;
```

JohnAZoidberg, 2025-01-08:
> Can you run with a debugger and see if `fu_util_update_device_changed_cb` gets called multiple times for some reason? Or add some prints there to see what devices it hits.
>
> My guess is you're [...]

In `fu_util_update_device_changed_cb` in `src/fu-tool.c` I changed it to

```
        if (fwupd_device_has_flag(device, FWUPD_DEVICE_FLAG_NEEDS_REBOOT)) {                                                                                 
                priv->completion_flags |= FWUPD_DEVICE_FLAG_NEEDS_REBOOT;                                                                                    
                g_warning("must reboot because of %s",                                                                                                       
                        fwupd_device_get_name(device));                                                                                                      
        }                                                                                                                                                    
```

And it said the following very early when I ran `sudo venv/bin/fwupdtool update -v`

```
14:39:25.379 FuMain               must reboot because of Internal SPI Controller
14:39:25.380 FuMain               must reboot because of Micron XXX      
14:39:25.380 FuMain               must reboot because of System Firmware
14:39:25.381 FuMain               must reboot because of Camera
```

superm1, 2025-01-08:
Can you do it with a debugger and get a backtrace when it's on the wrong device to understand how that happened?

superm1, 2025-01-08:
Does https://github.com/fwupd/fwupd/tree/superm1/inhibit-fwupdtool help?

If not; can you please share a sanitized `-vv` log for any strings about pre-release hardware you don't want to share or some emulation data to repro it?

JohnAZoidberg, 2025-01-08:
First time hitting the breaktpoint because of "Internal SPI Controller":
[image]

Is there something specific I should look for?

superm1, 2025-01-08:
Ah I think the problem is that in `fu_util_install()` we set the device to the release before we know it's valid, and at that time we have the device-change signal connected.  This connection isn't really needed though because `fu_util_update` calls `fu_util_install_release` which call `fu_util_install`.  

Try this! https://github.com/fwupd/fwupd/pull/8284

JohnAZoidberg, 2025-01-08:
> [inhibit-fwupdtool](https://github.com/fwupd/fwupd/tree/superm1/inhibit-fwupdtool)

Doesn't work, still asks me to reboot, and I hit the same breakpoint.

JohnAZoidberg, 2025-01-08:
> Try this! https://github.com/fwupd/fwupd/pull/8284

Works, I'm not hitting any breakpoint and I'm not asked to reboot.

closed 2025-01-08 by superm1