fwupd/fwupd #8296: PS175 is not being detected with v2.0.3

mightymos, 2025-01-10:
**Describe the bug**
A clear and concise description of what the bug is.
PS175 on Displayport to HDMI small board is not being detected in Elitedesk 800 G3.

**Steps to Reproduce**
Steps to reproduce the behavior.
Install fwupd.
Try to get devices.

**Expected behavior**
A clear and concise description of what you expected to happen.
Expected PS175 to be detected as in:
https://github.com/fwupd/fwupd/issues/6750

**fwupd version information**
Please provide the version of the daemon and client.
v2.0.3

```shell
fwupdmgr --version
```

```
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.3
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libjcat           0.2.0
runtime   com.hughsie.libxmlb           0.3.21
runtime   org.kernel                    6.12.6-amd64
runtime   org.freedesktop.fwupd         2.0.3
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
apt

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```
```
HP HP EliteDesk 800 G3 SFF
│
├─Unknown Device:
│     Device ID:          b5881e53bf55f4e454a88a354e1f9432e5780820
│     Vendor:             PNP:VSC
│     Serial Number:      UBX152800522
│     GUID:               18926853-019f-59e5-9c9c-4d05d871a8ef ← DRM\VEN_VSC&DEV_D22B
│     Device Flags:       • Can tag for emulation
│   
├─Unknown Device:
│     Device ID:          96928709c665f953e18829015803502c7fb9a1ef
│     Vendor:             PNP:VSC
│     Serial Number:      UBX152800522
│     GUID:               18926853-019f-59e5-9c9c-4d05d871a8ef ← DRM\VEN_VSC&DEV_D22B
│     Device Flags:       • Can tag for emulation
│   
├─Core™ i5-7500 CPU @ 3.40GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f8
│     Vendor:             Intel
│     GUIDs:              809a0b93-8a12-5338-a571-ad5583acf896 ← CPUID\PRO_0&FAM_06&MOD_9E
│                         d0f754d5-1395-5573-bc83-85ba955da70a ← CPUID\PRO_0&FAM_06&MOD_9E&STP_9
│     Device Flags:       • Internal device
│   
├─E2M2 64GB:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express solid state drive
│     Current version:    10100080
│     Vendor:             O2 Micro, Inc. (PCI:0x1217)
│     Serial Number:      MBC599D0048B5
│     GUIDs:              bf4c6932-98df-572a-8b79-6fdf20d46ca3 ← NVME\VEN_1217&DEV_8760
│                         959e6dbe-018a-5a3a-a4f3-9bc9ed2206fc ← NVME\VEN_1217&DEV_8760&SUBSYS_12170002
│                         26d3ea8a-46a1-5f36-a69b-653f6c269491 ← E2M2 64GB
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Can tag for emulation
│   
├─HD Graphics 630:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    04
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUID:               83d2b615-717b-5030-bc06-2994db598c7b ← PCI\VEN_8086&DEV_5912
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│                         • Can tag for emulation
│   
├─System Firmware:
│ │   Device ID:          0112f729778aa87193f281f4c6c6cb6197aa4cdc
│ │   Summary:            UEFI System Resource Table device (updated via NVRAM)
│ │   Current version:    131122
│ │   Minimum Version:    1
│ │   Vendor:             HP (DMI:HP)
│ │   Update State:       Success
│ │   GUID:               32c03543-d872-4a87-b33d-b590d6576beb
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │   Device Requests:    • Message
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  371
│       Minimum Version:  371
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            f1d99738-25d4-5f76-a2ec-87750d294993 ← UEFI\CRT_77F0715D0112EBF26BF15C0194B2395404E86731F853BBADCE4C4D745C3A78AD&ARCH_X64
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
├─TPM:
│     Device ID:          1d8d50a4dbc65618f5c399c2ae827b632b3ccc11
│     Current version:    7.62.12.13824
│     Vendor:             Infineon (TPM:IFX)
│     GUIDs:              5eebb112-75ad-5536-b173-a11eb3399402 ← TPM\VEN_IFX&DEV_0000
│                         ddf995da-1b32-5a8a-bc1b-8d5af4b38b51 ← TPM\VEN_IFX&MOD_SLB9670
│                         6d81ab63-db2e-50ac-934f-6be9accf5e02 ← TPM\VEN_IFX&DEV_0000&VER_2.0
│                         301555de-680d-5ddc-b995-7553fc9138f1 ← TPM\VEN_IFX&MOD_SLB9670&VER_2.0
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device can recover flash failures
│                         • Full disk encryption secrets may be invalidated when updating
│                         • Signed Payload
│                         • Can tag for emulation
│   
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            Hewlett-Packard Printing Device Infrastructure CA
      Vendor:             Hewlett-Packard Company
      GUID:               fa033ef8-3efa-5306-983e-37a196aba2ea ← UEFI\CRT_576F6A385DED2B5DF1FDA805614B347223B3FC4B
```
</details>

**Additional questions**

- Operating system and version:
Debian Testing (Trixie)
- Have you tried rebooting?
Yes.
- Is this a regression?
Unsure.

Board example:
[image]

mightymos, 2025-01-11:
For what it's worth, I was able to dump the MX25L2006E with a test clip.
There seems to be a similarity to the firmware versions here (though not identical):
https://gsdview.appspot.com/chromeos-localmirror/distfiles/?marker=distfiles%2Fnpu_plugin_elf-npu_ud_2024_32_rc1.tar.g%40

hughsie, 2025-01-11:
What's the full output from `sudo fwupdtool get-devices --plugins parade-lspcon -vv` please. Also, is your device bound to I2C device `1AF80175`? That's all we're matching at the moment.

mightymos, 2025-01-11:
I do not see a mention of 1AF80175 in the output.
How do I tell if the I2C device is bound or not?

<details>
<summary> fwupdtool output </summary>

```
user@debian:~$ sudo fwupdtool get-devices --plugins parade-lspcon -vv
18:39:20.760 FuDebug              verbose to debug (on console 1)
18:39:20.760 FuEngine             starting fwupd 2.0.3…
18:39:20.760 FuMain               locked /run/lock/fwupdtool
18:39:20.762 FuMain               failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Loading…                 [ -                                     ]18:39:20.763 FuConfig             loading config /etc/fwupd/fwupd.conf
18:39:20.763 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
18:39:20.763 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
18:39:20.763 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
18:39:20.763 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
18:39:20.763 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
18:39:20.763 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
18:39:20.763 FuConfig             ::configuration loaded
18:39:20.763 FuRemoteList         loading /usr/share/fwupd/metainfo
18:39:20.763 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
18:39:20.763 XbSilo               file: 04f886ff-eaee-8297-9575-b3e033ef525e, current:04f886ff-eaee-8297-9575-b3e033ef525e
18:39:20.763 XbSilo               loading silo with existing file contents
18:39:20.763 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
18:39:20.763 FuRemoteList         ::remote_list changed
18:39:20.763 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
18:39:20.763 FuRemoteList         ::remote_list changed
18:39:20.763 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
18:39:20.763 FuRemoteList         ::remote_list changed
18:39:20.764 FuEngine             priority local and vendor-directory is not download remote, so bumping
18:39:20.764 FuRemoteList         ordering lvfs-testing=lvfs+1
18:39:20.764 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Loading…                 [                                       ]18:39:20.764 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
18:39:20.764 FuHistory            got schema version of 14
Loading…                 [*                                      ]18:39:20.769 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-2.0.3/libfu_plugin_modem_manager.so)
18:39:20.770 FuQuirks             open database /var/cache/fwupd/quirks.db
18:39:20.771 FuQuirks             mtimes unchanged: quirks,pci.ids:1732500905,usb.ids:1734174757, doing nothing
18:39:20.771 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
18:39:20.771 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
18:39:20.771 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
18:39:20.771 XbSilo               file: 61f27f68-8fa1-5617-b11a-ab9ca6c33575, current:61f27f68-8fa1-5617-b11a-ab9ca6c33575
18:39:20.771 XbSilo               loading silo with existing file contents
Loading…                 [*                                      ]18:39:20.772 FuConfig             loading config /etc/fwupd/fwupd.conf
18:39:20.772 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
18:39:20.772 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
18:39:20.772 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
18:39:20.772 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
18:39:20.772 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
18:39:20.772 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
18:39:20.772 FuConfig             ::configuration loaded
18:39:20.772 FuStruct             FuStructSmbiosEp64:
  anchor_str: _SM3_
  entry_point_csum: 0x3b
  entry_point_len: 0x18
  smbios_major_ver: 0x3
  smbios_minor_ver: 0x0
  smbios_docrev: 0x0
  entry_point_rev: 0x1
  reserved0: 0x0
  structure_table_len: 0xdd6
  structure_table_addr: 0x9f762000
18:39:20.772 FuStruct             FuStructSmbiosStructure:
  type: 0x10
  length: 0x17
  handle: 0x0
18:39:20.772 FuStruct             FuStructSmbiosStructure:
  type: 0x13
  length: 0x1f
  handle: 0x5
[... 2112 lines cut ...]
  CanInvalidate:        false
FuUdevBackend:
  Name:                 udev
  Enabled:              true
  DoneSetup:            true
  CanInvalidate:        false
DoneColdplug:           true
FuBluezBackend:
  Name:                 bluez
  Enabled:              false
  DoneSetup:            false
  CanInvalidate:        false
FuParadeLspconPlugin:
  Name:                 parade_lspcon
  Flags:                ready

18:39:22.300 FuEngine             resetting update motd timeout

No detected devices
18:39:22.300 FuMain               No detected devices

18:39:22.300 FuProgress           raw timing data was { 0.001, 0.000, 0.001, 0.000 } -- steps were set as [ 1 1 3 95 ] but should have been [ 53 3 43 2 ] at ../libfwupdplugin/fu-context.c:1058
user@debian:~$ 
```
</details>

mightymos, 2025-01-11:
More output from i2cdetect:

```
user@debian:~$ sudo modprobe i2c_dev
user@debian:~$ sudo i2cdetect -l
i2c-0	smbus     	SMBus I801 adapter at efa0      	SMBus adapter
i2c-1	i2c       	i915 gmbus dpc                  	I2C adapter
i2c-2	i2c       	i915 gmbus dpb                  	I2C adapter
i2c-3	i2c       	i915 gmbus dpd                  	I2C adapter
i2c-4	i2c       	AUX B/DDI B/PHY B               	I2C adapter
i2c-5	i2c       	AUX C/DDI C/PHY C               	I2C adapter
i2c-6	i2c       	AUX D/DDI D/PHY D               	I2C adapter
```

hughsie, 2025-01-12:
> More output from i2cdetect

Ahh, so the PS175 isn't on your mainboard? Can you help me a bit with the logical and electrical topology?

mightymos, 2025-01-12:
Based on a program called ddccontrol (gddccontrol) it seems the two displayport outputs are at i2c-4 and i2c-5.
The HDMI output (i.e., displayport to hdmi) is then on i2c-6.

Only when I have an HDMI cable plugged in do I get this output with i2cdetect -y 6
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- 37 -- -- 3a -- -- -- -- -- 
40: 40 -- -- -- -- -- -- -- -- 49 -- -- -- -- -- -- 
50: 50 -- -- -- -- -- -- -- -- 59 -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --   
```

The Elitedesk 800 G3 has options boards for either outputting VGA, Displayport, or HDMI for one of the back panel output spaces. Two other displayports are permanently connected to the mainboard. A board to board connector is used to mount the option board to the mainboard with two screws. The option board for the HDMI contains a PS175 and programmable flash memory.

I can trace the board connector electrically if that helps, it would just help me if I knew what I was looking for.

I am not sure how a GUID is generated.
If it's just a matter of figuring out the relevant I2C address I can try to do that as well.

hughsie, 2025-01-13:
> If it's just a matter of figuring out the relevant I2C address

I guess you could try using i2c-6 in the quirk file, but more normally ACPI is involved to give it a well known name. Usually the i2c interface is actually different to the DDC one. Do you have any contact with HP for this hardware? I could ask Parade, but they're probably not going to be able to say anything without the approval of the ODM or OEM.

The other thing to try is to find a [working] firmware update program for the PS175 in the Elitedesk and then strace what it's doing, or the equivalent for a win32 binary.

mightymos, 2025-01-14:
I tried poking around at ACPI dumps from Linux but I either do not know what I am looking for or it was not present.
I have no professional contacts at HP lol.
I suppose it is possible I could attempt to contact customer service to see what they say.

The most promising result seems to be from this update program for an Intel NUC model:
https://www.intel.com/content/www/us/en/download/18730/26061/hdmi-firmware-update-tool-for-nuc6i7kyk.html

It does not "work" for updating but it gets partway through some checks even on my HP.

On the HP during execution I get a popup that says "The HDMI adapter has been detected. It is connected to an INTEL GPU."
That is accurate based on the known Intel HD Graphics 630.

With Ghidra I see strings like "PS175 FW Updater", "get PS175 demoboard!!!", "Cannot get ACK. \nUpdate failed", and "IICRead_INTEL successful\n".
There also appears to be an Intel HEX file present.

I am slowly parsing the decompilation but again it seems to be the most promising avenue.
I did try logging API calls using various tools but that seemed to be just Windows stuff.

Any other advice or opinions would be appreciated.

mightymos, 2025-01-14:
While we're at it, why doesn't Parade have any firmware files on LVFS?

mightymos, 2025-01-15:
Not sure how much more time I will spend on this.
It appears the drivers have been buggy with or without firmware updates:
https://gitlab.freedesktop.org/drm/i915/kernel/-/issues/203

I did email Parade support to request information about a firmware updater.

Attaching dumped and extracted firmware files in case they are of interest to someone else:
[ps175_dumped_extracted_and_found_firmware.zip](https://github.com/user-attachments/files/18419641/ps175_dumped_extracted_and_found_firmware.zip)

mightymos, 2025-01-17:
Okay, most useful output so far I guess.

```
admin@hpelitedesk[~]$ sudo i2cdetect 6
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-6.
I will probe address range 0x08-0x77.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
admin@hpelitedesk[~]$ sudo i2cdump 6 0x40
No size specified (using byte-data access)
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-6, address 0x40, mode byte
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 44 50 2d 48 44 4d 49 20 41 44 41 50 54 4f 52 04    DP-HDMI ADAPTOR?
10: a8 00 1c f8 50 53 31 37 35 30 b2 00 00 78 0f 00    ?.??PS1750?..x?.
20: 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00    ..?.............
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
40: 01 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ??..............
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
```

mightymos, 2025-01-18:
I happen to have another machine that has a newer Parade adapter inside:
```
mightymos@mightym715q:~$ sudo i2cdetect -l
i2c-0   smbus           SMBus PIIX4 adapter port 0 at 0b00      SMBus adapter
i2c-1   smbus           SMBus PIIX4 adapter port 2 at 0b00      SMBus adapter
i2c-2   smbus           SMBus PIIX4 adapter port 1 at 0b20      SMBus adapter
i2c-3   i2c             AMDGPU DM i2c hw bus 0                  I2C adapter
i2c-4   i2c             AMDGPU DM i2c hw bus 1                  I2C adapter
i2c-5   i2c             AMDGPU DM i2c hw bus 2                  I2C adapter
i2c-6   i2c             AMDGPU DM aux hw bus 0                  I2C adapter
i2c-7   i2c             AMDGPU DM aux hw bus 1                  I2C adapter
i2c-8   i2c             AMDGPU DM aux hw bus 2                  I2C adapter
mightymos@mightym715q:~$ sudo i2cdetect 5
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-5.
I will probe address range 0x08-0x77.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: 30 -- -- -- -- -- -- 37 -- -- -- -- -- -- -- --
40: 40 -- -- -- -- -- -- -- -- -- 4a 4b -- -- -- --
50: 50 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
mightymos@mightym715q:~$ sudo i2cdump 5 0x40
No size specified (using byte-data access)
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-5, address 0x40, mode byte
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 44 50 2d 48 44 4d 49 20 41 44 41 50 54 4f 52 04    DP-HDMI ADAPTOR?
10: a0 00 1c f8 50 53 38 34 30 32 a2 00 00 78 0f 00    ?.??PS8402?..x?.
20: 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00    ..?.............
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
mightymos@mightym715q:~$
```

I am wondering if the 1c f8 are part of an identifier because they are similar to I2C\NAME_1AF80175:00:00.
I see the same sequence in the binary files.

Apparently fwpdtool firmware-parse can extract version, VID, PID from binary files but that is apparently not available for parade_lspcon.

I cannot find anything like a name similar to to I2C\NAME under /sys but I do see the i2c and aux interfaces.

hughsie, 2025-01-18:
@Pingyao5115 is there anything you can help with here? Thanks.