fwupd/fwupd #8325: Update of the Lenovo Thunderbolt 4 Docking Station does't work with PCB version 4

Haewelmann, 2025-01-15:
**Describe the bug**
The update of the docking station with PCB version 4 ends with an error. Devices with a PCB version other than 4 must be disconnected from the USB cable during the update and reconnected. This is indicated by the two messages in the terminal:

"Please unplug the device USB cable."
... ~1min later ...
"Please re-insert the device USB cable."

These outputs are missing with PCB version 4. I noticed this when updating from version 10.13 to 10.16. Also from version 10.16 to 10.18.

**Steps to Reproduce**
start the update (10.13 -> 10.16):
fwupdmgr install 63a54fecbb2c02745c835c6b12c140dc98404bbdaa9abc498f709763771e567f-1016_NoIOT.cab
...
the update ends with this issue:
WARNING: This package has not been validated, it may not work properly.
Restarting device…       [******************                     ] 7 minutes remaining……
failed to wait for attach replug: device 691357701be9715ae7179d965f4fbfefdeb6e828 did not come back

**Expected behavior**
The update must finish without errors. It may be necessary to issue a message that the USB cable must be reconnected.

**fwupd version information**
fwupd version 1.9.27
linux kernel 6.8.12

hughsie, 2025-01-15:
Would it be possible to test with 2.0.3? If that works I can work out what needs backporting.

Haewelmann, 2025-01-15:
Yes, I can try that.

Haewelmann, 2025-01-22:
The two versions 2.0.3 and 2.0.4 can't detect the docking station (both v4 and non-v4). 
With version 2.0.2 at commit 94268e ("Improve the algorithm for detecting multipledevice-id matches") the docks are visible in fwupd (fwupdmgr get-devices). However, this also causes the problem that the update ends with an error and the message to reconnect the cable is missing.

Haewelmann, 2025-01-23:
I'm confused. It seems that versions 2.0.4 can sometimes detect my docking station. I can't explain why. Could it be that the function 'fu_udev_backend_netlink_setup' plays a role in this? But that is only a very vague assumption.