fwupd/fwupd #8327: uefi-dbx snapd unit tests are failing

superm1, 2025-01-15:
@superm1 is

	  File "/github/workspace/plugins/uefi-dbx/tests/snapd.py", line 70, in assert_prepare_req
	    with open(os.path.join(current_app.datadir, "dbx-update.auth"), "rb") as inf:
		 ~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
	FileNotFoundError: [Errno 2] No such file or directory: '/github/workspace/plugins/uefi-dbx/tests/dbx-update.auth'

...on your radar? I assume it's missing the fwupd-test-firmware blobs.

_Originally posted by @hughsie in https://github.com/fwupd/fwupd/issues/8323#issuecomment-2591866034_

superm1, 2025-01-15:
CC @bboozzoo

bboozzoo, 2025-01-15:
IF you want, feel free to assign it to me. I'm travelling today, but I'll try to have a look in the evening or tomorrow morning.

superm1, 2025-01-15:
No worries. If you're traveling I'll have a try and if I fails you can get to it then.

bboozzoo, 2025-01-15:
Found a short while to take a quick look and open a PR. I think it may be it, but let's see if the CI is happy now.

bboozzoo, 2025-01-15:
The tests are passing now.

Looking at the workflow steps though, I wonder if this part is correct:

https://github.com/fwupd/fwupd/blob/c4e6ceeb5218f4af257d2c6a1636f474da11a4cf/.github/workflows/matrix.yml#L46-L49

Shouldn't there be `-e CI=$CI -e CI_NETWORK=$CI_NETWORK` in that snippet too? This would change the behavior of get_test_firmware.sh too.

superm1, 2025-01-15:
> The tests are passing now.
>
> Looking at the workflow steps though, I wonder if this part is correct: [...]

Yes I think that's a good suggestion, can you propose that too?

closed 2025-01-15 by superm1

bboozzoo, 2025-01-15:
W dniu śr., 15 sty 2025 o 17:20 Mario Limonciello ***@***.***>
napisał(a):

> The tests are passing now.
>
> Looking at the workflow steps though, I wonder if this part is correct: [...]

Yeah. Will do.

bboozzoo, 2025-01-15:
opened https://github.com/fwupd/fwupd/pull/8333 please have a look 

Some tests failed with an error like below and I'm not sure the error makes any sense in the context of the code changes:

```
failed to download https://fwupd.org/downloads/779eda568d6146da1c02ee36d65e3b0dcc9b0094b924ccc976ab90b01967ed51-emulation.zip: failed to download file: HTTP/2 stream 1 was not closed cleanly: INTERNAL_ERROR (err 2)
FAIL: fwupd/fwupd.test (Child process exited with code 1)
```

PR #8329 (2025-01-15): trivial: ci: pass a path to uefi-dbx test artifacts
Pass a path to uefi-dbx test artifacts when running mock snapd API service.

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation

Maybe fixes: #8327