fwupd/fwupd #8331: LegionHid2 fails to reload, but only on 1_9_X

superm1, 2025-01-15:
When running an update on legion-hid2, I see that with 1_9_X (1.9.26 specifically) that the update fails with:

```sh
failed to reload device: failed to claim HID interface: USB error on device 1a86:e310 : No such device (it may have been disconnected) [-4]
```

But trying the exact same thing on main (2.0.3 release specifically) it works.  Any ideas on what's missing for a backport?

superm1, 2025-01-15:
1.9.26 has this:
```
22:17:31.254 FuHidDevice          ignoring: USB error on device 1a86:e310 : Entity not found [-5]
```

Here's the full update logs:

[fwupd_1_9_26.txt](https://github.com/user-attachments/files/18427509/fwupd_1_9_26.txt)
[fwupd_2_0_3.txt](https://github.com/user-attachments/files/18427512/fwupd_2_0_3.txt)

hughsie, 2025-01-15:
I'm not sure - but I'd certainly see the GError domain and code is what we expect to ignore. 1.9.x was a bit fast-and-loose about what error domains were being returned.

superm1, 2025-01-15:
We were using gusb in 1.9.x, so the error codes are definitely different.

We're missing 245eadb6c77863e89540825a6a884fc3e150d980 but that was fallout from the move away from gusb (2c6c4dc9).

superm1, 2025-01-15:
Maybe the wrong device is being picked to pass to reload; IE it's due to missing 94268e76fb0c2a43bb3364235aa820b8121312e0

superm1, 2025-01-15:
Nope; that's not it.  I did a backport of 7cc3b202e42b48050ff4559d03ead826b36dbb82 and 94268e76fb0c2a43bb3364235aa820b8121312e0 but can still repro

superm1, 2025-01-16:
As the hardware with Legion Hid2 isn't even shipping yet I decided it's not worth identifying the patches for fwupd 1.9.x in this case.  I'm going to close as won't fix.

closed 2025-01-16 by superm1