fwupd/fwupd #8347: fwupd 1.9.27 daemon cannot open the device correctly (re-connected to the host system)

CragW, 2025-01-18:
**Describe the bug**
When a device is disconnected and then reconnected to the host system, the daemon cannot open the device correctly to perform real communication.

Simply disconnecting and reconnecting the device does not resolve the issue. Restarting the daemon service is required to recover, for example:

```
$ systemctl restart fwupd.service
```

Hereunder is a small slice from  [fwupd-1.9.27-kestrel-update-device-not-opened.txt](https://github.com/user-attachments/files/18462855/fwupd-1.9.27-kestrel-update-device-not-opened.txt) which comes from the daemon runtime log.

```
Jan 17 23:03:33 fwupd[23602]: 15:03:33.098 FuEngine             install phase now composite-prepare
Jan 17 23:03:33 fwupd[23602]: 15:03:33.098 FuPlugin             composite_prepare(dell_dock)
Jan 17 23:03:33 fwupd[23602]: 15:03:33.098 FuPlugin             composite_prepare(dell_kestrel)
Jan 17 23:03:33 fwupd[23602]: 15:03:33.098 FuDeviceLocker       ignoring close error on aborted open: Device 413c:b06e has not been opened
Jan 17 23:03:33 fwupd[23602]: 15:03:33.098 FuDeviceLocker       ignoring close error on aborted open: failed to release HID interface: Device 413c:b06e has not been opened
Jan 17 23:03:33 fwupd[23602]: 15:03:33.098 FuIdle               uninhibiting: timeout,signals by update
Jan 17 23:03:33 fwupd[23602]: 15:03:33.098 FuIdle               now inhibited: timeout
Jan 17 23:03:33 fwupd[23602]: 15:03:33.098 FuEngine             clearing device-changed allowlist as transaction done
Jan 17 23:03:33 fwupd[23602]: 15:03:33.098 FuMain               Emitting PropertyChanged('Status'='idle')
Jan 17 23:03:33 fwupd[23602]: 15:03:33.099 FuClientList         client :1.343 removed
Jan 17 23:03:33 fwupd[23602]: 15:03:33.099 FuMain               connected clients: 0
Jan 17 23:03:33 fwupd[23602]: 15:03:33.099 FuIdle               uninhibiting: timeout by connected-clients
```

Meanwhile, below error messages are seen on the client side:

```
$ fwupdmgr install ./foo.cab --allow-reinstall
Authenticating...    [
failed to prepare composite action: failed to composite_prepare using dell_kestrel: device recovery not possible: failed to open device: USB error on device 413c:b06e : No such device (it may have been disconnected) [-4]
```

Issue is not reproducible from the source in the main branch.

**Steps to Reproduce**
$ fwupdmgr install foo.cab --allow-reinstall

**Expected behavior**
Updated firmware successfully.

**fwupd version information**

```shell
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.27
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.27
runtime   org.kernel                    6.8.0-51-generic

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): apt

**Additional questions**

- Operating system and version: Ubuntu 24.04
- Have you tried rebooting? Yes.
- Is this a regression? Yes.

hughsie, 2025-01-18:
Dude, it's your plugin -- you're the assignee :)

superm1, 2025-01-20:
https://github.com/fwupd/fwupd/pull/8356

PR #8356 (2025-01-20): dell-kestrel: cleanup the devices when disconnected
Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation

Fixes #8347
closed 2025-01-20 by superm1