fwupd/fwupd #8357: "20230501 # formally version 371 is not a valid number"

grawity, 2025-01-20:
**Describe the bug**

fwupd 2.0.4 doesn't offer any dbx updates beyond version 371, although version 20241101 seems to be published. The daemon logs `20230501 # formally version 371 is not a valid number`.

```
$ fwupdmgr get-updates
Firmware metadata has not been updated for 30 days and may not be up to date.
Update now? (Requires internet connection) [y|N]: y
Updating lvfs
Downloading…             [************************************** ]
Successfully downloaded new metadata: Updates have been published for 8 of 11 local devices
Successfully uploaded report
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • Prometheus
 • Prometheus IOTA Config
 • System Firmware
 • Thunderbolt host controller
 • Unifying Receiver
 • WDC PC SN730 SDBQNTY-512G-1001
No updates available
(returned 2)
```

```
systemd[1]: Starting Firmware update daemon...
fwupd[1679395]: 12:46:52.237 FuDevice             20230501 # formally version 371 is not a valid number (guessed plain)
fwupd[1679395]: 12:46:52.237 FuDevice             20230501 # formally version 371 is not a valid number (guessed plain)
fwupd[1679395]: 12:46:56.398 FuMain               Daemon ready for requests (locale en_US.UTF-8)
```

**Steps to Reproduce**

1. Upgrade to fwupd 2.0.4, restart fwupd.service if already running.
2. Run `fwupdmgr get-updates`.

**Expected behavior**

It should offer the 2024 or 2025 dbx update.

**fwupd version information**

Installed via pacman on Arch Linux:

```shell
$ pacman -Q fwupd
fwupd 2.0.4-1

$ fwupdmgr --version
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.4
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.2
runtime   org.freedesktop.fwupd-efi     1.7
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.2
runtime   org.kernel                    6.12.6-arch1-1
runtime   org.freedesktop.fwupd         2.0.4
```

<details>

<summary>**fwupd device information**</summary>
[... 314 lines cut ...]
│   ├── entries/
│   │   ├── arch.conf
│   │   └── arch-lts.conf
│   ├── entries.srel
│   ├── loader.conf
│   └── random-seed
├── initramfs-linux.img
├── initramfs-linux-lts.img
├── intel-ucode.img
├── Lenovo.zip
├── vmlinuz-linux
└── vmlinuz-linux-lts

6 directories, 13 files
```
</details>

**Additional questions**

- Operating system and version: [Arch Linux]
- Have you tried rebooting? [no]
- Is this a regression? [not sure]
- Are you using an NVMe disk? [yes]
- Is secure boot enabled? [no]
- Is this a Lenovo system [yes] with 'Boot Order Lock' turned on in the BIOS? [no]

hughsie, 2025-01-20:
OOpa. Can you try cherry picking the patch in https://github.com/fwupd/fwupd/pull/8358 please.

grawity, 2025-01-20:
> OOpa. Can you try cherry picking the patch in [#8358](https://github.com/fwupd/fwupd/pull/8358) please.

Just tested with the patch; it no longer complains (but still doesn't offer any dbx updates).

closed 2025-01-20 by superm1

hughsie, 2025-01-20:
>

hughsie, 2025-01-20:
> but still doesn't offer any dbx updates

Can you try `fwupdmgr refresh --force` please. If that doesn't work then please grab me the output of `sudo fwupdtool get-devices --plugins uefi_dbx -vv` please.

grawity, 2025-01-21:
> Can you try `fwupdmgr refresh --force` please. If that doesn't work

I tried this yesterday and it did not, but today it worked – I got an update to 20241101.

(Seems to have been strictly incremental though, i.e. dbx only [grew](https://nullroute.lt/tmp/txt/20250121.073315.OaeJkN.txt) from 20kB to 22kB according to `efi-readvar`, despite the blog post saying "more hashes were removed than added", but I suppose that's fine – I don't really understand how those updates are applied, and `df /sys/firmware/efi/efivars` says I'm at 27% usage so that's probably fine.)