fwupd/fwupd #8368: Dell WD19 upgrade issue

wojciechczyz, 2025-01-21:
**Describe the bug**
update is found and downloaded but cannot perform

**Steps to Reproduce**
sudo fwupdtool get-devices
sudo fwupdtool refresh
sudo fwupdtool get-updates
sudo fwupdtool update

**Expected behavior**
Expected correct upgrade to conclude

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
Fails
fwupdtool --version
Failed to lock: failed to open /Users/xxx/.cache/fwupd/fwupdtool
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

brew on MacOS

<summary>**fwupd device information**</summary>

Loading?                 [************************************** ]
Apple Mac14,5
?
??WD22TB4:
  ?   Device ID:          b3db560e794a8ceb981cf5118b5818ac75d0eafd
  ?   Summary:            High performance dock
  ?   Current version:    01.01.00.07
  ?   Vendor:             Dell Inc. (USB:0x413C)
  ?   Install Duration:   1 minute
  ?   Serial Number:      BYFXYT3/3135888993113304
  ?   GUID:               cd357cf1-40b2-5d87-b8df-bb2dd82774aa ? USB\VID_413C&PID_B06E&hub&embedded
  ?   Device Flags:       ? Updatable
  ?                       ? Supported on remote server
  ?                       ? Device stages updates
  ?                       ? Device can recover flash failures
  ?                       ? Device is usable for the duration of the update
  ?                       ? Signed Payload
  ?
  ??RTS5413 in Dell dock:
  ?     Device ID:        0bcab2ad4789ecbcca1bf84409a79681c8c700aa
  ?     Summary:          USB 3.1 Generation 1 Hub
  ?     Current version:  01.22
  ?     Vendor:           Dell Inc. (USB:0x413C)
  ?     Install Duration: 14 seconds
  ?     GUIDs:            b27d25f1-019d-5718-b41a-02ddaefe5577 ? USB\VID_413C&PID_B06F
  ?                       ac5b774c-b49d-566b-9255-85f0f7f8a4ed ? USB\VID_413C&PID_B06F&hub
  ?     Device Flags:     ? Updatable
  ?                       ? Supported on remote server
  ?                       ? Device stages updates
  ?                       ? Device is usable for the duration of the update
  ?                       ? Signed Payload
  ?                       ? Can tag for emulation
  ?
  ??USB4 controller in Dell dock:
        Device ID:        71413822e87deae707da948bd4b7e537af463c19
        Summary:          USB4 controller
        Current version:  41.81
        Vendor:           Dell Inc. (USB:0x8087, TBT:0x00D4)
        Install Duration: 46 seconds
        GUIDs:            b3f8b518-ebdd-5969-9b44-6b096eff5b9b ? USB\VID_8087&PID_0B40
                          c6e7a14f-f13c-55e5-99de-d59461db1d27 ? USB\VID_8087&PID_0B40&HUB_0000
        Device Flags:     ? Updatable
[... 272 lines cut ...]

**Peripherals connected to the dock**
Please describe all devices connected to the dock.  Be as specific as possible,
including USB devices, hubs, monitors, and downstream type-C devices.

**Verbose daemon logs**
First enable daemon verbose logs collection.

```shell
fwupdmgr modify-config "VerboseDomains" "*"
```

Then try to reproduce the issue.  Even if it doesn't reproduce, please attach the
daemon verbose logs collected from the system journal.

```shell
journalctl -b -u fwupd.service
```

**Additional questions**

- Operating system and version:
- Have you tried unplugging the dock or any peripherals from your machine?
- Have you tried to power cycle the dock from the AC adapter?
- Is this a regression?

ciarancoffey, 2025-01-28:
I picked up one of these docs for cheap. I went to upgrade the firmware and googled the error and arrived here.

I'm on linux so I'll add some more logs

```$ fwupdmgr --version 
compile   info.libusb                   1.0.27
compile   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.2
compile   org.freedesktop.fwupd         2.0.4
runtime   org.freedesktop.fwupd-efi     1.7
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.2
runtime   org.kernel                    6.12.10-arch1-1
runtime   org.freedesktop.fwupd         2.0.4
```

`journalctl -b -u fwupd.service  > wd19.txt`
```
Jan 28 14:35:05 HOSTNAME systemd[1]: fwupd.service: Deactivated successfully.
Jan 28 14:35:05 HOSTNAME systemd[1]: fwupd.service: Consumed 5.382s CPU time, 194.7M memory peak.
Jan 28 14:35:05 HOSTNAME systemd[1]: Starting Firmware update daemon...
Jan 28 14:35:05 HOSTNAME fwupd[161085]: 14:35:05.637 FuDevice             20230501 # formally version 371 is not a valid number (guessed plain)
Jan 28 14:35:05 HOSTNAME fwupd[161085]: 14:35:05.637 FuDevice             20230501 # formally version 371 is not a valid number (guessed plain)
Jan 28 14:35:09 HOSTNAME fwupd[161085]: 14:35:09.265 FuPluginDfu          failed to parse interface data: no image idx 33 found in firmware
Jan 28 14:35:09 HOSTNAME fwupd[161085]: 14:35:09.265 FuPluginDfu          DFU version 0x0000 invalid, v1.1 assumed
Jan 28 14:35:09 HOSTNAME fwupd[161085]: 14:35:09.420 FuMain               Daemon ready for requests (locale en_IE.UTF-8)
Jan 28 14:35:09 HOSTNAME systemd[1]: Started Firmware update daemon.
Jan 28 14:35:16 HOSTNAME fwupd[161085]: 14:35:16.752 FuEngine             failed to cleanup failed composite action: failed to composite_cleanup using dell_dock: failed to subclass open: no proxy
Jan 28 14:36:22 HOSTNAME fwupd[161085]: 14:36:22.728 FuEngine             failed to cleanup failed composite action: failed to composite_cleanup using dell_dock: failed to subclass open: no proxy```

david-barbion, 2025-01-28:
Same here on that device:
```
├─WD19TBS:
│ │   Device ID:          4940a095e87a030ab4507472901d8fe8cc7b9785
│ │   Summary:            High performance dock
│ │   Current version:    01.01.00.07
│ │   Vendor:             Dell Inc. (USB:0x413C)
│ │   Install Duration:   1 minute
│ │   Serial Number:      GFQL0L3/3134063727574170
│ │   Update State:       Failed
│ │   Update Error:       failed to open device for prepare: failed to subclass open: no proxy
│ │   Last modified:      2025-01-28
│ │   GUID:               cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
```

olistr, 2025-01-31:
same issue here for a dw22 on macbook pro 16

jkuma, 2025-02-08:
same issue here for a dw22 on Dell Inc. XPS 13 9300

divico, 2025-02-13:
same issue trying to upgrade ` WD19TB from 01.01.00.07 to 01.01.00.10`

obbardc, 2025-02-14:
Also reproduces here, is the root cause known/under investigation?

hughsie, 2025-02-15:
> is the root cause known/under investigation

I don't have the hardware, @CragW do you have a spare WD19TB you can send me?

CragW, 2025-02-16:
> I don't have the hardware, [@CragW](https://github.com/CragW) do you have a spare WD19TB you can send me?

Some devices are missing after 940a3b353 , there instance id for `mst` isn't really added to the device hence it was failed to be added to the plugin and failing all the rest components.

commit 28857984 (2025-02-20): dell-dock: bugfix for mst instance id

Fixes #8368
commit c6322bf5 (2025-02-20): dell-dock: bugfix for mst instance id

Fixes #8368
commit 4f8e123a (2025-02-20): dell-dock: bugfix for mst instance id

Fixes #8368
PR #8481 (2025-02-20): dell-dock: bugfix for mst instance id
Fixes #8368

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
closed 2025-02-20 by superm1