fwupd/fwupd #8369: Error writing to file descriptor: Invalid argument for DBX update on openQA

hughsie, 2025-01-21:
Copied from https://bugzilla.redhat.com/show_bug.cgi?id=2339077:

> Since fwupd-2.0.4-1.fc42 landed in Rawhide earlier today, all openQA GNOME and KDE graphical update tests have failed. Those are tests where we make sure at least one updated package is available, then run through an update normally in the graphical tool (Software on GNOME, Discover on KDE).
> 
> It seems like with this updated fwupd, the tests are seeing a UEFI DBX update, which I don't think has ever been the case in openQA testing before. It looks like they try and install this update, and it fails. That causes the apps not to follow their usual flow that openQA is expecting, but unexpectedly show some kind of error message. GNOME says "Sorry, something went wrong". KDE says "There was an issue during the update or installation process. Please try again later."
> 
> GNOME doesn't apparently log anything about the issue (it has a button to press for more details, but openQA doesn't click it), but KDE does log what seems to be the problem:
> 
> Jan 20 11:04:10 fedora plasma-discover[2481]: Fwupd Error 18 failed to write-firmware: failed to write data to efivarsfs: Error writing to file descriptor: Invalid argument
> 
> This is how the openQA VM's UEFI stuff is set up:
> 
> [2025-01-20T22:29:24.079048Z] [debug] [pid:3159702] running `/usr/bin/qemu-img create -f qcow2 -F raw -b /usr/share/edk2/ovmf/OVMF_CODE.secboot.fd /var/lib/openqa/pool/20/raid/pflash-code-overlay0 1966080`
> [2025-01-20T22:29:24.181895Z] [debug] [pid:3159702] Formatting '/var/lib/openqa/pool/20/raid/pflash-code-overlay0', fmt=qcow2 cluster_size=65536 extended_l2=off compression_type=zlib size=1966080 backing_file=/usr/share/edk2/ovmf/OVMF_CODE.secboot.fd backing_fmt=raw lazy_refcounts=off refcount_bits=16
> [2025-01-20T22:29:24.182193Z] [debug] [pid:3159702] running `/usr/bin/qemu-img create -f qcow2 -F raw -b /usr/share/edk2/ovmf/OVMF_VARS.secboot.fd /var/lib/openqa/pool/20/raid/pflash-vars-overlay0 131072`
> [2025-01-20T22:29:24.308512Z] [debug] [pid:3159702] Formatting '/var/lib/openqa/pool/20/raid/pflash-vars-overlay0', fmt=qcow2 cluster_size=65536 extended_l2=off compression_type=zlib size=131072 backing_file=/usr/share/edk2/ovmf/OVMF_VARS.secboot.fd backing_fmt=raw lazy_refcounts=off refcount_bits=16
> ...
> [2025-01-20T22:29:24.311497Z] [debug] [pid:3159702] starting: /usr/bin/qemu-system-x86_64 -device virtio-vga,edid=on,xres=1024,yres=768 -only-migratable -chardev ringbuf,id=serial0,logfile=serial0,logappend=on -serial chardev:serial0 -audiodev none,id=snd0 -device intel-hda -device hda-output,audiodev=snd0 -global isa-fdc.fdtypeA=none -m 3072 -machine q35,smm=on -cpu host -netdev user,id=qanet0,net=172.16.2.0/24 -device virtio-net,netdev=qanet0,mac=52:54:00:12:34:56 -object rng-random,filename=/dev/urandom,id=rng0 -device virtio-rng-pci,rng=rng0 -device qemu-xhci -device usb-tablet -smp 2 -enable-kvm -no-shutdown -vnc :110,share=force-shared -device virtio-serial -chardev pipe,id=virtio_console,path=virtio_console,logfile=virtio_console.log,logappend=on -device virtconsole,chardev=virtio_console,name=org.openqa.console.virtio_console -chardev pipe,id=virtio_console_user,path=virtio_console_user,logfile=virtio_console_user.log,logappend=on -device virtconsole,chardev=virtio_console_user,name=org.openqa.console.virtio_console_user -chardev socket,path=qmp_socket,server=on,wait=off,id=qmp_socket,logfile=qmp_socket.log,logappend=on -qmp chardev:qmp_socket -S -global driver=cfi.pflash01,property=secure,value=on -device virtio-scsi-pci,id=scsi0 -blockdev driver=file,node-name=hd0-overlay0-file,filename=/var/lib/openqa/pool/20/raid/hd0-overlay0,cache.no-flush=on -blockdev driver=qcow2,node-name=hd0-overlay0,file=hd0-overlay0-file,cache.no-flush=on,discard=unmap -device virtio-blk,id=hd0-device,drive=hd0-overlay0,bootindex=0,serial=hd0 -blockdev driver=file,node-name=hd1-file,filename=/var/lib/openqa/pool/20/raid/hd1,cache.no-flush=on -blockdev driver=qcow2,node-name=hd1,file=hd1-file,cache.no-flush=on,discard=unmap -device virtio-blk,id=hd1-device,drive=hd1,serial=hd1 -drive id=pflash-code-overlay0,if=pflash,file=/var/lib/openqa/pool/20/raid/pflash-code-overlay0,unit=0,readonly=on -drive id=pflash-vars-overlay0,if=pflash,file=/var/lib/openqa/pool/20/raid/pflash-vars-overlay0,unit=1
> 
> That is, openQA creates overlays on the OVMF_CODE.secboot.fd and OVMF_VARS.secboot.fd pflash and vars images from edk2, and attaches those to the VM. It sets readonly=on for the pflash overlay, but not for the vars overlay.
> 
> I don't know if the issue is that the fwupd shouldn't show the update for the VM, or the update should actually *work*, or it should fail but the failure should be handled...but the current state is not workable.

I think the issue isn't the new fwpud per-se, more that this is the first version of fwupd that can apply the new style of dbx update and I don't know why, but it isn't applying on the qemu machine. I kinda suspect we've never tried before (the version in /usr/share/edk2/ovmf/OVMF_VARS.secboot.fd is 371, i.e. what used to be the latest version) and it's always going to have failed with this SB setup.

My initial theory was that pflash-vars-overlay0 wasn't big enough

* pflash-vars-overlay0 created: 131072 (128kB)
* EfivarsNvramUsed according to fwupd: 33490 (32kb)
* DBX file: 15462 (16kb)

I created the much smaller VARS, although it's still enough to apply it in theory:

    sudo qemu-img create -f qcow2 -F raw -b /usr/share/edk2/ovmf/OVMF_VARS.secboot.fd /var/lib/libvirt/qemu/nvram/fedora-rawhide_VARS.qcow2 131072`
    virsh -c qemu:///system start fedora-rawhide --reset-nvram

That worked; it's not the VARS store *size*. It's also failing with `Error writing to file descriptor: Invalid argument` which kinda suggests it's not out of space either. @superm1 I'm out of ideas, anything that pops out at you?

hughsie, 2025-01-21:
Some background... `Invalid argument` is actually `EINVAL`, which is actually `EFI_INVALID_PARAMETER` which looking at AuthService.c in edk2 I'm unsure how we're failing given the non openQA image works fine.

AdamWill, 2025-01-21:
So it looks like this works if I have openQA use the "4M" versions of the code and vars images. From googling around, the only difference is indeed that those provide more space for EFI variables. It's still a bit odd that @hughsie couldn't reproduce this with the non-4M files, but oh well, that's probably good enough for now. I'll adjust our openQA deployments to use the 4M files permanently and things should be OK.

closed 2025-01-21 by hughsie