fwupd/fwupd #8370: Does fwupd check DB before DBX update? Is there a warning for revoked files present before DBX is applied?

sei-vsarvepalli, 2025-01-21:
**Describe the question**
In the light of latest Microsoft updates on Jan 14th 2025, it looks like the revocation of `Microsoft Windows Production PCA (Product Certificate Authority) 2011` certificate is included in the update. This means any bootloader or driver signed under that certificate will no longer work. Can fwupd tool verify that there is no revoked binary or firmware present before applying DBX? When I test this on a Linux install with a vulnerable driver store din EFI filesystem, I did not get a warning. May be my version is too old? 

I can see I was Abel to update the DBX to bock the latest Howyar vulnerability CVE-2024-7344 or VU#529659 

```
dbxtool -l | grep -i CDB7C90D3AB8833D5324F5D8516D41FA990B9CA721FE643FFFAEF9057D9F9E48
 463: {microsoft} {sha256} cdb7c90d3ab8833d5324f5d8516d41fa990b9ca721fe643fffaef9057d9f9e48
```

**fwupd version information**

```shell
fwupdmgr --version
```
```
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.0
compile   org.freedesktop.fwupd         1.9.27
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.8.0-51-generic
runtime   org.freedesktop.fwupd         1.9.27
```

hughsie, 2025-01-21:
> Can fwupd tool verify that there is no revoked binary or firmware present before applying DBX

Yup, we do that now if you use fwupdmgr or fwupdtool to deploy the dbx update. We've had that for some time. If you're using either the original dbxtool (or the fwupd-provided backwards compatibility version) you can do anything you like with no checks. e.g. google for "Blocked executable in the ESP, ensure grub and shim are up to date" to see all the real people that hit this. :)

sei-vsarvepalli, 2025-01-21:
Thanks! I think I did the `fwupdtool install-blob DBXUpdate-20250114.x64.bin` which bypasses these checks.  It is impossible to know the PXE, HTTP and any other Network Media is up to date, perhaps a warning is best both in the XML and anybody who cares to read knows that these may be impacted.