fwupd/fwupd #8382: Updating Quectel 5G module RM520N-GL fails

iyanmv, 2025-01-23:
**Describe the bug**
Trying to update 5G module Quectel RM520N-GL on a Lenovo Thinkpad X1 Carbon Gen 12 fails.

Before today [I was not seeing any updates](https://github.com/fwupd/firmware-lenovo/issues/491) at all for this module.

**Steps to Reproduce**
1. Run `fwupdmgr update`. The following available update is shown:
```
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade RM520N-GL from RM520NGLAAR03A03M4G_04.204.04.204 to                  ║
║ RM520NGLAAR03A03M4G_04.211.04.211?                                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Firmware release RM520NGLAAR03A03M4G_04.211.04.211 packaged for LVFS.        ║
║                                                                              ║
║ • Performance improvements                                                   ║
║ • Stability improvements                                                     ║
║                                                                              ║
║ RM520N-GL and all connected devices may not be usable while updating.        ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]:
```
2. Press enter
3. After a few seconds, update fails with this error message: `failed to write-firmware: no firehose port provided for filename`

**Expected behavior**
Update should complete without errors

**fwupd version information**
```
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.4
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.2
runtime   org.freedesktop.fwupd-efi     1.7
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.2
runtime   org.kernel                    6.13.0-arch1-1
runtime   org.freedesktop.fwupd         2.0.4
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `pacman` ([2.0.4-1](https://archlinux.org/packages/extra/x86_64/fwupd/))

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```
LENOVO 21KDS00600
│
├─WD22TB4:
│ │   Device ID:          d79875839126c155c7a0e57d34c2fb27db428b7e
│ │   Summary:            High performance dock
│ │   Current version:    01.01.00.10
│ │   Vendor:             Dell Inc. (USB:0x413C)
│ │   Install Duration:   1 minute
│ │   Serial Number:      CZCVK24/3143864680307620
│ │   GUID:               cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │ 
│ └─USB4 controller in Dell dock:
│       Device ID:        6a057b01eafb9e4c547cbaba877ad802b568b6fc
[... 421 lines cut ...]
│     Device Flags:       • Updatable
│                         • Signed Payload
│                         • Can tag for emulation
│   
└─USB4 Retimer:
      Device ID:          16d35c510618116e7899ae45e62605572872e713
      Summary:            A physical layer protocol-aware, software-transparent extension device that forms two separate electrical link segments
      Current version:    23.81
      Vendor:             THUNDERBOLT:0x8087
      GUIDs:              74fb1b87-53e6-5115-bd02-a9fefd06ed9a ← THUNDERBOLT\VEN_8087&DEV_0D9C
                          dac6f95d-7c5b-52b5-8c79-a0436a5ef784 ← TBT-80870d9c-retimer0-0:1.1
      Device Flags:       • Internal device
                          • Updatable
                          • Device stages updates
                          • Unsigned Payload
                          • Can tag for emulation
```

</details>

**Additional questions**

- Operating system and version: Arch Linux
- Have you tried rebooting? Yes, several times. Update always fails in the same way.
- Is this a regression? I don't know. This is the first time I'm trying to update this module. Until today I never received any updates.

lvoegl, 2025-01-31:
The Firehose port does not seem to be assigned anywhere in version 2 for PCI based modems.
This has been reworked for version 2+. I've implemented the port finding for USB modems (which is the hardware I have) in [fu-mm-device.c](https://github.com/fwupd/fwupd/blob/main/plugins/modem-manager/fu-mm-device.c#L1042), but for PCI modems someone would have to implement something similar. The rest seems to work fine.
I think version 1.9.x should work out of the box in your case.

iyanmv, 2025-01-31:
@lvoegl I can confirm that update works with 1.9.x. In particular, I used this:

```
compile   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.2
compile   org.freedesktop.fwupd         1.9.28
runtime   org.freedesktop.fwupd-efi     1.7
compile   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libxmlb           0.3.21
runtime   org.freedesktop.Passim        0.1.8
runtime   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libjcat           0.2.2
runtime   org.kernel                    6.13.0-arch1-2
runtime   org.freedesktop.fwupd         1.9.2
```

MankWang, 2025-02-11:
@lvoegl 
The modification still has problems, because the update flow is the following function, and the `self->port_edl` is still NULL.

[image]

I have added `fu_mm_device_ensure_edl_port` under `fu_mm_device_qcdm_switch_to_edl` , but it still doesn't work.

lvoegl, 2025-02-14:
> I have added `fu_mm_device_ensure_edl_port` under `fu_mm_device_qcdm_switch_to_edl` , but it still doesn't work.

Is the error still `no firehose port provided for filename`?
Can you please provide the `-vv` debug logs?

MankWang, 2025-02-17:
My modification is as follows:

[image]

Here is the log:

```
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade RM520N-GL from RM520NGLAAR03A03M4G_04.211.04.211 to                  ║
║ RM520NGLAAR03A03M4G_04.212.04.212?                                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Firmware release RM520NGLAAR03A03M4G_04.212.04.212 packaged for LVFS.        ║
║                                                                              ║
║ • Performance improvements                                                   ║
║ • Stability improvements                                                     ║
║                                                                              ║
║ RM520N-GL and all connected devices may not be usable while updating.        ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Writing…                 [****                                   ] Less than one minute remaining…
Successfully installed firmware
failed to write-firmware: no firehose port provided for filename
```

```
ls /dev/ | grep firehose
wwan0firehose0
```

```
fwupd -vv

...
02:20:24.461 FuStruct             FuStructCabFile:
  usize: 0x35e
  uoffset: 0x0
  index: 0x0
  date: 0x5a45
  time: 0xdd0
  fattr: 0x0 [none]
02:20:24.461 FuStruct             FuStructCabFile:
  usize: 0xdff
  uoffset: 0x35e
  index: 0x0
  date: 0x5a45
  time: 0xdd0
  fattr: 0x0 [none]
02:20:24.461 FuStruct             FuStructCabFile:
  usize: 0x1235
  uoffset: 0x115d
  index: 0x0
  date: 0x5a45
  time: 0xdd0
  fattr: 0x0 [none]
02:20:24.461 FuStruct             FuStructCabFile:
  usize: 0x78c9b83
  uoffset: 0x2392
  index: 0x0
  date: 0x5a38
  time: 0x30d5
  fattr: 0x20 [arch]
02:20:25.357 Jcat                 Jcat engine kind 'sha1' not allowed
02:20:25.357 Jcat                 Jcat engine kind 'sha1' not allowed
02:20:25.357 Jcat                 got 1 PKCS7 signatures
02:20:25.358 Jcat                 verified: JcatResult:
  Timestamp:             2025-02-05T01:46:33Z
  Authority:             O=Linux Vendor Firmware Project,CN=LVFS CA
  JcatPkcs7Engine:
    Kind:                pkcs7
    VerifyKind:          signature
[... 596 lines cut ...]
02:24:27.271 FuEngine             4117c99d-0660-5db6-bbfe-65ed036debe5 matched 1 releases
02:24:27.271 FuEngine             bb3a7757-29b3-5950-9dba-71cc18c6fb86 was not found: indexed string 'bb3a7757-29b3-5950-9dba-71cc18c6fb86' was unfound
02:24:27.271 FuEngine             5971a208-da00-5fce-b5f5-1234342f9cf7 was not found: indexed string '5971a208-da00-5fce-b5f5-1234342f9cf7' was unfound
02:24:27.281 FuEngine             f8ba2887-9411-5c36-9cee-88995bb39731 matched 2 components
02:24:27.281 FuRelease            remote lvfs has kind=download and so marking as trusted
02:24:27.281 FuEngine             requirement 1.9.28 ge 2.0.6 -> org.freedesktop.fwupd passed
02:24:27.281 FuEngine             add trusted-report to com.microsoft.dbx.x64.firmware:20241101 as trusted: FwupdReport:
  Flags:                from-oem

02:24:27.281 FuRelease            remote lvfs has kind=download and so marking as trusted
02:24:27.282 FuEngine             failed to set release for component: Device UEFI dbx does not support org.uefi.dbx, only org.uefi.dbx2
02:24:27.282 FuRelease            remote lvfs has kind=download and so marking as trusted
02:24:27.282 FuEngine             failed to set release for component: Device UEFI dbx does not support org.uefi.dbx, only org.uefi.dbx2
02:24:27.282 FuRelease            remote lvfs has kind=download and so marking as trusted
02:24:27.282 FuEngine             failed to set release for component: Device UEFI dbx does not support org.uefi.dbx, only org.uefi.dbx2
02:24:27.282 FuRelease            remote lvfs has kind=download and so marking as trusted
02:24:27.283 FuEngine             failed to set release for component: Device UEFI dbx does not support org.uefi.dbx, only org.uefi.dbx2
02:24:27.283 FuRelease            remote lvfs has kind=download and so marking as trusted
02:24:27.283 FuEngine             failed to set release for component: Device UEFI dbx does not support org.uefi.dbx, only org.uefi.dbx2
02:24:27.283 FuEngine             f8ba2887-9411-5c36-9cee-88995bb39731 matched 1 releases
02:24:27.283 FuEngine             fccbb6ea-e20e-58ad-bf8a-7fb7d43ff4c2 was not found: indexed string 'fccbb6ea-e20e-58ad-bf8a-7fb7d43ff4c2' was unfound
02:24:27.283 FuEngine             12c86995-0b90-5ec5-98f3-7a6ed4ca50e0 was not found: indexed string '12c86995-0b90-5ec5-98f3-7a6ed4ca50e0' was unfound
02:24:27.283 FuEngine             06b4e2aa-91af-508b-b06e-65e3b3189e97 was not found: indexed string '06b4e2aa-91af-508b-b06e-65e3b3189e97' was unfound
02:24:27.283 FuEngine             writing motd target /home/test/tools/fwupd/venv/bin/../dist/cache/fwupd/motd.d/85-fwupd
```

lvoegl, 2025-02-21:
Interesting...
```
devices = g_udev_client_query_by_subsystem(client, subsystem);
for (GList *l = devices; l != NULL; l = g_list_next(l)) {
	if (g_str_has_prefix(g_udev_device_get_sysfs_path(G_UDEV_DEVICE(l->data)),
			     device_sysfs_path)) {
		device_file = g_strdup(g_udev_device_get_device_file(l->data));
		if (device_file != NULL)
			break;
	}
}
```
since this was used to find the `device_file` before version 2.0.0, I thought `fu_udev_device_get_device_file` gives the device file, because there is a udev device for the modem (also PCI) in EDL mode. But apparently `g_udev_device_get_device_file` (or `udev_device_get_devnode`) does something different here if I'm not overlooking something.

I closed the related pull request since unfortunately I cannot dig any deeper without the required hardware.

hughsie, 2025-02-21:
@lvoegl also note I've been working on https://github.com/fwupd/fwupd/pull/8486 -- it's pre-alpha but is a lot more understandable than the version in the modem-manager plugin IMHO. I'd appreciate your comments and suggestions very much! On that vein, if you can send me a RM520N-GL and some firmware I'd be very grateful.

MankWang, 2025-02-24:
Can this be restored to the code before 2.0.0 first? Currently, this issue has already caused many modules of Quectel to be unable to be upgraded.

MankWang, 2025-02-25:
RM520N-GL, EM160R-GL, EM061K-GL. Currently, none of the modules I am testing can be upgraded.
RM520N-GL, EM160R-GL are PCIE modules.
EM061K-GL is USB module.
If you need logs, I can also provide.

hughsie, 2025-02-25:
>  RM520N-GL, EM160R-GL are PCIE modules.
> EM061K-GL is USB module.

Can you share what they are? e.g. QMI+PDC, MBIM-QDU, Firehose?

MankWang, 2025-02-25:
They all use firehose for upgrading.

hughsie, 2025-02-25:
> They all use firehose for upgrading.

Triggered into firehose using MBIM-QDU?

MankWang, 2025-02-26:
The PCIE modules are triggered to enter firehose by sending switching commands through the wwan0qcdm0 port .

[image]

The USB modules switch to edl mode by this mbim message

[image]

shenyunet, 2025-02-26:
The RM520n-GL module with a vendor ID of 1EAC is an OEM version, while the standard version has a module ID of 17CB:0801. The 1EAC version may not be able to upgrade successfully and requires a special method to change the module ID to 17CB.

TGX03, 2025-03-20:
Changing the module ID could potentially be dangerous, as the 1EAC version is the module shipped in Lenovo laptops. The UEFI on Lenovo laptops enforces a whitelisting for WWAN modules, allowing only Lenovo OEM versions and otherwise refusing to boot. Meaning if the module ID would be changed, the module could no longer be used in the laptop.

hughsie, 2025-05-04:
@MankWang I've merged the modem refactor -- the new code is much more understandable. @iyanmv can you retest and close if it now works please.

MankWang, 2025-05-06:
[fwupd.log](https://github.com/user-attachments/files/20055238/fwupd.log)
[fwupdmgr.log](https://github.com/user-attachments/files/20055237/fwupdmgr.log)
@hughsie  There is still a problem, the upgrade writing process flashed by without actually writing the firmware. 
Please refer to the attached log for the detailed process.

iyanmv, 2025-05-08:
I get the exact same issue as @MankWang

vvdaal, 2025-05-23:
Just adding my two cents here, exact same issue.
Using a ThinkPad T14s Gen 6 with the RM520N-GL

If useful, happy to try things or give logs or give more information.

Queztion, 2025-07-07:
Issue still reproducible in X1 2in1 Gen 9 with RM520N-GL using Fedora 42; fwupdmgr 2.0.12

[fedora_fwupdmg_2.0.12.zip](https://github.com/user-attachments/files/21100741/fedora_fwupdmg_2.0.12.zip)

Environment is as follows:
fwupdmgr: fwupd 2.0.12
ModemManager: ModemManager.x86_64     1.22.0-5.fc42
QMI: libqmi.x86_64 1.34.0-7.fc42
MBIM: libmbim.x86_64 1.30.0-5.fc42
Kernel: 
6.15.4-200.fc42.x86_64

OS: Fedora 42

Issue still reproducible:
device version not updated on success, RM520NGLAAR03A03M4G_JP_04.206.04.206 != RM520NGLAAR03A03M4G_JP_04.209.04.209

Queztion, 2025-07-16:
Issue still reproducible in X1 2in1 Gen 9 with EM160R using Fedora 42; fwupdmgr 2.0.12

fwupdmgr: fwupd 2.0.12
ModemManager: ModemManager.x86_64     1.22.0-5.fc42
QMI: libqmi.x86_64 1.34.0-7.fc42
MBIM: libmbim.x86_64 1.30.0-5.fc42
Kernel: 
6.15.4-200.fc42.x86_64

OS: Fedora 42

Logs:

[2.0.12_fedora_em160.zip](https://github.com/user-attachments/files/21254175/2.0.12_fedora_em160.zip)

Error message after update:

Restarting device…       [*************************************  ] Less than one minute remaining…
failed to attach: failed to wait for attach replug: device 8e92eda1ab953dd9f27d11cd320c36e5d1127030 did not come back

NOTE:
I can't seem to find this device anywhere: 8e92eda1ab953dd9f27d11cd320c36e5d1127030.

Even if I list down fwupdmgr get-devices

hughsie, 2025-07-25:
@iyanmv can you try with hughsie/modems-less-raw please? i.e. what's in https://github.com/fwupd/fwupd/pull/9070

iyanmv, 2025-08-29:
> [@iyanmv](https://github.com/iyanmv) can you try [#9182](https://github.com/fwupd/fwupd/pull/9182) please

I'll be able to try that branch in a few hours. I'll post the results here.

iyanmv, 2025-08-30:
I was able to update the firmware with that branch 🎉

iyanmv, 2025-08-30:
Oh... great! Now I cannot boot the laptop 🤦‍♂

I got the following error after rebooting. I can't even enter the BIOS settings. I guess this update should not be in the stable channel.

[image]

iyanmv, 2025-08-30:
> Changing the module ID could potentially be dangerous, as the 1EAC version is the module shipped in Lenovo laptops. The UEFI on Lenovo laptops enforces a whitelisting for WWAN modules, allowing only [...]

Maybe this 🙄

1alessandro1, 2025-08-30:
If anybody posts their own working PCI IDs, on Quectel modules you can change those if EDL mode works.

hughsie, 2025-08-31:
@iyanmv oh no! If you remove the WWAN card then the laptop will continue booting -- but that's obviously not the proper solution. @mrhpearson this appears to be the same as what happened to mine -- I've demoted https://fwupd.org/lvfs/firmware/108829 as a precaution. 

@1alessandro1 how do you change the PCI IDs in EDL mode please?

1alessandro1, 2025-08-31:
Does the 520n appear under /dev/MHI_EDL or other 9008 endpoint? 

It's required to have a working model to compare with yours.

lvoegl, 2025-09-05:
> ​The issue seems to be that the new version of fwupd erases the raw data partition where the module configuration is stored.

The `rawprogram.xml` seems to include this line:
```
<erase a_rawdata_start_sector="6528" b_rawdata_num_sector="640" c_project_name="RM520NGLAP_LNV_VB" d_project_type="STD" vendor="quectel" />
```
The Firehose implementation filters only [well-known xml attributes](https://github.com/fwupd/fwupd/blob/330e2302465983eb17dee6f7eb146e7dcddf2e4e/plugins/qc-firehose/fu-qc-firehose-impl.c#L451-L456) for erases, which makes the above line result in
```
<erase />
```
As far as I know this means 'erase all flash storage' - including calibration and all sorts of other data.

hughsie, 2025-09-05:
@lvoegl great detective work, although eek for my modem calibration data. Does https://github.com/fwupd/fwupd/pull/9204 solve things?

hughsie, 2025-09-05:
> By the way, how did you manage to boot the Lenovo laptop with the module installed after the firmware upgrade

Lenovo are preparing a special BIOS firmware which will allow the laptop to boot when it has these special VID/PIDs.

closed 2025-09-05 by hughsie

Queztion, 2025-09-10:
Hello @hughsie ,

There is good news and bad news.

**GOOD NEWS:** 
We have tested fwupdmgr 2.0.15 with 9204 patch based on latest COPR build, and fw upgrade/downgrade is successful.

Logs attached for: fw downgade
RM520NGLAAR03A03M4G_JP_04.207.04.207 to 
RM520NGLAAR03A03M4G_JP_04.206.04.206 

[fedora_rm520_fwupd_2.0.15v2.zip](https://github.com/user-attachments/files/22252629/fedora_rm520_fwupd_2.0.15v2.zip)

Successful upgrade also conducted for
RM520NGLAAR03A03M4G_JP_04.206.04.206  to
RM520NGLAAR03A03M4G_JP_04.207.04.207  to
RM520NGLAAR03A03M4G_JP_04.209.04.209

Test conducted on X1 2n1 Gen9 with Fedora 42 and fwup 2.0.15-0.1224.202050909 build.

**BAD NEWS:**
HOWEVER, We are still getting the following errors in fwupdtool logs:
08:53:08.421 FuEngine             failed to add device /org/freedesktop/ModemManager1/Modem/2: failed to add device using on modem_manager: failed to subclass open: failed to open /dev/wwan0qcdm0: Permission denied

I am not sure if that refers to another issue or some issue that might come up in the future.

Also, the test was conducted on another system with RM520.
We are still looking for ways to recover the WWAN card where HWID got changed and reaching out to relevant teams.

But all in all, thank you and everyone else for your support!

vishnuocv, 2025-09-18:
@iyanmv I've sent a mail to you @ me@iyanmv.com. Please have a check.
Thank you.

iyanmv, 2025-09-18:
> [@iyanmv](https://github.com/iyanmv) I've sent a mail to you @ [me@iyanmv.com](mailto:me@iyanmv.com). Please have a check. Thank you.

Ups, yeah, I didn't see it, thanks!