fwupd/fwupd #8389: 8bitdo pro 2 not recognized

FeyrisTan, 2025-01-23:
Controller works fine both wired and with bluetooth.
However fwupd does not show the controller. 

**fwupd version information**
2.0.4

```shell
❯ fwupdtool get-report-metadata
Loading…                 [*****************                      ]21:03:16.313 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/host9/target9:0:0/9:0:0:0/block/sdb: failed to setup: failed to read version: command fail with status 2, senseKey 0x05, asc 0x26, ascq 0x00
Loading…                 [******************                     ]21:03:16.324 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-9/2-9.2/2-9.2:1.0/host8/target8:0:0/8:0:0:0/block/sdc: failed to setup: failed to read version: command fail with status 2, senseKey 0x05, asc 0x20, ascq 0x00
Waiting…                 [*************************************  ]
BatteryLevel:           101
BatteryThreshold:       10
BootTime:               1737664860
CompileVersion(com.hughsie.libjcat): 0.2.2
CompileVersion(com.hughsie.libxmlb): 0.3.21
CompileVersion(info.libusb): 1.0.27
CompileVersion(org.freedesktop.Passim): 0.1.8
CompileVersion(org.freedesktop.fwupd): 2.0.4
CpuArchitecture:        x86_64
CpuModel:               Intel 13th Gen Intel Core™ i5-13600K
DisplayState:           connected
DistroId:               arch
DistroName:             Arch Linux
DistroPrettyName:       Arch Linux
FwupdSupported:         True
HostBaseboardManufacturer: ASRock
HostBaseboardProduct:   Z790 PG Lightning
HostBiosMajorRelease:   05
HostBiosMinorRelease:   1b
HostBiosVendor:         American Megatrends International, LLC.
HostBiosVersion:        15.04
HostEnclosureKind:      3
HostFamily:             To Be Filled By O.E.M.
HostFirmwareMajorRelease: ff
HostFirmwareMinorRelease: ff
HostProduct:            Z790 PG Lightning
HostSku:                To Be Filled By O.E.M.
HostVendor:             ASRock
KernelCmdline:          zswap.enabled=0 nowatchdog
KernelName:             Linux
KernelVersion:          6.13.0-2-cachyos
LidState:               unknown
PassimDownloadSaving:   0
PowerState:             unknown
RuntimeVersion(com.hughsie.libjcat): 0.2.2
RuntimeVersion(com.hughsie.libxmlb): 0.3.21
RuntimeVersion(org.freedesktop.fwupd): 2.0.4
RuntimeVersion(org.freedesktop.fwupd-efi): 1.7
RuntimeVersion(org.kernel): 6.13.0-2-cachyos
DeviceId:               23bdd675150095da1f734b59f381b23c669dbe69
  pre:
      CapsuleApplyMethod: cod
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot
      MissingCapsuleHeader: False
  post:
      LastAttemptStatus: 0x0
      LastAttemptVersion: 0x5e0
DeviceId:               aa57281a0d8785aa3ead30ef2ef4336bddaf3949
  pre:
      CapsuleApplyMethod: cod
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot
      MissingCapsuleHeader: False
  post:
      LastAttemptStatus: 0x0
      LastAttemptVersion: 0x0
DeviceId:               cef254ddb406e77ace86f4cd92e2a7db5aa308fc
  pre:
[... 261 lines cut ...]
          <device_path>VenHw(46B29808-06F0-4CA0-9F92-5C4967829278)</device_path>
          <am_healthy>0x1</am_healthy>
        </firmware>
        <firmware gtype="FuAcpiPhatHealthRecord">
          <version_raw>0x1</version_raw>
          <data size="0x24b">[GInputStream]</data>
          <guid>24721fc5-2760-4391-aa5a-5c0324be9419</guid>
          <device_path>PciRoot(0x0)</device_path>
          <am_healthy>0x1</am_healthy>
        </firmware>
      </firmware>
      
      LinuxLockdown:    none
      Pcr0_SHA256:      a52c94c78d0b846f86144d684054b432da54d9dc0a293723e59c01309ad53c2b
      TpmEventLog:      0x00000008 d681837b3b8f5fd685bfad042ca78e6884e10e1c8f272dbccd2fd10be738a7fa [BSsQp8fZZUGBQCrd6Ur2PA==]
      0x8000000a 22ce9e3428e2b401df130690c0e5353674e56a9e3289326cd78c69fdc393444f [HU1lYXN1cmVkIFVFRkkgRmlybXdhcmUgVm9sdW1lAADT/wAAAAAAAAsAAAAAAA==]
      0x8000000a f2d59cffcb0a83631a41166a1a0822baed786eb85ad4fa29df1ae8a7fafa5eae [KUZ2KEI3QTNDQUFBLTRFMTAtNDRBNS04QzJDLUQyRjM4RDVCODA4MykAAACx/wAAAAAAAAwAAAAAAA==]
      0x8000000a fa98bdf34990f69981e246f8ffc8410111a1a11aab7055e7e27b7adee0131d52 [B0ZWX01BSU4AAEf/AAAAAAAAagAAAAAA]
      0x00000004 df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119 [AAAAAA==]
      PCR0: a52c94c78d0b846f86144d684054b432da54d9dc0a293723e59c01309ad53c2b
      TpmFamily:        2.0
      EfivarsNvramUsed: 51773
      SecureBoot:       Disabled
      UEFIUXCapsule:    Enabled
```

hughsie, 2025-01-23:
Can you upload the full output of `sudo fwupdtool get-devices --plugins ebitdo -vv` please.

FeyrisTan, 2025-01-23:
```shell
21:22:16.326 FuDebug              verbose to debug (on console 0)
21:22:16.326 FuEngine             starting fwupd 2.0.5...
21:22:16.326 FuMain               locked /run/lock/fwupdtool
21:22:16.328 FuMain               failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Loading…: 0%
21:22:16.328 FuConfig             loading config /etc/fwupd/fwupd.conf
21:22:16.328 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
21:22:16.328 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
21:22:16.328 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
21:22:16.328 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
21:22:16.328 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
21:22:16.328 FuConfig             Failed to open file "/var/etc/fwupd/fwupd.conf": No such file or directory
21:22:16.328 FuConfig             ::configuration loaded
21:22:16.328 FuRemoteList         loading /usr/share/fwupd/metainfo
21:22:16.328 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
21:22:16.328 XbSilo               file: 91d9df29-8f58-4952-bb9c-88e037d61388, current:91d9df29-8f58-4952-bb9c-88e037d61388
21:22:16.328 XbSilo               loading silo with existing file contents
21:22:16.328 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
21:22:16.328 FuRemoteList         ::remote_list changed
21:22:16.328 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
21:22:16.328 FuRemoteList         ::remote_list changed
21:22:16.328 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
21:22:16.328 FuRemoteList         ::remote_list changed
21:22:16.328 FuEngine             priority local and vendor-directory is not download remote, so bumping
21:22:16.328 FuRemoteList         ordering lvfs-testing=lvfs+1
21:22:16.328 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Loading…: 1%
Loading…: 2%
21:22:16.329 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
21:22:16.329 FuHistory            got schema version of 14
Loading…: 3%
Loading…: 4%
21:22:16.332 FuPlugin             load(/usr/lib/fwupd-2.0.5/libfu_plugin_modem_manager.so)
21:22:16.332 FuQuirks             open database /var/cache/fwupd/quirks.db
21:22:16.333 FuQuirks             mtimes unchanged: quirks,pci.ids:1735913538,usb.ids:1735913538,pnp.ids:1735913538,oui.txt:1735913538, doing nothing
21:22:16.333 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
21:22:16.333 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
21:22:16.333 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
21:22:16.333 XbSilo               file: e0526dd4-3fb9-2956-7ea5-916ade364da5, current:e0526dd4-3fb9-2956-7ea5-916ade364da5
21:22:16.333 XbSilo               loading silo with existing file contents
Loading…: 5%
21:22:16.333 FuConfig             loading config /etc/fwupd/fwupd.conf
21:22:16.333 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
21:22:16.333 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
21:22:16.333 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
21:22:16.333 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
21:22:16.333 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
21:22:16.333 FuConfig             Failed to open file "/var/etc/fwupd/fwupd.conf": No such file or directory
21:22:16.333 FuConfig             ::configuration loaded
21:22:16.333 FuStruct             FuStructSmbiosEp64:
  anchor_str: _SM3_
  entry_point_csum: 0x8e
  entry_point_len: 0x18
  smbios_major_ver: 0x3
  smbios_minor_ver: 0x5
  smbios_docrev: 0x0
  entry_point_rev: 0x1
  reserved0: 0x0
  structure_table_len: 0xdc2
  structure_table_addr: 0x65bcd000
21:22:16.333 FuStruct             FuStructSmbiosStructure:
  type: 0x0
  length: 0x1a
  handle: 0x0
21:22:16.333 FuStruct             FuStructSmbiosStructure:
  type: 0x1
  length: 0x1b
  handle: 0x1
21:22:16.333 FuStruct             FuStructSmbiosStructure:
[... 4249 lines cut ...]
  Name:                 uefi
  Enabled:              true
  DoneSetup:            true
  CanInvalidate:        false
FuUdevBackend:
  Name:                 udev
  Enabled:              true
  DoneSetup:            true
  CanInvalidate:        false
DoneColdplug:           true
FuBluezBackend:
  Name:                 bluez
  Enabled:              true
  DoneSetup:            true
  CanInvalidate:        false
FuEbitdoPlugin:
  Name:                 ebitdo
  Flags:                ready

21:22:16.348 FuEngine             failed to update history database: device ID 362301da643102b9f38477387e2193e57abaa590 was not found
Idle…: 0%
21:22:16.348 FuEngine             resetting update motd timeout
No detected devices
21:22:16.348 FuMain               No detected devices
```

hughsie, 2025-01-24:
That looks like a new PID that we don't support -- https://github.com/fwupd/fwupd/blob/main/plugins/ebitdo/ebitdo.quirk

It might be possible to add support for it by copying the quirk entry and adding PID 3208 -- but 8bitdo have been ignoring us for some time so if it fails we have no idea how to fix it.

FeyrisTan, 2025-01-24:
I created the file ebitdo.quirk in /usr/share/fwupd/quirks.d/ and added this entry:

[USB\VID_2DC8&PID_3208]
Plugin = ebitdo
Flags = is-bootloader
InstallDuration = 120

Now I get: failed to add device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-5: failed to setup: failed to send to device on ep 0x01: USB error: Input/Output Error [-1]

hughsie, 2025-01-24:
I think it's probably due to the 8bitdo protocol changing; it's been drifting for some time. e.g. https://github.com/fwupd/fwupd/commit/3604b897b5072b74250d369b60730e849f419709

hughsie, 2025-05-04:
8bitdo doesn't want to add support, sorry.

closed 2025-05-04 by hughsie