fwupd/fwupd #8405: Core Dump error during fwupd.service bootup

eboye, 2025-01-27:
**Describe the bug**
In system logs and inspecting the systemctl status of fwupd service there's an error that it failed to boot.
If the service is restarted after boot, at some point, it starts the service as it should.

**Steps to Reproduce**
Not sure what is causing it.

**Expected behavior**
It should start the service without errors.

**fwupd version information**
fwupd: 2.0.4.1
fwupd-efi: 1.7-1

```shell
fwupdtool get-report-metadata

Loading…                 [****                                   ]23:30:47.428 FuPluginUefiCapsule  skipping device that failed coldplug: ESRT GUID '00000000-0000-0000-0000-000000000000' was not valid
Loading…                 [*****************                      ]23:30:48.035 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:03.1/0000:0a:00.0/0000:0b:00.0/0000:0c:00.0: failed to setup: cannot parse datastream
Waiting…                 [*************************************  ]
BatteryLevel:           101
BatteryThreshold:       10
BootTime:               1737912792
CompileVersion(com.hughsie.libjcat): 0.2.2
CompileVersion(com.hughsie.libxmlb): 0.3.21
CompileVersion(info.libusb): 1.0.27
CompileVersion(org.freedesktop.Passim): 0.1.8
CompileVersion(org.freedesktop.fwupd): 2.0.4
CpuArchitecture:        x86_64
CpuModel:               Advanced Micro Devices, Inc. AMD Ryzen 7 5800X 8-Core Processor
DisplayState:           connected
DistroId:               arch
DistroName:             Arch Linux
DistroPrettyName:       Arch Linux
FwupdSupported:         True
HostBaseboardManufacturer: Gigabyte Technology Co., Ltd.
HostBaseboardProduct:   X570 AORUS ELITE
HostBiosMajorRelease:   05
HostBiosMinorRelease:   11
HostBiosVendor:         American Megatrends International, LLC.
HostBiosVersion:        F36e
HostEnclosureKind:      3
HostFamily:             X570 MB
HostFirmwareMajorRelease: ff
HostFirmwareMinorRelease: ff
HostProduct:            X570 AORUS ELITE
HostSku:                Default string
HostVendor:             Gigabyte Technology Co., Ltd.
KernelCmdline:          rootfstype=ext4
KernelName:             Linux
KernelVersion:          6.12.10-arch1-1
LidState:               unknown
PassimDownloadSaving:   20054132
PowerState:             unknown
RuntimeVersion(com.hughsie.libjcat): 0.2.2
RuntimeVersion(com.hughsie.libxmlb): 0.3.21
RuntimeVersion(org.freedesktop.fwupd): 2.0.4
RuntimeVersion(org.freedesktop.fwupd-efi): 1.7
RuntimeVersion(org.kernel): 6.12.10-arch1-1
      LinuxLockdown:    none
      Pcr0_SHA256:      6ceb0265ff2837da28ee95af0d7e063cfdf0a543ce94f927de9265070d4fe8a7
      TpmEventLog:      0x00000008 d681837b3b8f5fd685bfad042ca78e6884e10e1c8f272dbccd2fd10be738a7fa [BSsQp8fZZUGBQCrd6Ur2PA==]
      0x8000000a c220eb72c92129da415dc8b1cbe6dc10fa0b25c8dab6d6482c5f8c3425335f64 [KUZ2KEY2NDlGQzJELUMwRTYtNDI2Mi1BRDUxLTBDRTY0Rjc2NDI5RikAAPDj/wAAAAAAEAAAAAAAAA==]
      0x8000000a 31f157c595070c460d11cddf3fcece50ca5337a1c8f516c28c520c961e6b187e [B0ZWX01BSU4A8Jb/AAAAAAAATQAAAAAA]
      0x00000001 ec8c6eaf428a2320d34d0fd7a823ec57ba30adf2fe65302f7574ac5f90740e1b [FEVtYmVkZGVkIFVFRkkgRHJpdmVyGNCItwAAAAAA3AEAAAAAAA==]
      0x00000004 df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119 [AAAAAA==]
      PCR0: 6ceb0265ff2837da28ee95af0d7e063cfdf0a543ce94f927de9265070d4fe8a7
      TpmFamily:        2.0
      EfivarsNvramUsed: 28934
      SecureBoot:       Disabled
[... 141 lines cut ...]
      GUIDs:              74324928-8228-5088-a43d-0c5b85082861 ← IDE\WDC_WD40EZRZ-00GXCB0____________________80.00A80
                          04909b26-8cba-52e1-8905-435f5b538ead ← IDE\0WDC_WD40EZRZ-00GXCB0____________________
                          e46f6c45-e987-53c6-b991-87590e86a249 ← WDC WD40EZRZ-00GXCB0
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Can tag for emulation
    

```

</details>

**Additional questions**

- Operating system and version:
Archlinux, kernel 6.12.10-arch1-1

- Have you tried rebooting?
Yes, I'll try rebooting few more times and report back if this is the case every time.

- Is this a regression? 
Not sure

superm1, 2025-01-27:
Do you have a core dump file to attach?

Also does the journal for fwupd.service show anything interesting?

eboye, 2025-01-27:
This is the coredump from systemctl status of the service when it crashed.

```shell
systemd-coredump[225112]: [🡕] Process 3528 (fwupd) of user 0 dumped core.
                                                        
                                                        Stack trace of thread 3528:
                                                        #0  0x00007c57b1a47d1b g_logv (libglib-2.0.so.0 + 0x62d1b)
                                                        #1  0x00007c57b1a47f94 g_log (libglib-2.0.so.0 + 0x62f94)
                                                        #2  0x00007c57b1a9ee73 n/a (libglib-2.0.so.0 + 0xb9e73)
                                                        #3  0x00007c57b1a3fb6c g_main_context_new_with_flags (libglib-2.0.so.0 + 0x5ab6c)
                                                        #4  0x00007c57b1c3667c g_dbus_connection_send_message_with_reply_sync (libgio-2.0.so.0 + 0x10167c)
                                                        #5  0x00007c57b1c43491 n/a (libgio-2.0.so.0 + 0x10e491)
                                                        #6  0x00007c57b1c4fa9a n/a (libgio-2.0.so.0 + 0x11aa9a)
                                                        #7  0x00007c57b1c4fc9b g_dbus_proxy_call_sync (libgio-2.0.so.0 + 0x11ac9b)
                                                        #8  0x000059fd5e4f2457 n/a (n/a + 0x0)
                                                        #9  0x00007c57b1c3ced2 n/a (libgio-2.0.so.0 + 0x107ed2)
                                                        #10 0x00007c57b1a42559 n/a (libglib-2.0.so.0 + 0x5d559)
                                                        #11 0x00007c57b1aa5257 n/a (libglib-2.0.so.0 + 0xc0257)
                                                        #12 0x00007c57b1a43287 g_main_loop_run (libglib-2.0.so.0 + 0x5e287)
                                                        #13 0x000059fd5e4f4c1b n/a (n/a + 0x0)
                                                        #14 0x000059fd5e4eb709 n/a (n/a + 0x0)
                                                        #15 0x00007c57b17bbe08 n/a (libc.so.6 + 0x25e08)
                                                        #16 0x00007c57b17bbecc __libc_start_main (libc.so.6 + 0x25ecc)
                                                        #17 0x000059fd5e4ebbd5 n/a (n/a + 0x0)
                                                        ELF object binary architecture: AMD x86-64
```

I've rebooted 3 times now and can't see this anymore. I'll report back if it happens again and what changed in the process if anything.

superm1, 2025-01-27:
Has glib been upgraded perhaps between attempts?

eboye, 2025-01-28:
Could be. I see it was upgraded few days ago in /var/log/pacman.log:

```
[2025-01-24T13:39:00+0100] [ALPM] upgraded glibc (2.40+r16+gaa533d58ff-2 -> 2.40+r66+g7d4b6bcae91f-1)
```

But I think I did a reboot in the meantime. But this can be the cause if relevant:

```
[2025-01-27T01:13:40+0100] [ALPM] transaction started
[2025-01-27T01:13:40+0100] [ALPM] upgraded gtk3 (1:3.24.43-4 -> 1:3.24.48-1)
[2025-01-27T01:13:40+0100] [ALPM] upgraded webrtc-audio-processing-1 (1.3-3 -> 1.3-4)
[2025-01-27T01:13:40+0100] [ALPM] transaction completed
[2025-01-27T01:13:40+0100] [ALPM] running '30-systemd-update.hook'...
[2025-01-27T01:13:40+0100] [ALPM] running 'glib-compile-schemas.hook'...
[2025-01-27T01:13:40+0100] [ALPM] running 'gtk-query-immodules-3.0.hook'...
[2025-01-27T01:13:40+0100] [ALPM] running 'paccache-hook.hook'...
```

As this is the last update from today before last reboot.

hughsie, 2025-05-04:
Okay, please re-open if this happens again.

closed 2025-05-04 by hughsie