fwupd/fwupd #8433: Missing GUID in Redfish FirmwareInventory should not ignore all devices

Arno500, 2025-02-06:
**Describe the bug**
When scrapping an HPE iLO 5 Redfish interface , devices that do not have a detected GUID makes the Redfish plugin stop when coldplugging, like for:
```
response: {
  "@odata.context" : "/redfish/v1/$metadata#SoftwareInventory.SoftwareInventory",
  "@odata.etag" : "W/\"1DF5B87E\"",
  "@odata.id" : "/redfish/v1/UpdateService/FirmwareInventory/8/",
  "@odata.type" : "#SoftwareInventory.v1_2_0.SoftwareInventory",
  "Id" : "8",
  "Description" : "SystemRomBackup",
  "Name" : "Redundant System ROM",
  "Oem" : {
    "Hpe" : {
      "@odata.context" : "/redfish/v1/$metadata#HpeiLOSoftwareInventory.HpeiLOSoftwareInventory",
      "@odata.type" : "#HpeiLOSoftwareInventory.v2_1_0.HpeiLOSoftwareInventory",
      "DeviceClass" : null,
      "DeviceContext" : "System Board"
    }
  },
  "Updateable" : true,
  "Version" : "A46 v3.00 (01/26/2024)"
}
```
It shows: `disabling plugin because: failed to coldplug using redfish: no GUID for device`
In this situation, I believe the device itself should be ignored instead of removing all of them.

**Steps to Reproduce**
Enable the Redfish scrapping using `redfish.conf`

**Expected behavior**
I expect to see all the compatible devices (with a GUID, obviously)

**fwupd version information**
```
runtime   com.dell.libsmbios            2.4
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.gusb          0.3.0
runtime   org.kernel                    6.9.12-1.el8.x86_64
compile   org.freedesktop.fwupd         1.7.8
runtime   org.freedesktop.fwupd         1.7.8
```

```shell
~ fwupdtool get-report-metadata
Command not found

Use fwupdtool --help for help
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): DNF

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
~ fwupdmgr get-devices --show-all-devices
WARNING: UEFI ESP partition not detected or configured
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
ProLiant XL225n Gen10 Plus
│
├─EPYC 7502P 32-Core Processor:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             AMD
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         997104da-ec11-5e04-9d94-ac5501fbf609
│                         2b9136a8-1b96-5f53-9a69-57baf6f94c9b
│                         a477c4dd-63da-5027-afd3-43ae87333563
│     Device Flags:       • Internal device
│
├─I350 Gigabit Network Connection:
│     Device ID:          7ba99e3d78111c3cbf4dbedfef81370058667697
│     Current version:    01
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              04fc369c-0f7e-5fd6-8c60-c1b24c7b9cf7
│                         63f9a955-32ba-51be-9dbc-8c4e4e659f37
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│
├─Integrated Lights-Out Standard Management Processor Support and Messaging:
│     Device ID:          fd3da06f3cd443d272abae5b53e0faad38a6bd56
│     Current version:    07
│     Vendor:             Hewlett-Packard Company (PCI:0x103C)
│     GUIDs:              a7d7e715-7349-5a26-93a8-7de0d964d019
│                         951821d4-b679-5c07-b088-474071ea3fb1
│                         1ddffdce-4b27-5f5f-99ff-f6286b7d0dd6
│                         61ca3d96-355b-5431-abe2-dc241b0ca17b
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│
├─MGA G200eH3 (iLO5 VGA):
│     Device ID:          29364b287a9e482b1c8aa179cf1402c4198e948b
│     Current version:    02
│     Vendor:             Matrox Electronics Systems Ltd. (PCI:0x102B)
│     GUIDs:              bfd0464d-ee0f-520c-870c-ed1975810cef
│                         3c97f57b-7c5f-56f7-8c42-fc1e7a56ce9f
│                         9fda2e56-849d-5147-a61e-628c70326b26
│                         17f14a5a-452c-5215-ba10-0851b8b23b48
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│
├─MT27710 Family [ConnectX-4 Lx]:
│     Device ID:          8a5b008ab82a08b03628499b735addc8b416282b
│     Vendor:             Mellanox Technologies (PCI:0x15B3)
│     GUIDs:              ae494b97-70c9-5752-991e-c0bbbc75c4f3
│                         a5761d96-bbb6-572a-8870-6565e33c220e
│                         c0a8c171-a928-58b8-b0c4-9cde53db38cc
│                         55871977-4246-5902-a6f9-58226165d751
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│
├─MT27710 Family [ConnectX-4 Lx]:
│     Device ID:          0d24e848d3dd7a64d4c70289087a6d2e983807c1
│     Vendor:             Mellanox Technologies (PCI:0x15B3)
│     GUIDs:              ae494b97-70c9-5752-991e-c0bbbc75c4f3
│                         a5761d96-bbb6-572a-8870-6565e33c220e
│                         c0a8c171-a928-58b8-b0c4-9cde53db38cc
│                         55871977-4246-5902-a6f9-58226165d751
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│
├─UEFI Device Firmware:
│     Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│     Summary:            UEFI ESRT device
│     Current version:    14321010
│     Minimum Version:    2
│     Vendor:             DMI:HPE
│     Update State:       Success
│     Update Error:       Not updatable as UEFI ESP partition not detected
│     GUID:               3792094f-4eb9-11e2-9d5a-f9377442e311
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│
├─UEFI Device Firmware:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    4
│     Minimum Version:    4
│     Vendor:             DMI:HPE
│     Update State:       Success
│     Update Error:       Not updatable as UEFI ESP partition not detected
│     GUID:               164ffef2-15d3-4f7c-bae6-72097de8af78
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│
├─UEFI dbx:
│     Device ID:          362301da643102b9f38477387e2193e57abaa590
│     Summary:            UEFI revocation database
│     Current version:    77
│     Minimum Version:    77
│     Vendor:             UEFI:Linux Foundation
│     Install Duration:   1 second
│     GUIDs:              2498853f-4aa9-50ec-8103-db973a870f69
│                         7c45fbad-bf85-5c16-af0d-5bdd8a4264a7
│                         c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│                         8b0c62fa-23e7-518d-b3b5-140d8afa6079
│                         d1264ba7-d302-581c-a295-9eb50889d4ba
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Only version upgrades are allowed
│                         • Signed Payload
│
└─VK000480GWTHA:
      Device ID:          c2148b001c29143462428b5943859a01d0616635
      Summary:            ATA drive
      Current version:    HPG5
      Vendor:             HPE (OUI:ace42e, ATA:0x1590)
      GUIDs:              76592505-d2df-5c46-904b-baa5835cf497
                          cb0eac44-9b14-53fa-a889-a9184819a473
                          9214953f-2c02-571b-b883-74f3195efa37
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

</details>

**Additional questions**

- Operating system and version:  Rocky Linux 8.10
- Have you tried rebooting? :P
- Is this a regression? No

hughsie, 2025-02-06:
> I believe the device itself should be ignored instead of removing all of them.

Agree! Would you consider creating a pull request with this change? We can help you.

superm1, 2025-02-06:
Before doing a PR; make sure that it still fails on a current version too (1.7.x is EOL for ~8 months).

Arno500, 2025-02-06:
Indeed! The code seemed to still have this behavior, so I did not particularly try to rebuild a more recent version on my system, but that's of course a prerequisite :)

hughsie, 2025-05-04:
@Arno500 can we close this now?

Arno500, 2025-05-04:
Absolutely, sorry for forgetting about it :)

closed 2025-05-04 by Arno500