fwupd/fwupd #8435: failed to write-firmware: Blocked executable in the ESP, ensure grub and shim are up to date: failed to load /boot/efi/EFI/fedora/: Error opening file /boot/efi/EFI/fedora: Is a directory

mcatanzaro, 2025-02-06:
**Describe the bug**

I'm unable to update my dbx database because fwupd is trying to open the directory /boot/efi/EFI/fedora/ without the `O_DIRECTORY` flag, which does not work:

```shell
$ fwupdmgr update
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 20230301 to 20241101?                                  ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the list of forbidden signatures (the "dbx") to the latest      ║
║ release from Microsoft.                                                      ║
║                                                                              ║
║ An insecure version of Howyar's SysReturn software was added, due to a       ║
║ security vulnerability that allowed an attacker to bypass UEFI Secure Boot.  ║
║                                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Verifying…               [                                       ]
Successfully installed firmware
failed to write-firmware: Blocked executable in the ESP, ensure grub and shim are up to date: failed to load /boot/efi/EFI/fedora/: Error opening file /boot/efi/EFI/fedora: Is a directory
```

I assume there is not actually any blocked executable in the ESP. Looks like the check for whether there is a blocked executable is broken.

**Steps to Reproduce**
Steps to reproduce the behavior. I can reproduce by running `fwupmgr update`

**Expected behavior**
Upgrade should succeed

**fwupd version information**
fwupd-2.0.5-1.fc42

```shell
BatteryLevel:           101
BatteryThreshold:       10
BootTime:               1738850940
CompileVersion(com.hughsie.libjcat): 0.2.2
CompileVersion(com.hughsie.libxmlb): 0.3.21
CompileVersion(info.libusb): 1.0.27
CompileVersion(org.freedesktop.Passim): 0.1.8
CompileVersion(org.freedesktop.fwupd): 2.0.5
CpuArchitecture:        x86_64
CpuModel:               Advanced Micro Devices, Inc. AMD Ryzen 9 7950X 16-Core Processor
DisplayState:           connected
DistroId:               fedora
DistroName:             Fedora Linux
DistroPrettyName:       Fedora Linux 42 (Workstation Edition Prerelease)
DistroVariant:          workstation
DistroVersion:          42
FwupdSupported:         True
HostBaseboardManufacturer: ASUSTeK COMPUTER INC.
HostBaseboardProduct:   ProArt B650-CREATOR
HostBiosMajorRelease:   19
HostBiosMinorRelease:   06
HostBiosVendor:         American Megatrends Inc.
HostBiosVersion:        2506
HostEnclosureKind:      3
HostFamily:             To be filled by O.E.M.
HostFirmwareMajorRelease: ff
HostFirmwareMinorRelease: ff
HostProduct:            System Product Name
HostSku:                SKU
HostVendor:             ASUS
KernelName:             Linux
KernelVersion:          6.14.0-0.rc1.15.fc42.x86_64
LidState:               unknown
PassimDownloadSaving:   17295130
PlatformArchitecture:   x86_64
PowerState:             unknown
[... 224 lines cut ...]
├── symvers-6.11.11-300.fc41.x86_64.xz
├── symvers-6.12.10-200.fc41.x86_64.xz
├── symvers-6.12.8-200.fc41.x86_64.xz
├── symvers-6.14.0-0.rc1.15.fc42.x86_64.xz -> /lib/modules/6.14.0-0.rc1.15.fc42.x86_64/symvers.xz
├── System.map-6.11.11-300.fc41.x86_64
├── System.map-6.12.10-200.fc41.x86_64
├── System.map-6.12.8-200.fc41.x86_64
├── System.map-6.14.0-0.rc1.15.fc42.x86_64
├── vmlinuz-0-rescue-b8b00b2ea67f4aba873bcc16ecc6b8f1
├── vmlinuz-6.11.11-300.fc41.x86_64
├── vmlinuz-6.12.10-200.fc41.x86_64
├── vmlinuz-6.12.8-200.fc41.x86_64
└── vmlinuz-6.14.0-0.rc1.15.fc42.x86_64

6 directories, 22 files
```

**Additional questions**

- Operating system and version: Fedora 42
- Have you tried rebooting? not today, but I reboot daily and this update has been failing for a while now
- Is this a regression? not sure, it's been broken for a while
- Are you using an NVMe disk? yes
- Is secure boot enabled? yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? no

hughsie, 2025-02-06:
> Looks like the check for whether there is a blocked executable is broken.

Weird; there are two more reports on the LVFS with that exact error too. Can you attach the output of `sudo efibootmgr` pls.

hughsie, 2025-02-06:
OOpa, you've done that; ignore me.

hughsie, 2025-02-06:
>     data: 00 00 42 4f

That bit is somewhat unexpected. @vathpela  -- any ideas what that's supposed to be?

vathpela, 2025-02-06:
This is garbage ASUS is adding to boot entries their firmware has created (in violation of the spec /recommendations/) and tagged with those bytes in the "data" portion to imply some form of ownership, instead of actually putting something in the spec to represent that meaningfully.  I honestly could not tell you why some vendors do this.

hughsie, 2025-02-06:
@mcatanzaro can you do `cp /sys/firmware/efi/efivars/Boot0001-8be4df61-93ca-11d2-aa0d-00e098032b8c ~/` and attach it to this bug please; I'd like to add this to the fwupd test suite and want to make sure it's byte identical.

vathpela, 2025-02-06:
... I've blamed ASUS here but I've just realized that the ASUS machine I've got put "42 4f 0a" there instead.  Same firmware vendor and board vendor though, so probably they just messed up between two different board's firmwares.

mcatanzaro, 2025-02-06:
[Boot0001-8be4df61-93ca-11d2-aa0d-00e098032b8c.zip](https://github.com/user-attachments/files/18695202/Boot0001-8be4df61-93ca-11d2-aa0d-00e098032b8c.zip)

mcatanzaro, 2025-02-06:
> Can you attach the output of `sudo efibootmgr` pls.

BTW your issue report template should probably specify use of `sudo` or root where required. I used it for the `tree` command since it shows more output that way, but I remember from a previous bug report that fwupdmgr many not show the full desired output when run unprivileged.

hughsie, 2025-02-06:
@mcatanzaro i think this is fixed by https://github.com/fwupd/fwupd/pull/8436 -- can you try the COPR build of fwupd in https://copr.fedorainfracloud.org/coprs/rhughes/fwupd/build/8622318/ please.

mcatanzaro, 2025-02-06:
Yes, confirmed fixed. Thanks!

hughsie, 2025-02-06:
Known issue [wiki page](https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-dbx-failure-due-to-LoadOption-garbage-data) created, and added as a rule on the LVFS: 

[image]

closed 2025-02-07 by hughsie

piratenpanda, 2025-02-15:
I'm seeing the same on EndeavourOS with fwupd 2.0.6 on a ROG STRIX X670E-F GAMING WIFI. What info do you need?

superm1, 2025-02-15:
Open your own report with your own logs following the template.