fwupd/fwupd #8439: fwupdplugin-self-test test failing in 2.0.5

rishabh-chromium, 2025-02-07:
**Describe the bug**
`fwupdplugin-self-test` test failing on ChromeOS

**Steps to Reproduce**
Run unit test

**fwupd version information**
2.0.5

fwupd by itself works fine. I have attached the logs of the failure

[error.txt](https://github.com/user-attachments/files/18711228/error.txt)

**Additional questions**

- Operating system and version: ChromeOS
- Have you tried rebooting? Yes
- Is this a regression? Yes

hughsie, 2025-02-07:
The relevant lines are:

     libfwupdplugin/fu-self-test.c:1833:fu_common_kernel_search_func: assertion failed (g_strchomp(result2) == ""): (" foo/bar" == "")

@superm1 can you see why fu_kernel_reset_firmware_search_path() wouldn't work on ChromeOS? I'm also a bit confused as I don't think this code has changed for a very long time...

superm1, 2025-02-07:
Maybe the jail changed on Chrome and is presenting something kind of weird from that path?

rishabh-chromium, 2025-02-09:
Unit tests aren't run in minijail, just the daemon itself once it is flashed on chromeos

superm1, 2025-02-09:
Could the  build host have changed that search path potentially?

rishabh-chromium, 2025-02-10:
I realized that this test is now failing on 2.0.0 too. That means something definitely changed on our end. Which path should it be searching for?

superm1, 2025-02-13:
this should be fixed in main hopefully with the new locker stuff (see the tagged commits above)

closed 2025-02-13 by superm1