fwupd/fwupd #8440: Update failure ("failed to attach: failed to wait for attach replug: device ... did not come back") bricked my thunderbolt dock

haunma, 2025-02-08:
**Describe the bug**
I recently received a new-in-box HP Thunderbolt G4 dock (4J0A2AA) and connected it to the USB 4 port on my AMD Thinkpad P16s gen 2 (7840U).  Everything appeared to work correctly:  dmesg looked good, PCI passthrough worked correctly as evidenced by the attaching of the internal 2.5G Intel ethernet chip, and USB ports were working.  I did not check video at this point but it was likely working too.

I ran "fwupdadm update" which immediately recognized the dock and informed me that there was a substantially newer firmware with a couple of fixes specifically for AMD systems, so I decided to accept the update then and there.  The update proceeded fine at first, then appeared to stall near the end.  I let it sit another ten minutes until fwupdadm timed out with an error.  At that point I disconnected the dock from the laptop and made sure its LED was not blinking, then removed power for a minute.  Rest of the sad story after this update log:

```
$ fwupdmgr get-history
WARNING: UEFI capsule updates not available or enabled in firmware setup
See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
LENOVO 21K9CTO1WW
│
├─UEFI dbx:
│ │   Device ID:          362301da643102b9f38477387e2193e57abaa590
│ │   Previous version:   220
│ │   Update State:       Success
│ │   Last modified:      2025-02-01
│ │   GUID:               5971a208-da00-5fce-b5f5-1234342f9cf7
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │                       • Only version upgrades are allowed
│ │                       • Signed Payload
│ │
│ └─Secure Boot dbx Configuration Update:
│       New version:      371
│       Remote ID:        lvfs
│       Release ID:       35287
│       Summary:          UEFI Secure Boot Forbidden Signature Database
│       Variant:          x64
│       License:          Proprietary
│       Size:             21.2 kB
│       Created:          2023-05-09
│       Urgency:          High
│         Tested:         2024-11-20
│         Distribution:   fedora 39 (workstation)
│         Old version:    217
│         Version[fwupd]: 1.9.20
│         Tested:         2024-06-06
│         Distribution:   ubuntu 22.04
│         Old version:    267
│         Version[fwupd]: 1.9.15
│         Tested:         2024-06-04
│         Distribution:   fedora 39 (workstation)
│         Old version:    77
│         Version[fwupd]: 1.9.20
│         Tested:         2024-04-24
│         Distribution:   ubuntu 22.04
│         Old version:    83
│         Version[fwupd]: 1.9.16
│         Tested:         2024-03-21
│         Distribution:   fedora 39 (workstation)
│         Old version:    217
│         Version[fwupd]: 1.9.15
│         Tested:         2024-02-20
│         Distribution:   fedora 39 (workstation)
│         Old version:    77
│         Version[fwupd]: 1.9.5
│         Tested:         2024-01-12
│         Distribution:   fedora 39 (workstation)
│         Old version:    220
│         Version[fwupd]: 1.9.11
│         Tested:         2023-07-11
│         Distribution:   fedora 38 (workstation)
│         Old version:    211
│         Version[fwupd]: 1.9.2
│         Tested:         2023-07-03
│         Distribution:   ubuntu 22.04
[... 261 lines cut ...]
│
└─Unifying Receiver:
      Device ID:          4a67519fb668532a7178b08040629f2672d6cc26
      Summary:            Miniaturised USB wireless receiver
      Current version:    RQR12.03_B0025
      Bootloader Version: BOT01.02_B0015
      Vendor:             USB:0x046D|HIDRAW:0x046D
      Install Duration:   30 seconds
      GUID:               279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:       • Updatable
                          • Unsigned Payload
                          • Can tag for emulation

────────────────────────────────────────────────
Devices that were not updated correctly:
 • Thunderbolt Dock G4 (1.4.20.0 → 1.5.25.0)
Devices that have been updated successfully:
 • UEFI dbx (220 → 371)
```

**Additional questions**

- Operating system and version:   Void Linux (kernel 6.12.9_1 #1 SMP PREEMPT_DYNAMIC Fri Jan 10 00:53:27 UTC 2025 x86_64 GNU/Linux)
- Have you tried rebooting?  Yes
- Is this a regression?  Uncertain

hughsie, 2025-02-08:
> The next power up immediately started a flash pattern on the dock LED: three long, two short, pause, then repeat

I'm confused why it didn't enumerate at all, even in bootloader recovery state. @IfxBrent can you help us and tell us what might be happening? I've never seen this behaviour before and I've played with ccgx_dmx a **lot**.

>  Void Linux

We don't do any testing on Void, but I don't think it should affect anything like this. It might be worth plugging it into a Windows or Fedora machine just to check.

Out of interest, where did you purchase the dock *from*? I ask because I've known clones from other vendors behave badly when updating the firmware.

IfxBrent, 2025-02-08:
I am not sure what happened to the device. Dock should be able to recover itself even if the update is failure. 
Did this dock provide power laptop after the firmware update and before "securestart"?

haunma, 2025-02-08:
I purchased the dock from an IT reseller as a "new / open-box" unit.  It came in what appears to be the original box with official HP labels on the outside.  I can't rule out some kind of reject unit, of course, although the serial number checks out on the HP web site and lists a warranty period from January 2024 to January 2025.  I assumed some corporate IT dept surplussed their stock after it went out of warranty.  I'm going to try engaging with HP support on Monday, but as the second owner of a 13-month-old dock of unknown provenance, I'm not optimistic.

It would help to know what the three long / two short flash pattern means.  It's clearly intended to be diagnostic, but not documented anywhere on the open internet.  In particular, does it indicate a hardware failure?  There are warnings that an interrupted firmware update can lead to hardware damage.  This is obviously possible, but wearing my EE hat, I'm skeptical.  Thoughts?

If the diagnostic does indicate a HW fault and the odds of that being caused by the failed update are close to zero, then I would feel like I have a case to take up with the reseller.  But as it stands, the dock *was* working fine when I unboxed it, and stopped working after the attempted update, so claiming defective HW doesn't seem fair.

@IfxBrent, could you clarify your question?  I don't know what state(s) the dock was in, at what points.  HP does not document the LED patterns in detail, even the fast-flash vs slow-flash is a little obscure.  I couldn't say whether SureStart was activated.  I will say that at some point during the failed upgrade, it did re-enumerate in my dmesg (and again looked normal, with PCIe passthough etc).  But now it appears completely dead except for the LED flash pattern and the fan being constantly on.  No power delivery, no enumeration.  (It was charging the laptop prior to the failed update.)

haunma, 2025-02-08:
I should add that I'm happy to perform any further testing you think could narrow down the problem, both for future users and for the outside chance that the dock can be recovered.  I'll try plugging it into a Windows machine but AFAICT the TB4 stuff isn't alive at all; I'd be shocked if Windows notices that anything connected.

I'm also happy to open up the unit and poke around, although I don't see any magic reset buttons in the teardown pictures :/

hughsie, 2025-02-08:
> interrupted firmware update can lead to hardware damage. This is obviously possible, but wearing my EE hat

Am also an EE :) You can interrupt the firmware update with ccgx_dmc at any point in the update; it'll just fall back to recovery.

> I purchased the dock from an IT reseller as a "new / open-box" unit .. warranty period from January 2024 to January 2025

My suspicion would be that the unit has a hardware fault, unfortunately. @IfxBrent might be able to shed some light on the flashing LED, that might help, but even if you could flash the SPI directly I don't know the exact way the dock has failed. I'd certainly be taking it back to the reseller if that was me...

Richard

IfxBrent, 2025-02-09:
The question about power delivery is to clarify whether the USB PD part is alive or not. Looks your device USB PD has no function as well. 
Regarding LED behaviors, it might different from products or OEM vendors.  Probably go back to reseller or HP's customer service is a good way to figure out what happened to the device.

mark0wnik, 2025-04-25:
Hey, I have recently purchased (what was described to be 100% working) this exact same dock in this exact state - 3 long, 2 short flashes.
Would you be able to desolder and provide dumps of all the memory chips from the device? 
(3 on main board - U2701, U3501, U3305; 2 on daughter board - U4002, U453)
I've already tried looking through official firmware update but it's either compressed or not complete - only a partial update

IfxBrent, 2025-04-28:
Hi, for a recently purchased unit, i suggest you contact HP service to get a replace one, no matter it is a corrupted update or chip issue. It would be helpful for HP to clarity the root cause and solve it in the beginning.

haunma, 2025-04-28:
> Hey, I have recently purchased (what was described to be 100% working) this exact same dock in this exact state - 3 long, 2 short flashes. From what I know from HP laptops it means 'memory failure' - maybe a corrupted update?

I'm wondering if this is the same unit that I had previously!  I would not be surprised if the seller (this was on Ebay) decided to try again, perhaps this time without a return guarantee.  OTOH if returns are accepted I would take advantage of that...

If you can share the serial number, I may be able to track it down from my Ebay history and see if this is in fact the same unit.  I think it would be helpful to know if this behavior is a rare fluke, or something which happens more frequently.

FWIW I bought another G4 dock from a different seller, and I am very happy with it.  It has slightly newer firmware but could still be updated.  I haven't dared to try... yet.

danividanivi, 2025-06-25:
I have experienced something similar right now, updating from Thunderbold Dock G4 1.4.20.0 to 1.5.22.0.
My HP dock stopped working and started blinking its LED. I tried disconnecting everything, waiting a bit unpowered, but when I reconnect again the laptop it will blink forever.
Fortunately I could get another dock from my company, but I am not willing to update it again to try my luck.

kagurazakakotori, 2025-07-10:
> (After further testing it occured to me that only change necessary to make it work would probably be fixing offsets 40000-40FFF and 3C0000-3C0FFF in original U4002 dump)

@mark0wnik Will this make everything work? I've also experiencing this with updating 1.5.25.0->1.5.27.0, and had no luck after waiting for the led blinking for hours :(

mrwiora, 2026-10-22:
I had the same issue - but after upgrading from 1.5.25 to 1.5.27
The most probable reason was, that I have super restrictive USB Devices, which need to be allowed individually.
After a reboot of course all temporary rules were gone and the device could not report back to the OS.

However getting rid of the cached "failed update" was a nightmare.
The Device was always shown stuck in the state:
```
└─Thunderbolt Dock G4:
  │   Device ID:          8ca...
  │   Previous version:   1.5.25.0
  │   Install Duration:   14 minutes
  │   Update State:       Failed
  │   Update Error:       failed to attach: failed to wait for attach replug: device 8ca... did not come back
  │   Last modified:      2026-10-22 09:04:00
  │   GUID:               3d0...
  │   Device Flags:       • System requires external power source
  │                       • Supported on remote server
  │                       • Device stages updates
  │                       • Device can recover flash failures
  │                       • Updatable
  │                       • Signed Payload
  │                       • Can tag for emulation
```

Clearing failed with:

```
# fwupdmgr clear-results
0. Cancel
...

    21. aa4... [Thunderbolt Dock G4]
...
Choose device [0-29]: 21
Failed to find aa4... in history database: No devices found
``` 

all other attempts have failed.
Finally I found:
```
systemctl restart fwupd
fwupdmgr get-devices --no-unreported-check
```
and this reports now the correct version.