fwupd/fwupd #8441: Firmware update report failed to upload because of missing SSL cert

kimdre, 2025-02-08:
**Describe the bug**
I updated my firmware drivers. After the update the cli tool asked me to upload a report of the successful update.
The upload failed with
```
failed to upload file: SSL: no alternative certificate subject name matches target hostname 'fwupd.org'
```

**Steps to Reproduce**

Run 
```
sudo fwupdmgr refresh --force && sudo fwupdmgr get-updates && sudo fwupdmgr update
```

**Expected behavior**
Upload should probably have worked?

**fwupd version information**
Please provide the version of the daemon and client.

Installed via dnf on Fedora 41

```shell
BatteryLevel:           101
BatteryThreshold:       10
BootTime:               1738597384
CompileVersion(com.hughsie.libjcat):0.2.2
CompileVersion(com.hughsie.libxmlb):0.3.21
CompileVersion(org.freedesktop.Passim):0.1.8
CompileVersion(org.freedesktop.fwupd):1.9.28
CompileVersion(org.freedesktop.gusb):0.4.9
CpuArchitecture:        x86_64
CpuModel:               Advanced Micro Devices, Inc. AMD Ryzen 7 7800X3D 8-Core Processor
DisplayState:           connected
DistroId:               fedora
DistroName:             Fedora Linux
DistroPrettyName:       Fedora Linux 41 (Workstation Edition)
DistroVariant:          workstation
DistroVersion:          41
FwupdSupported:         True
HostBaseboardManufacturer:Micro-Star International Co., Ltd.
HostBaseboardProduct:   MAG B650 TOMAHAWK WIFI (MS-7D75)
HostBiosMajorRelease:   05
HostBiosMinorRelease:   20
HostBiosVendor:         American Megatrends International, LLC.
HostBiosVersion:        1.C0
HostEnclosureKind:      3
HostFamily:             To be filled by O.E.M.
HostFirmwareMajorRelease:ff
HostFirmwareMinorRelease:ff
HostProduct:            MS-7D75
HostSku:                To be filled by O.E.M.
HostVendor:             Micro-Star International Co., Ltd.
KernelCmdline:          modprobe.blacklist=nouveau rd.driver.blacklist=nouveau
KernelName:             Linux
KernelVersion:          6.12.11-200.fc41.x86_64
LidState:               unknown
PassimDownloadSaving:   5661965
PowerState:             unknown
RuntimeVersion(com.hughsie.libjcat):0.2.2
RuntimeVersion(com.hughsie.libxmlb):0.3.21
RuntimeVersion(org.freedesktop.Passim):0.1.8
RuntimeVersion(org.freedesktop.fwupd):1.9.28
RuntimeVersion(org.freedesktop.fwupd-efi):1.6
RuntimeVersion(org.freedesktop.gusb):0.4.9
RuntimeVersion(org.kernel):6.12.11-200.fc41.x86_64
DeviceId:               823e346631882bd3e8ed3258d84b4cd6f53bbe03
  pre:
      CapsuleApplyMethod:nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader:False
  post:
      LastAttemptStatus:0x0
      LastAttemptVersion:0x1
DeviceId:               362301da643102b9f38477387e2193e57abaa590
  pre:
      EfivarNvramUsed:  84010
      LinuxLockdown:    none
      Pcr0_SHA256:      847ce6ed3b25960ae8a76169d3adb68547a0d97e7163121e3c7fb6d417ed484c
      TpmEventLog:      0x00000008 d681837b3b8f5fd685bfad042ca78e6884e10e1c8f272dbccd2fd10be738a7fa [BSsQp8fZZUGBQCrd6Ur2PA==]
      0x8000000a 7d52fc656d66e80cfcd69f3d9098276c33b1f2d8ed8b2ec6d6aa2bbf17667f65 [D0ZWX01BSU5fV1JBUFBFUgAAAAEAAAAAAAC5AAAAAAA=]
      0x00000004 df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119 [AAAAAA==]
      PCR0: 847ce6ed3b25960ae8a76169d3adb68547a0d97e7163121e3c7fb6d417ed484c
      TpmFamily:        2.0
      EfivarNvramUsed:  84010
      SecureBoot:       Enabled
      UEFIUXCapsule:    Enabled

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?

hughsie, 2025-02-08:
Hmm, there's thousands of reports being logged on the LVFS this morning so I think this might be something specific to your system. Just to check; is the date and time set correctly?

kimdre, 2025-02-08:
Yes, maybe it was just a temporary thing. I don't know how I can retry the upload.

hughsie, 2025-02-08:
> I don't know how I can retry the upload.

fwupdmgr report-history --force

kimdre, 2025-02-08:
Thank you, now it worked.

closed 2025-02-08 by kimdre