fwupd/fwupd #8442: Secure Boot dbx Configuration Update Failure

nattymcfly5, 2025-02-09:
**Describe the bug**
Using either Gnome software or from the terminal I am getting - 'Remote peer disconnected' 'Unknown' - errors with a core dump according to journalctl

**Steps to Reproduce**
attempt to update Secure boot dbx Configuration update 20230501-20241101

**Expected behavior**
Update should take place with no errors 

**fwupd version information**
fwupdmgr --version  
compile   org.freedesktop.fwupd         1.9.28
compile   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.2
runtime   org.freedesktop.fwupd-efi     1.6
compile   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.2
runtime   org.freedesktop.Passim        0.1.8
runtime   org.freedesktop.gusb          0.4.9
runtime   org.kernel                    6.5.6-300.fc39.x86_64
runtime   org.freedesktop.fwupd         1.9.28

```shell
fwupdtool get-report-metadata

BatteryLevel:           101
BatteryThreshold:       10
BootTime:               1739035499
CompileVersion(com.hughsie.libjcat):0.2.2
CompileVersion(com.hughsie.libxmlb):0.3.21
CompileVersion(org.freedesktop.Passim):0.1.8
CompileVersion(org.freedesktop.fwupd):1.9.28
CompileVersion(org.freedesktop.gusb):0.4.9
CpuArchitecture:        x86_64
CpuModel:               Intel 12th Gen Intel Core™ i7-12700K
DisplayState:           connected
DistroId:               fedora
DistroName:             Fedora Linux
DistroPrettyName:       Fedora Linux 41 (Workstation Edition)
DistroVariant:          workstation
DistroVersion:          41
FwupdSupported:         True
HostBaseboardManufacturer:Micro-Star International Co., Ltd.
HostBaseboardProduct:   MEG Z690I UNIFY (MS-7D29)
HostBiosMajorRelease:   05
HostBiosMinorRelease:   18
HostBiosVendor:         American Megatrends International, LLC.
HostBiosVersion:        1.20
HostEnclosureKind:      3
HostFamily:             Default string
HostFirmwareMajorRelease:ff
HostFirmwareMinorRelease:ff
HostProduct:            MS-7D29
HostSku:                Default string
HostVendor:             Micro-Star International Co., Ltd.
KernelCmdline:          rd.driver.blacklist=nouveau modprobe.blacklist=nouveau enforcing=0
KernelName:             Linux
KernelVersion:          6.5.6-300.fc39.x86_64
LidState:               unknown
PassimDownloadSaving:   89086087
PowerState:             unknown
RuntimeVersion(com.hughsie.libjcat):0.2.2
RuntimeVersion(com.hughsie.libxmlb):0.3.21
RuntimeVersion(org.freedesktop.Passim):0.1.8
RuntimeVersion(org.freedesktop.fwupd):1.9.28
RuntimeVersion(org.freedesktop.fwupd-efi):1.6
RuntimeVersion(org.freedesktop.gusb):0.4.9
RuntimeVersion(org.kernel):6.5.6-300.fc39.x86_64
[... 273 lines cut ...]
├── initramfs-6.5.6-300.fc39.x86_64-nouveau.img
├── loader
│   └── entries  [error opening dir]
├── lost+found  [error opening dir]
├── symvers-6.2.14-703.inttf.fc38.x86_64.gz -> /lib/modules/6.2.14-703.inttf.fc38.x86_64/symvers.gz
├── symvers-6.2.15-703.inttf.fc38.x86_64.gz -> /lib/modules/6.2.15-703.inttf.fc38.x86_64/symvers.gz
├── symvers-6.5.6-300.fc39.x86_64.xz -> /lib/modules/6.5.6-300.fc39.x86_64/symvers.xz
├── System.map-6.2.14-703.inttf.fc38.x86_64
├── System.map-6.2.15-703.inttf.fc38.x86_64
├── System.map-6.5.6-300.fc39.x86_64
├── vmlinuz-0-rescue-92b9b390efe849bfbea4708060139a76
├── vmlinuz-6.2.14-703.inttf.fc38.x86_64
├── vmlinuz-6.2.15-703.inttf.fc38.x86_64
└── vmlinuz-6.5.6-300.fc39.x86_64

```

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?
- Are you using an NVMe disk?
- Is secure boot enabled?
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?

hughsie, 2025-02-09:
Eek! This smells like https://github.com/fwupd/fwupd/commit/6c2af55f373d154e5f52e746e9c18fbb13001b5e -- but can you get a backtrace out of the coredump for me please.

mezpahlan, 2025-04-08:
Hi @hughsie I think I am affected by this too. If you let me know how I can try and run some troubleshooting commands. I'm about 2/10 on the command line 😅 .

Ayu-zh, 2025-04-10:
old reddit records suggest updating via command line fixed this. But not for me. When I boot via command line and reboot, a boot menu appears asking me to select one of 3 options which are different versions of fedora workstation

Deepakkrcse, 2025-04-16:
> old reddit records suggest updating via command line fixed this. But not for me. When I boot via command line and reboot, a boot menu appears asking me to select one of 3 options which are different versions of fedora workstation

Same for me, not working with command line also

hughsie, 2025-05-04:
Going to need a coredump or a backtrace to debug further.

mezpahlan, 2025-05-06:
Hi @hughsie I'm happy to do this but could you point me in the right direction for how I should be able to generate a coredump / backtrace? I assume I need to attempt the firmware upgrade and then gather the logs? I haven't done that before so any pointers would be appreciated. 👍

mezpahlan, 2025-05-22:
So I _think_ I managed to get it to work. I couldn't actually get the backtrace or coredump to work. My Linux knowledge isn't that good 😢. Whatever I did must have made a difference but this is what I eventually did.

- Changed Core boot configuration in my bios to factory reset. I had previously messed about with it whilst trying to install Linux on my Windows laptop.
- Removed the Ubuntu deb version of fwupd and installed the latest Snap stable. The deb was on version 1.9 and the Snap version 2.
- Ran the update again.

I'm running Ubuntu 24.04 LTS if that helps anyone.

I think that might have worked 👍.

closed 2025-05-22 by superm1