fwupd/fwupd #8452: Unable to download embargo metadata

MankWang, 2025-02-11:
**Describe the question**
fwupdmgr refresh couldn't download embargo metadata, and print the information below. I have tried three devices and encountered this problem.

```
Updating xxxxx-embargo
Downloading…             [            /                          ]
Failed to download metadata for xxxxx-embargo: Failed to download, server response was 404: 
<!DOCTYPE html>
<!-- Copyright 2015 Richard Hughes <richard@hughsie.com>
     SPDX-License-Identifier: GPL-2.0-or-later -->
<html lang="en">
  <head>
    <title>LVFS: Permission denied</title>
    <meta http-equiv="Content-Type" content="text/html;charset=utf-8"/>
    <link rel="stylesheet" href="https://cdn.fwupd.org/static/theme.css">
    <link rel="stylesheet" href="https://cdn.fwupd.org/static/style.css">
    <meta name="viewport" content="width=device-width, initial-scale=1.0">
    <link rel="shortcut icon" href="https://cdn.fwupd.org/static/favicon.ico"/>

  </head>
  <body>
    <main class="main" id="top">

<!-- navigation -->

<nav class="navbar navbar-dark fs--1 font-weight-semi-bold bg-dark navbar-expand-lg">

  <div class="container">
    <a class="navbar-brand" href="/">
      <div class="d-flex align-items-center text-primary">
        <span class="text-white">LVFS</span>
      </div>
    </a>
    <button class="navbar-toggler collapsed" type="button" data-toggle="collapse" data-target="#navbarStandard" aria-controls="navbarStandard" aria-expanded="false" aria-label="Toggle navigation">
      <span class="navbar-toggler-icon"></span>
    </button>
    <div class="collapse navbar-collapse" id="navbarStandard">
      <ul class="navbar-nav align-items-center">
        <li class="nav-item dropdown">
          <a class="nav-link dropdown-toggle" id="navbarDropdownPages" href="#" role="button" data-toggle="dropdown" aria-haspopup="true" aria-expanded="false">
            Home
          </a>
          <div class="dropdown-menu dropdown-menu-card" aria-labelledby="navbarDropdownPages">
            <div class="bg-white rounded-soft py-2">
              <a class="dropdown-item" href="/lvfs/firmware/new">New Firmware</a>
              <a class="dropdown-item" href="/lvfs/devices/">Supported Devices</a>
              <a class="dropdown-item" href="/lvfs/hsireports/devices">Device Security</a>
              <a class="dropdown-item" href="/lvfs/vendors/">Vendor List</a>
              <a class="dropdown-item" href="/lvfs/vendors/prodcerts">Product Certification</a>
              <a class="dropdown-item" href="/lvfs/docs/privacy">Privacy Report</a>
            </div>
          </div>
        </li>
        <li class="nav-item dropdown">
          <a class="nav-link dropdown-toggle" id="navbarDropdownDocumentation" href="#" role="button" data-toggle="dropdown" aria-haspopup="true" aria-expanded="false">
            Documentation
          </a>
          <div class="dropdown-menu dropdown-menu-card" aria-labelledby="navbarDropdownDocumentation">
            <div class="bg-white rounded-soft py-2">
              <a class="dropdown-item" href="/lvfs/docs/vendors">Hardware Vendors</a>
              <a class="dropdown-item" href="/lvfs/docs/users">End Users</a>
              <a class="dropdown-item" href="/lvfs/docs/developers">Developers</a>
              <a class="dropdown-item" href="/lvfs/docs/introduction">External Documentation</a>
              <div class="dropdown-divider"></div>
              <a class="dropdown-item" href="/lvfs/docs/agreement">Vendor Agreement</a>
              <a class="dropdown-item" href="/lvfs/docs/consulting">Consulting</a>
            </div>
          </div>
        </li>
      </ul>

      <ul class="navbar-nav align-items-center d-none d-lg-block ml-3">
        <li class="nav-item">
          <form class="form-inline search-box" action="/lvfs/search">
            <input class="form-control" type="search" placeholder="Search for firmware..." name="value" aria-label="Search" style="background: none; color: rgba(255, 255, 255, 0.7);"/>
          </form>
        </li>
      </ul>

      <ul class="navbar-nav align-items-center ml-auto">

        <li class="nav-item">
          <a class="btn btn-sm btn-falcon-default" href="/lvfs/login1" role="button">
            Login
          </a>
        </li>

      </ul>
    </div>
  </div>
</nav>

<div class="container mt-3">

<div class="row flex-center min-v
```

**fwupd version information**
compile   org.freedesktop.fwupd         1.9.27
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.0
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.27
runtime   org.kernel                    6.12.0-next-20241120-dirty

```shell
fwupdtool get-report-metadata
```

hughsie, 2025-02-11:
The `fwupdmgr refresh -vv` output would be useful. Can you also try re-downloading the embargo.conf from https://fwupd.org/lvfs/metadata/ and check the `Username` and `Password` is set correctly please.

MankWang, 2025-02-18:
The `Username` and `Password` are both correct, but still print `Permission denied`

 ```
fwupdmgr refresh --force -vv
(pkttyagent:3452): GLib-GIO-DEBUG: 10:41:57.911: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:3450): GLib-GIO-DEBUG: 10:41:57.923: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:3450): dconf-DEBUG: 10:41:57.923: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:3450): dconf-DEBUG: 10:41:57.923: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:3450): dconf-DEBUG: 10:41:57.923: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:3450): dconf-DEBUG: 10:41:57.923: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:3450): dconf-DEBUG: 10:41:57.923: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:3450): GLib-GIO-DEBUG: 10:41:57.923: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(fwupdmgr:3450): GLib-GIO-DEBUG: 10:41:57.925: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:3450): GLib-GIO-DEBUG: 10:41:57.925: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:3450): dconf-DEBUG: 10:41:57.927: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:3450): dconf-DEBUG: 10:41:57.928: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:3450): dconf-DEBUG: 10:41:57.928: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:3450): dconf-DEBUG: 10:41:57.928: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:3450): dconf-DEBUG: 10:41:57.928: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:3450): Fwupd-DEBUG: 10:41:57.930: Emitting ::status-changed() [idle]
WARNING: UEFI capsule updates not available or enabled in firmware setup
See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Updating xxxxxxxx-embargo
Fwupd-INFO: 10:41:57.941: downloading https://xxxxxx%40xxxx.com:XXXXXXXXXXXXX@fwupd.org/downloads/firmware-xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx.xml.zst.asc/auth
(fwupdmgr:3450): Fwupd-DEBUG: 10:41:57.942: Emitting ::status-changed() [downloading]
Downloading…             [                          |            ]Fwupd-INFO: 10:41:58.958: download progress: 100%
(fwupdmgr:3450): Fwupd-DEBUG: 10:41:58.958: Emitting ::status-changed() [idle]
Fwupd-INFO: 10:41:58.958: status-code was 404

Failed to download metadata for xxxxxxxxxxxxx-embargo: Failed to download, server response was 404: 
<!DOCTYPE html>
<!-- Copyright 2015 Richard Hughes <richard@hughsie.com>
     SPDX-License-Identifier: GPL-2.0-or-later -->
<html lang="en">
  <head>
    <title>LVFS: Permission denied</title>
    <meta http-equiv="Content-Type" content="text/html;charset=utf-8"/>
    <link rel="stylesheet" href="https://cdn.fwupd.org/static/theme.css">
    <link rel="stylesheet" href="https://cdn.fwupd.org/static/style.css">
    <meta name="viewport" content="width=device-width, initial-scale=1.0">
    <link rel="shortcut icon" href="https://cdn.fwupd.org/static/favicon.ico"/>

  </head>
  <body>
    <main class="main" id="top">

<!-- navigation -->

<div class="container">

  <nav class="navbar navbar-vertical navbar-expand-lg navbar-light navbar-glass">
    <a class="navbar-brand text-left" href="/lvfs/dashboard">
      <div class="d-flex align-items-center text-primary py-3">
        <div class="d-inline-flex flex-center">
          <img class="mr-2" src="https://cdn.fwupd.org/static/avatar.png" alt="" width="40" />
          <span class="text-sans-serif">LVFS</span>
        </div>
      </div>
    </a>
    <div class="collapse navbar-collapse" id="navbarVerticalCollapse">
      <ul class="navbar-nav flex-column">
        <li class="nav-item">
          <a class="nav-link dropdown-indicator" href="#home" data-toggle="collapse" role="button" aria-expanded="true" aria-controls="home">
            <div class="d-flex align-items-center">
              <span class="nav-link-icon">
                <img class="fa" src="https://cdn.fwupd.org/static/img/home-solid.svg" width="16px" height="16px" />
              </span>
              <span>Home</span>
            </div>
          </a>
[... 29 lines cut ...]
            <li class="nav-item ">
              <a class="nav-link" href="/lvfs/upload/firmware">Upload new</a>
            </li>
            <li class="nav-item ">
              <a class="nav-link" href="/lvfs/firmware/state/private">State :: Private</a>
            </li>
            <li class="nav-item ">
              <a class="nav-link" href="/lvfs/firmware/state/embargo">State :: Embargo</a>
            </li>
            <li class="nav-item ">
              <a class="nav-link" href="/lvfs/firmware/state/testing">State :: Testing</a>
            </li>
            <li class="nav-item ">
              <a class="nav-link" href="/lvfs/firmware/state/stable">State :: Stable</a>
            </li>
            <li class="nav-item ">
              <a class="nav-link" href="/lvfs/firmware/state/deleted">State :: Deleted</a>
            </li>
            <li class="nav-item ">
              <a class="nav-link" href="/lvfs/firmware/events">State :: Events</a>
            </li>
            <li class="nav-item ">
              <a class="nav-link" href="/lvfs/firmware/">State :: All</a>

```

superm1, 2025-02-18:
Are you using the token or your password?  It should be the token.

MankWang, 2025-02-18:
Yes, I am using a newly created token in profile settings.

superm1, 2025-02-18:
I wonder if your username is causing issue.  I notice in your above it's got a `%40`:
```xxxxxx%40xxxx.com```

MankWang, 2025-02-18:
I entered the @ symbol, and this %40 may have been escaped by fwupd.

superm1, 2025-02-18:
OK, maybe @hughsie needs to look at LVFS side to see what could be going wrong.

hughsie, 2025-02-18:
> xml.zst.asc

This isn't supported. Can you redownload the embargo.conf file from https://fwupd.org/lvfs/metadata/ please -- I suspect you're trying to use a keyring of GPG or something.

MankWang, 2025-02-18:
This is strange, I have already downloaded embargo.conf file again, and the content in the file is as follows. It is `xml.gz`, not `xml.zst.asc`

```
[fwupd Remote]
Enabled=true
Title=Embargoed for xxxxxxxxxxxx
RefreshInterval=3600
MetadataURI=https://fwupd.org/downloads/firmware-xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx.xml.gz
ReportURI=https://fwupd.org/lvfs/firmware/report
OrderBefore=lvfs,fwupd
# ↓ MODIFY ↓ -- the email address you use to login to the LVFS
Username=xxxxx@xxxxx.com
# ↓ MODIFY ↓ -- a *USER TOKEN* from https://fwupd.org/lvfs/profile
Password=xxxxxxxxx
```

hughsie, 2025-02-18:
> . It is xml.gz, not xml.zst.asc

Right, we re-write the xml.gz to xml.zst if the fwupd version is new enough. If you use this new embargo file do you get the asc download error?

MankWang, 2025-02-18:
I am using version 1.9.27 because versions after 2.0.0 can not upgrade the PCIE module of Quectel. At present, this issue is still being fixed in #8382 .
I can refresh successfully using the latest fwupd version, but in this case, versions before 2.0.0 can not be used. Is this expected? Or what do I have to do to continue using the old version?