fwupd/fwupd #8454: Update not working: failed to write data to efivarfs: Error writing to file descriptor: Invalid argument

Jens-TGS, 2025-02-12:
**Describe the bug**
Cannot run update. Getting error message "failed to write data to efivarfs: Error writing to file descriptor: Invalid argument" on a Lenovo ThinkCentre M70q Gen 4

**Steps to Reproduce**
fwupdmgr refresh
fwupdmgr update

**Expected behavior**
Update is running without problems like it did before on those machines (we do have 30 of them running Linux Mint).

**fwupd version information**
`root@213pc01:~# fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.27
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.27
runtime   org.kernel                    6.8.0-53-generic`

```shell
root@213pc01:~# fwupdtool get-report-metadata
Waiting…                 [*************************************  ]
BatteryLevel:           101
BatteryThreshold:       25
BootTime:               1739333229
CompileVersion(com.hughsie.libjcat):0.2.0
CompileVersion(com.hughsie.libxmlb):0.3.18
CompileVersion(org.freedesktop.fwupd):1.9.27
CompileVersion(org.freedesktop.gusb):0.4.8
CpuArchitecture:        x86_64
CpuModel:               Intel 13th Gen Intel Core™ i5-13400T
DisplayState:           connected
DistroId:               linuxmint
DistroName:             Linux Mint
DistroPrettyName:       Linux Mint 22.1
DistroVersion:          22.1
FwupdSupported:         True
HostBaseboardManufacturer:LENOVO
HostBaseboardProduct:   332D
HostBiosMajorRelease:   01
HostBiosMinorRelease:   26
HostBiosVendor:         LENOVO
HostBiosVersion:        M4VKT26A
HostEnclosureKind:      23
HostFamily:             ThinkCentre M70q Gen 4
HostFirmwareMajorRelease:01
HostFirmwareMinorRelease:0a
HostProduct:            12E4S0UQ00
HostSku:                LENOVO_MT_12E4_BU_Think_FM_ThinkCentre M70q Gen 4
HostVendor:             LENOVO
KernelName:             Linux
KernelVersion:          6.8.0-53-generic
LidState:               unknown
PowerState:             unknown
RuntimeVersion(com.hughsie.libjcat):0.2.0
RuntimeVersion(com.hughsie.libxmlb):0.3.x
RuntimeVersion(org.freedesktop.fwupd):1.9.27
RuntimeVersion(org.freedesktop.fwupd-efi):1.4
RuntimeVersion(org.freedesktop.gusb):0.4.8
RuntimeVersion(org.kernel):6.8.0-53-generic
DeviceId:               8a3bd64788bb0b0e8e4daa89392d48201ee98625
  pre:
      CapsuleApplyMethod:cod
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader:False
[... 790 lines cut ...]
│       ├── xzio.mod
│       ├── zfs.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zstd.mod
├── initrd.img -> initrd.img-6.8.0-53-generic
├── initrd.img-6.8.0-47-generic
├── initrd.img-6.8.0-53-generic
├── initrd.img.old -> initrd.img-6.8.0-47-generic
├── vmlinuz -> vmlinuz-6.8.0-53-generic
├── vmlinuz-6.8.0-47-generic
├── vmlinuz-6.8.0-53-generic
└── vmlinuz.old -> vmlinuz-6.8.0-47-generic

14 directories, 306 files
```

**Additional questions**

- Operating system and version: Linux Mint 22.1
- Have you tried rebooting? Yes
- Is this a regression? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? Unknown

hughsie, 2025-05-04:
What update are you trying to deploy?

cowboysmall, 2025-07-31:
I see the same on a Dell XPS 15 9560. 

When attempting to "Upgrade KEK CA from 2011 to 2023" using the command `fwupdmgr update`.

I can create a separate issue with all details if it would help.