fwupd/fwupd #8455: FuPluginFlashrom     failed to set bios info: no structures with type 00

cmmrandau, 2025-02-12:
**Describe the bug**
Getting these errors in syslog, don't know for how long. Don't know how it interferes with normal operations. 

**Steps to Reproduce**
It's in syslog, also in fwudptool

**Expected behavior**
-

**fwupd version information**
Please provide the version of the daemon and client.

sudo fwupdtool get-report-metadata
Loading…                 [****                                   ]05:30:48.990 FuPluginFlashrom     failed to set bios info: no structures with type 00
Loading…                 [*************************************  ]
WARNING: UEFI capsule updates not available or enabled in firmware setup
See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Waiting…                 [*************************************  ]
BatteryLevel:           101
BatteryThreshold:       10
BootTime:               1739289483
CompileVersion(com.hughsie.libjcat):0.2.0
CompileVersion(com.hughsie.libxmlb):0.3.18
CompileVersion(org.freedesktop.fwupd):1.9.27
CompileVersion(org.freedesktop.gusb):0.4.8
CpuArchitecture:        aarch64
DisplayState:           unknown
DistroId:               ubuntu
DistroName:             Ubuntu
DistroPrettyName:       Ubuntu 24.04.2 LTS
DistroVersion:          24.04
FwupdSupported:         True
HostBiosVersion:        20231030
HostFamily:             brcm,bcm2712
HostProduct:            Raspberry Pi 5 Model B Rev 1.0
HostVendor:             raspberrypi
KernelCmdline:          vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 multipath=off reboot=w snd_bcm2835.enable_hdmi=1 rootfstype=ext4 fixrtc snd_bcm2835.enable_compat_alsa=0 8250.nr_uarts=1 vc_mem.mem_base=0x3fc00000 coherent_pool=1M dwc_otg.lpm_enable=0 pci=pcie_bus_safe smsc95xx.macaddr=D8:3A:DD:DB:3F:1C bcm2708_fb.fbdepth=16 rootwait bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 bcm2708_fb.fbwidth=640
KernelName:             Linux
KernelVersion:          6.8.0-1018-raspi
LidState:               unknown
PowerState:             unknown
RuntimeVersion(com.hughsie.libjcat):0.2.0
RuntimeVersion(com.hughsie.libxmlb):0.3.x
RuntimeVersion(org.freedesktop.fwupd):1.9.27
RuntimeVersion(org.freedesktop.gusb):0.4.8
RuntimeVersion(org.kernel):6.8.0-1018-raspi

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Default install in Ubuntu 24.04

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

sudo fwupdmgr get-devices --show-all-devices
WARNING: UEFI capsule updates not available or enabled in firmware setup
See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
raspberrypi Raspberry Pi 5 Model B Rev 1.0
│
├─Unknown Device:
│     Device ID:          61541a7f2ebac655c8832fded8fa8311dc84a0b0
│     GUID:               77e49f18-23b1-5030-bd40-7f28207ce489 ← GPIO\ID_pinctrl-rp1
│   
├─Unknown Device:
│     Device ID:          c9c2e330b74bee84a0e711f2fe360354818ad0b1
│     GUID:               6f8c4c13-b857-5e90-a051-69dcd6242bfb ← GPIO\ID_gpio-brcmstb@107d508500
│   
├─Unknown Device:
│     Device ID:          dc3575c0980dd6b9579c2c8b8f2a2f21fba229c8
│     GUID:               e7b8a94a-9fcb-5c52-aae3-11d2703a923f ← GPIO\ID_gpio-brcmstb@107d508520
│   
├─Unknown Device:
│     Device ID:          127c73cd72d15f4d0546cf27d47929f3fcc496a3
│     GUID:               ebd99237-c4f4-5db7-b327-8d70b2ad596c ← GPIO\ID_gpio-brcmstb@107d517c00
│   
├─Unknown Device:
│     Device ID:          94efa0c20e7a0f1a26f5271577b4545cd411c005
│     GUID:               2ed0dd1b-4b39-50d2-9a36-0507a204921a ← GPIO\ID_gpio-brcmstb@107d517c20
│   
├─5 Model B Rev 1.0 (bios):
│     Device ID:          dbee8bd3b1ae0316ad143336155651eedb495a0e
│     Current version:    20231030
│     Vendor:             Raspberry Pi (DMI:raspberrypi)
│     GUIDs:              aff00a55-6aaf-53bf-bf40-4b4d7d82c9f7 ← FLASHROM\VENDOR_raspberrypi&PRODUCT_Raspberry Pi 5 Model B Rev 1.0&REGION_BIOS
│                         98147e00-02ec-50aa-8a7b-d8072fa30cb1 ← FLASHROM\GUID_abe2c65a-0295-5818-a254-2ec5379881c5
│                         909e4cd3-bdf0-581d-955c-f7709d9fb316 ← raspberrypi&brcm,bcm2712&Raspberry Pi 5 Model B Rev 1.0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs shutdown after installation
│                         • Cryptographic hash verification is available
│   
└─PSSD T7:
      Device ID:          11ef15e1570f0c84c63a97d0439dfa7d956ad081
      Summary:            SCSI device
      Current version:    0
      Vendor:             Samsung (SCSI:Samsung)
      Serial Number:      5000000000000001
      GUIDs:              2e7ad008-734e-5cf1-a504-9bfe3d9400e7 ← SCSI\VEN_Samsung&DEV_PSSD-T7
                          2a36e5f9-ddfc-5500-ad47-83d175f8556d ← SCSI\VEN_Samsung&DEV_PSSD-T7&REV_0
      Device Flags:       • Internal device
    

</details>

**Additional questions**

- Operating system and version:
- Ubuntu 24.04.2
- Have you tried rebooting?
- Yes
- Is this a regression?
Don't know

hughsie, 2025-02-12:
>  HostVendor: raspberrypi

Is this using a UEFI capable firmware on a raspberrypi? i.e. does `sudo efibootmgr` work?

cmmrandau, 2025-02-12:
No, that package was not installed but running it shows "sudo efibootmgr
EFI variables are not supported on this system."

cmmrandau, 2025-02-23:
So what is going on here? :)

closed 2025-05-04 by hughsie