fwupd/fwupd #8456: failed to write-firmware: could not set DP_BUF

xrn, 2025-02-12:
**Describe the bug**
I am trying to make an update for X1 12th - Lenovo System Firmware Version 1.39

Error which I am getting:
failed to write-firmware: could not set DP_BUF with EFI/fedora/fw/fwupd-420e7cf1-cedf-4b5f-9ef0-f1896a946d8f.cap: failed to set /sys/firmware/efi/efivars/fwupd-420e7cf1-cedf-4b5f-9ef0-f1896a946d8f-0-0abba7dc-e516-4167-bbf5-4d9d1c739416 as mutable: failed to open: No such file or directory

**Steps to Reproduce**
sudo fwupdmgr get-updates

**Expected behavior**
Installation should work 

**fwupd version information**
fwupd-2.0.5-1.fc42.x86_64

```shell
fwupdtool get-report-metadata
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
dnf

- Operating system and version: Fedora 42
- Have you tried rebooting? Yes
- Is this a regression?

hughsie, 2025-02-12:
> failed to set /sys/firmware/efi/efivars/fwupd-420e7cf1-cedf-4b5f-9ef0-f1896a946d8f-0-0abba7dc-e516-4167-bbf5-4d9d1c739416 as mutable

If efivarfs loaded as read only? e.g. What does `cat /etc/mtab | grep efivarfs` say?

xrn, 2025-02-14:
@hughsie it seems that is is loaded

```
cat /etc/mtab | grep efivarfs 
efivarfs /sys/firmware/efi/efivars efivarfs rw,nosuid,nodev,noexec,relatime 0 0
```

hughsie, 2025-02-14:
Maybe a dumb comment -- but does /sys/firmware/efi/efivars/fwupd-420e7cf1-cedf-4b5f-9ef0-f1896a946d8f-0-0abba7dc-e516-4167-bbf5-4d9d1c739416 exist? The `fwupdtool get-report-metadata` output would be helpful too.

xrn, 2025-02-14:
1. /sys/firmware/efi/efivars/ - exists
2.  /sys/firmware/efi/efivars/fwupd-420e7cf1-cedf-4b5f-9ef0-f1896a946d8f-0-0abba7dc-e516-4167-bbf5-4d9d1c739416 - does not exists
3.  command output

```
BatteryLevel:           85
BatteryThreshold:       25
BootTime:               1739520388
CompileVersion(com.hughsie.libjcat): 0.2.2
CompileVersion(com.hughsie.libxmlb): 0.3.21
CompileVersion(info.libusb): 1.0.27
CompileVersion(org.freedesktop.Passim): 0.1.8
CompileVersion(org.freedesktop.fwupd): 2.0.5
CpuArchitecture:        x86_64
CpuModel:               Intel Core™ Ultra 7 155U
DisplayState:           connected
DistroId:               fedora
DistroName:             Fedora Linux
DistroPrettyName:       Fedora Linux 43 (Workstation Edition Prerelease)
DistroVariant:          workstation
DistroVersion:          43
FwupdSupported:         True
HostBaseboardManufacturer: LENOVO
HostBaseboardProduct:   21KC005RUK
HostBiosMajorRelease:   01
HostBiosMinorRelease:   1d
HostBiosVendor:         LENOVO
HostBiosVersion:        N3YET64W (1.29 )
HostEnclosureKind:      a
HostFamily:             ThinkPad X1 Carbon Gen 12
HostFirmwareMajorRelease: 01
HostFirmwareMinorRelease: 16
HostProduct:            21KC005RUK
HostSku:                LENOVO_MT_21KC_BU_Think_FM_ThinkPad X1 Carbon Gen 12
HostVendor:             LENOVO
KernelName:             Linux
KernelVersion:          6.14.0-0.rc2.22.fc43.x86_64
LidState:               open
PassimDownloadSaving:   2941128
PlatformArchitecture:   x86_64
PowerState:             battery-discharging
RuntimeVersion(com.hughsie.libjcat): 0.2.3
RuntimeVersion(com.hughsie.libxmlb): 0.3.21
RuntimeVersion(org.freedesktop.fwupd): 2.0.5
RuntimeVersion(org.freedesktop.fwupd-efi): 1.6
RuntimeVersion(org.kernel): 6.14.0-0.rc2.22.fc43.x86_64
DeviceId:               d87aeba8ce6cdff706492615b1c1e9f6f48ce3c6
  pre:
      CapsuleApplyMethod: nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader: False
  post:
      LastAttemptStatus: 0x0
      LastAttemptVersion: 0x0
DeviceId:               84c1ee7b500ec547692c1c661ce6f46feb7809d9
  pre:
      CapsuleApplyMethod: nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader: False
  post:
      LastAttemptStatus: 0x0
      LastAttemptVersion: 0x0
DeviceId:               81b48f03e77395bb1d700a59b19f75ae6ceb9e35
  pre:
      CapsuleApplyMethod: nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader: False
[... 354 lines cut ...]
      LinuxLockdown:    none
      Pcr0_SHA256:      48be510bf9e6e17f11ec2e500408f50941b08687700fd653eae3d63fc43a65a9
      TpmEventLog:      0x00000003 0000000000000000000000000000000000000000000000000000000000000000 [U3RhcnR1cExvY2FsaXR5AAM=]
      0x00000008 ae9b43fc86b5947de633a6883ed0f3aaa9183650ee1d7db402e9fa44427af105 [MgAwADAANwAyADMAMgAwADAAMgAwADAAAAA=]
      0x00000001 f05a222163f1c98eb8bb2aeadec4a32f2de93322e9988f227c420b14bdd4af9d [QgBvAG8AdAAgAEcAdQBhAHIAZAAgAE0AZQBhAHMAdQByAGUAZAAgAEkAQgBCAAAAAAA=]
      0x00000001 ac19a701413f612ade615eed0b73dd5d7d7743644bb6e0e0cef5a4bdcb1e5c6a [QgBJAE8AUwAgAE0AZQBhAHMAdQByAGUAZAAgAEIAbwBvAHQAIABHAHUAYQByAGQAIABQAG8AbABpAGMAeQAAAAAA]
      0x80000008 2439a8a0b2cc4d670cb4426e1e125f00694536af9c4ecc2393c9b94e56eaa2eb [AADA/wAAAAAAAAgAAAAAAA==]
      0x80000008 6c993c231371c62f25719efbbbda74194ceb76df411afd03933b7eefacd7f084 [AAC6/wAAAAAAAAYAAAAAAA==]
      0x80000008 8b730987b31ac3db4149524503bdf83f7f96c24420d53bc628f81608b0514ef2 [AACV/wAAAAAAABAAAAAAAA==]
      0x80000008 afb237e00e628bc960ad15cadb061fe7bee4d91136e3d8fa42299a9baf945347 [AACJ/wAAAAAAAAgAAAAAAA==]
      0x80000008 b4289dd5f5004ebfb94d2299ae597a9bc1a55f71df0adc8286fb0502544ab9fb [AAAa/wAAAAAAAGMAAAAAAA==]
      0x80000008 b44b6ab69419fad8933bc2f7ebc2987ff7852a96c9fc6fa146b987cec71831ce [AAB9/wAAAAAAAAwAAAAAAA==]
      0x80000008 169aefa422c748ca1bdf760fff2a9172d8661031e8eba07cea115e642887a066 [AADI/wAAAAAA4AoAAAAAAA==]
      0x00000001 d07b290b36b467f2ad300b907cf91f6a7ce2dcd94c3898ca90ee7446b9c759ee [QUNQSSBEQVRB]
      0x00000001 608072d8953921f15718897cbb0a47623e0c29fa7286d20beb0d733756acb643 [QUNQSSBEQVRB]
      0x00000011 79e5492ed29c1a8d2cb7d512d4eaa2cf5d358ed4c993a1b80ace7e8fdc23919d [SW50ZWxDU3hFRXZlbnQwMQAAAAAEAAAAAAMAADAAAAB63TC03Jz3uNgVchQygNZWZnI4v0oReCb/jM2e/DgiWhDCx4HaqUpoYcRhooTffcwCAwAAFgAAABIAAAAFADIIAQAAAAEAAAABAAAAAQACLQAAFgAAAAEAAAAAAAEAAQAAAAEAAAAAAAAAAQABAgAAMAAAAFhH/p4Z/rlNC1LMPF7Sa5GBbGUVhOAkWZjssVKLs0g3H6wj9jzz+j96Ce6WTwc3agICAAAWAAAACAcPAAAAIAQAAAAAAAAAAAAAAAABAAEOAAAwAAAAgSvMehRr8dr2x08a3SiKJV4ochcsVeks66vvO6kgSz9rJjOsPT+ZL5Sdk04kSRptAg4AABYAAAAIBw8AAAALBAAAAAAAAAAAAAAAAAEAAwQAAAEAAAABAwAAAAQAAAA3AAAAAcAAADAAAACahZBEsmtbfM7HN4Rq5qyE1XmmBxizIZ048qAX9DrlRm7C0OBaG3eRusGjLD2HvHUCwAAAFgAAAOgDAAAFADMHAQAAAAEAAAAAAAAAAQABwQAAMAAAAIoYyEMQIGwchIVIUBmaKjvvyRBkjKKcqhX3OQx4oCYvaREey89Gk35QzdC0uxr96gLBAAAWAAAAAoFxgAAAUEABAAAAAQAAAAAAAAABAAHCAAAwAAAAFScz3S+fUwStCrN0DvDAiUtWn0LgLRn2KlmksX3glwlTAn6JyUg+LlXGMdmQloyyAsIAABYAAAACAXGAAABQQAEAAAABAAAAAAAAAAEAA2QAAAQAAAABEAAAASEAADAAAAA1hx4FtDLSrXVRunQgcopKQqlhVYq7s7MeRR6ZwpFLjzZCMWJesN8/Y7o5UBlvRmECIQAAFgAAAAAACQABAJIGAAAAAAAAAAAAAAAAAQABKQAAMAAAAKn+2wsAM52TzF6T1B/tnMd9cRtHnGX6DO90MRDISuASJNNk2+kNuQcps/RfTTG8DgIpAAAWAAAABQAGAAAA/h0AAAAAAAAAAAAAAAABAAEGAAAwAAAAWyNkm55V3VBcP0cA/r+enRQVhdWbIWJJAF8RT+wQlYDHqrY6tVkUAV09KausmTvtAgYAABYAAAAwABoAAAAAAAAAAAAAAAAAAAA [... cut ...] QAKADLBQAAAAAAAAAAAAAAAAAAAQADAwAAAQAAAAA=]
      0x00000011 2046f8405292c0a34c3e2ca758b338643a97a51a1713991e6acbad36ef7266c1 [SW50ZWxDU3hFSW5mb0V2ZW50AAABAAAAhoBwfgEAAAA=]
      0x00000004 df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119 [AAAAAA==]
      PCR0: 48be510bf9e6e17f11ec2e500408f50941b08687700fd653eae3d63fc43a65a9
      TpmFamily:        2.0
      BootMgrDesc:      legacy
      EfivarsNvramUsed: 70062
      SecureBoot:       Disabled
      UEFIUXCapsule:    Enabled
```

xrn, 2025-02-17:
@hughsie any idea what can be wrong?

Sturmflut, 2025-03-02:
@hughsie can this bug report considered to be closed by #8500 as well?

superm1, 2025-03-02:
Yeah should be same problem, thanks for pointing it out.

closed 2025-03-02 by superm1

hughsie, 2025-03-02:
There's a COPR here with a Fedora build: https://copr.fedorainfracloud.org/coprs/rhughes/fwupd/