fwupd/fwupd #8461: Update the wiki to provide "resolutions" for each test

RainOfPain125, 2025-02-13:
**Is your feature request related to a problem? Please describe.**
I'm very frustrated trying to search on Google and the fwupdmgr documentation for information about how to fix "SPI write protection: disabled". I can't find any information on the topic regarding if a user can address and fix this issue on their own, or if this is a purely hardware limitation that the user cannot change.

**Describe the solution you'd like**
It appears that only 3 of the checks have a resolution written for them - "DMA protection", "CSME Version", and "UEFI SecureBoot". I can probably, sadly, assume that all of the other checks without a "resolution" aren't able to be fixed by the user. But it would be nice if it was **_explicitly_** stated as such - with something like

Resolution: Hardware-related, unable to be addressed by user.

idk sounds really simple and would save time for people like me in the future.

hughsie, 2025-02-13:
Is https://fwupd.github.io/libfwupdplugin/hsi.html better than the wiki in that regard?

RainOfPain125, 2025-02-13:
> Is https://fwupd.github.io/libfwupdplugin/hsi.html better than the wiki in that regard?

Sorry, that is what I was looking at and meant to refer to.

hughsie, 2025-02-13:
I think each one needs a resolution in https://github.com/fwupd/fwupd/tree/main/docs/hsi-tests.d -- i.e. a `resolution` key. Would you like to have a go and fill in the messing keys? We can check each one and suggest better ideas if that helps.

closed 2025-05-03 by hughsie