fwupd/fwupd #8464: Minor logging regression: "starting fwupd <version>…" not printed in fwupd 2.0

nicholasbishop, 2025-02-13:
**Describe the bug**
In fwupd 1.9, the start of the fwupd log output looks like this:
```
FuDebug              verbose to info (on console 0)
FuEngine             starting fwupd 1.9.28...
```

In fwupd 2.0, the code to print that log line is [still present](https://github.com/fwupd/fwupd/blob/2dc288e4ca81b0032461a886bcee929f5b1a3758/src/fu-engine.c#L8580), but it doesn't show up.

My guess is that the log handler is being installed a little later than in fwupd 1.9. Per https://docs.gtk.org/glib/func.log_default_handler.html, the default handler doesn't print info-level messages by default. If I set `G_MESSAGES_DEBUG=all` in the env, the message does appear:
```
(process:47962): GLib-GIO-DEBUG: 18:46:43.501: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ?gio-vfs?
FuEngine-INFO: 18:46:43.501: starting fwupd 2.0.6?
FuDebug              verbose to info (on console 0)
```
It's a very minor thing, but can be helpful to have this information when examining device logs.

**Steps to Reproduce**
Run the fwupd daemon, observe log output.

**Expected behavior**
The log should contain `FuEngine starting fwupd 2.0.x...` near the start.

**fwupd version information**
Tested fwupd 1.9.28 (log present), fwupd 2.0.1 (log missing), and latest master (commit 2dc288e4ca81b0032461a886bcee929f5b1a3758, log missing)

```shell
fwupdtool get-report-metadata
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

ChromeOS package

**Additional questions**

- Operating system and version: ChromeOS M135
- Have you tried rebooting? no
- Is this a regression? Yes

hughsie, 2025-02-14:
>  In fwupd 1.9, the start of the fwupd log output looks like this:

Even when `-v` isn't specified? I think the general idea was that we'd log the very minimum without `-v`. @superm1 opinions?

superm1, 2025-02-14:
> my guess is that the log handler is being installed a little later than in fwupd 1.9

Yeah this theory makes sense.  The actual line hasn't changed since fwupd 1.9.1 (4bcbcade2d).  

>  I think the general idea was that we'd log the very minimum without -v. @superm1 opinions?

I don't really feel strongly, but I would ask if you can try justify the value it brings to log it to convince me otherwise.  It's really simple to run `fwupdmgr --version`, or look at your package manager to determine what version is running.

nicholasbishop, 2025-02-14:
> Even when -v isn't specified? I think the general idea was that we'd log the very minimum without -v
You are right, without `-v` it doesn't show up even in 1.9. I didn't notice this since ChromeOS always runs the service with `-v`.

> I don't really feel strongly, but I would ask if you can try justify the value it brings to log it to convince me otherwise.

Sure thing. On ChromeOS, all output from fwupd is redirected to `/var/log/fwupd.log`. This log file is persistent, so if you reboot the device, the old logs are still there and the new run of fwupd is appended to the end. If an OS update occurs, the log file may contain logs from different versions of fwupd.

A bit more context: I noticed this missing log when looking at a user feedback report, which is essentially a big bundle of system logs. I was in the habit of searching the log for "starting fwupd" to find the start of logging for the most recent invocation of fwupd, and then I was very confused because that search only turned up fwupd 1.9 when I was expecting 2.0. The log bundle also includes `fwupdmgr --version`, which showed the correct version. Once I realized that log line was still present in the code but wasn't showing up in the actual log, I was worried we might be missing other logs, so I spent a while investigating it until I figured out the `G_MESSAGES_DEBUG=all` trick.

So in the end... it's definitely a very minor bug, and would just be nice to have the version back in the log. If that isn't feasible, I think it would be good to remove the ineffective "starting fwupd" log line in case anyone else has the same confusion as me about whether logs are getting dropped.

closed 2025-04-03 by superm1