fwupd/fwupd #8466: Flaky fu-self-test unit test in 2.0.5

rishabh-chromium, 2025-02-14:
**Describe the bug**
Failing about 50% of the times on a particular board on ChromeOS with this error:
```
ok 9 /fwupd/device
ok 10 /fwupd/device{filter}
ok 11 /fwupd/security-attr
ok 12 /fwupd/bios-attrs
ok 13 /fwupd/client_api
ok 14 /fwupd/client_api{undefined_setter}
ok 15 /fwupd/client_api{undefined_getter}
# child process (/fwupd/client_api{ro_props} [33]) exit status: 0 (success)
# child process (/fwupd/client_api{ro_props} [33]) stdout: "(/tmp/portage/sys-apps/fwupd-2.0.5-r51/work/fwupd-2.0.5-build/libfwupd/fwupd-self-test:33): Fwupd-DEBUG: 13:20:02.370: D-Bus system bus unavailable, skipping tests.\n(/tmp/portage/sys-apps/fwupd-2.0.5-r51/work/fwupd-2.0.5-build/libfwupd/fwupd-self-test:33): GLib-GIO-DEBUG: 13:20:02.429: _g_io_module_get_default: Found default implementation dummy (GDummyProxyResolver) for ?gio-proxy-resolver?\n"
# child process (/fwupd/client_api{ro_props} [33]) stderr: "\n(/tmp/portage/sys-apps/fwupd-2.0.5-r51/work/fwupd-2.0.5-build/libfwupd/fwupd-self-test:33): GLib-GObject-CRITICAL **: 13:20:02.435: g_object_set_is_valid_property: property 'daemon-version' of object class 'FwupdClient' is not writable\n"
not ok /fwupd/client_api{ro_props} - Fwupd:ERROR:../fwupd-2.0.5/libfwupd/fwupd-self-test.c:537:fwupd_client_api_ro_props: child process (/fwupd/client_api{ro_props} [33]) did not fail as expected
Bail out!
[1;36mstderr:[0m
```

Even when it does pass, the test takes more than 10 seconds (usually on 2s).

**Steps to Reproduce**
Run unit test multiple times

**fwupd version information**
2.0.5

**Additional questions**

- Operating system and version: ChromeOS
- Have you tried rebooting? Yes
- Is this a regression? Yes

rishabh-chromium, 2025-02-18:
@superm1 Any idea what this might be? Do you need any other logs?

superm1, 2025-02-18:
So these tests are specifically for exercising what should be undefined behavior in the daemon (like assertions).
They use https://docs.gtk.org/glib/func.test_undefined.html

It would be really ideal if we had a [CI reproducer](https://github.com/fwupd/fwupd/issues/8123) so we could easily instrument extra debugging on things that can be introducing it.  I don't really understand how this test can be flaky.  It _feels_ like a Glib bug, but I we're not tripping it in any other of our CI tests.

For now a workaround you can use is `-m no-undefined` when launching your tests.

superm1, 2025-02-18:
Maybe a hint that will jog @hughsie's mind.  I notice this message in stdout above:
> # child process (/fwupd/client_api{ro_props} [33]) stdout: "(/tmp/portage/sys-apps/fwupd-2.0.5-r51/work/fwupd-2.0.5-build/libfwupd/fwupd-self-test:33): Fwupd-DEBUG: 13:20:02.370: D-Bus system bus [...]

That _shouldn't_ be coming from this test.  That should be coming from attempts to run tests _after_ this one (`fwupd_client_remotes_func` and `fwupd_client_devices_func`).

superm1, 2025-02-18:
Is this maybe tied to clang vs gcc?

rishabh-chromium, 2025-02-18:
The PR didn't help a lot. Test still taking >20 seconds. We might be using a clang wrapper for our tests, I am not 100% sure. Trying to figure it out

superm1, 2025-02-18:
> The PR didn't help a lot. Test still taking >20 seconds. We might be using a clang wrapper for our tests, I am not 100% sure. Trying to figure it out

Besides taking a long time, are they at least passing now?

rishabh-chromium, 2025-02-19:
So after investigating further, the test is still failing even after this commit applied. It only fails if there is a resource crunch. For example there is a 100% failure rate if I run this test while simultaneously running a stress test on my machine (e.g. using https://mprep.info).

superm1, 2025-02-19:
Can you get new logs for any new inspiration?

rishabh-chromium, 2025-02-19:
Here is the entire error message. This probably has more context.

[error.txt](https://github.com/user-attachments/files/18876085/error.txt)

hughsie, 2025-05-03:
Fixed in https://github.com/fwupd/fwupd/pull/8497 I hope.

closed 2025-05-03 by hughsie