fwupd/fwupd #8469: failed to add device, maybe cause for slower boot

Artim96, 2025-02-17:
**Describe the bug**
By coincidence I found out that fwupd.service is slowing down bootup by quite a bit. These are the two top-most entries of `systemd-analyze blame`:

```
27.384s fwupd.service
 3.212s NetworkManager-wait-online.service
```

So I looked into the journal and found this entry:

```
Feb 17 18:24:07 fwupd[4824]: 17:24:07.426 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:08.1/0000:c1:00.3/usb1/1-2/1-2.2: failed to setup: failed to get firmware version info: failed to get GET_REPORT: failed to GetReport: USB error: Operation timed out [-7]
Feb 17 18:24:07 fwupd[4824]: 17:24:07.559 FuMain               Daemon ready for requests (locale en_US.UTF-8)
```

It's quite difficult finding out what that device is. `udevadm info -a -p /sys/devices/pci0000:00/0000:00:08.1/0000:c1:00.3/usb1/1-2/1-2.2` gives me a few clues:

```
ATTR{configuration}=="Speaker Volume"
ATTR{manufacturer}=="Framework"
```
together with the fact that it's a USB device and there are only two "USB devices" that fwupdmgr can't find firmware updates for ("Hub" and "DisplayPort Expansion Card"), my guess is this might be about the "DisplayPort Expansion Card".

The complete output of `fwupdtool get-devices -vv` can be found here: [fwupdtool.txt](https://github.com/user-attachments/files/18829982/fwupdtool.txt)

**Steps to Reproduce**
Bootup

**Expected behavior**
Bootup with no longer holdups by fwupd.service.

**fwupd version information**
Please provide the version of the daemon and client.

```
fwupdmgr --version
compile   info.libusb                   1.0.27
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.0
compile   org.freedesktop.fwupd         2.0.5
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.0
runtime   org.kernel                    6.12.13-amd64
runtime   org.freedesktop.fwupd         2.0.5
```

```shell
fwupdtool get-report-metadata
```

[get-report-metadata.txt](https://github.com/user-attachments/files/18830395/get-report-metadata.txt)

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

apt

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

[show-all-devices.txt](https://github.com/user-attachments/files/18830418/show-all-devices.txt)

</details>

**Additional questions**

- Operating system and version: Debian Testing
- Have you tried rebooting? Yes
- Is this a regression? Unknown

superm1, 2025-02-17:
If you boot without the blamed cards inserted are things OK?

Artim96, 2025-02-17:
When booting with the DP card removed, both issues are being resolved.

superm1, 2025-02-17:
@JohnAZoidberg thoughts?

JohnAZoidberg, 2025-02-18:
I tried with and old version of fwupd:

```
> time sudo fwupdtool get-devices --plugins ccgx
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
Framework Laptop (13th Gen Intel Core)
│
└─DisplayPort Expansion Card:
      Device ID:          c56306082369348c7a362d311e535dd52620742c
      Current version:    3.0.17.101
      Bootloader Version: 3.0.2.900
      Vendor:             Framework (USB:0x32AC)
      Serial Number:      11AD1D0030123F183A0C0B00
      GUIDs:              7e137108-1fe0-5820-b962-6603fa7b2070 ← USB\VID_32AC&PID_0003
                          a671c02b-adb7-5f2d-b7b6-818cb56c9deb ← USB\VID_32AC&PID_0003&REV_0000
                          f30e2ece-b80f-525d-b651-b95c612cec05 ← USB\VID_32AC&PID_0003&MODE_FW2
      Device Flags:       • Updatable
                          • Device stages updates
                          • Unsigned Payload
    

________________________________________________________
Executed in   14.66 secs      fish           external
   usr time    2.56 millis  633.00 micros    1.93 millis
   sys time   14.56 millis   89.00 micros   14.47 millis

> sudo fwupdtool --version
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
compile   org.freedesktop.fwupd         1.9.2
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.0
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.9.3-76060903-generic
runtime   org.freedesktop.fwupd         1.9.2
```

And with the latest main (3df77b396):

```
> time fwupdtool get-devices --plugins ccgx
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
Framework Laptop (13th Gen Intel Core)
│
└─DisplayPort Expansion Card:
      Device ID:          28a4aa67dfde939ffdb66e6975c4b676643e27dc
      Current version:    3.0.17.101
      Bootloader Version: 3.0.2.900
      Vendor:             Framework (USB:0x32AC)
      Serial Number:      11AD1D0030123F183A0C0B00
      GUIDs:              7e137108-1fe0-5820-b962-6603fa7b2070 ← USB\VID_32AC&PID_0003
                          f30e2ece-b80f-525d-b651-b95c612cec05 ← USB\VID_32AC&PID_0003&MODE_FW2
      Device Flags:       • Updatable
                          • Device stages updates
                          • Unsigned Payload
                          • Can tag for emulation
    

________________________________________________________
Executed in  267.51 millis    fish           external
   usr time    9.29 millis    0.00 micros    9.29 millis
   sys time   26.56 millis  703.00 micros   25.85 millis

> fwupdtool --version
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
compile   info.libusb                   1.0.25
compile   org.freedesktop.fwupd         2.0.7
compile   com.hughsie.libxmlb           0.3.20
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.0
runtime   com.hughsie.libxmlb           0.3.20
compile   org.freedesktop.fwupd.source  2.0.6-2-g3df77b396
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.9.3-76060903-generic
runtime   org.freedesktop.fwupd         2.0.7
```

So it seems this issue is fixed already. Previously it would take 14 seconds on the first try and now it only takes 0.3 seconds.
But even with the old fwupd version, subsequent runs succeed in roughly 0.3 seconds.
Only after unplugging and replugging it's back to 14s.

JohnAZoidberg, 2025-02-18:
I also tried the HDMI card - exact same behavior.

Artim96, 2025-02-18:
Is there a way to say for sure which module did cause this? Booting up my PC today also didn't show any delays from fwupd, even though the DP module was attached. I'm actually seeing another issue I was not yet able to further figure out: https://community.frame.work/t/audio-expansion-card-connection-sometimes-unreliable/62951

In short, sometimes the audio jack module isn't showing up as an audio device, even when headphones are connected. Due to the fact that shutting down/rebooting the PC takes a lot longer when this happens, my guess is it's an issue with the Kernel 6.12 series. This is supported by the fact that in the time I have been using 6.13.2 (compiled from source, but based on the config Debian used for 6.13 rc7) this didn't show up, while on 6.12.x it shows up quite often. Also, the only other time(s) I've had a much slower reboot times was when there was some (unknown to me) issue in both 6.11 and 6.12 until both (or the same?) issues where fixed.

Would it be possible this could be triggered by the very same issue, and fwupd.service taking that long isn't the cause but a result? I'll have to test for a couple of days, as the issue doesn't happen that reliably, but if the audio jack module could have been the issue and not the DP module, this could be connected.

JohnAZoidberg, 2025-02-19:
> Is there a way to say for sure which module did cause this?

As in my example, you can make fwuptool just use the ccgx plugin.
Then it'll only connect to devices supported by this plugin, which is only the HDMI or DP Expansion cards.

Could you please try the latest fwupd main commit to see if it's faster?

Artim96, 2025-02-19:
Do I get the latest main commit by just cloning the repo and compiling it, without any `git checkout`? If so, there's no difference beyond the usual variation between runs.

The version from Debian testing repo:
```
fwupdmgr --version 
compile   info.libusb                   1.0.27
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.0
compile   org.freedesktop.fwupd         2.0.5
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.0
runtime   org.kernel                    6.12.13-amd64
runtime   org.freedesktop.fwupd         2.0.5
```

```
time sudo fwupdtool get-devices --plugins ccgx
real	0m0,400s
user	0m0,011s
sys	0m0,015s
```

And the one I just compiled:
```
fwupdmgr --version 
compile   info.libusb                   1.0.27
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.0
compile   org.freedesktop.fwupd         2.0.5
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.0
runtime   org.kernel                    6.12.13-amd64
runtime   org.freedesktop.fwupd         2.0.5
```

```
time sudo fwupdtool get-devices --plugins ccgx
real	0m0,386s
user	0m0,008s
sys	0m0,012s
```

JohnAZoidberg, 2025-02-19:
Please try running it right after you unplug and replug the card.
I only see the slow run the very first time.

> Do I get the latest main commit by just cloning the repo and compiling it, without any git checkout? If so, there's no difference beyond the usual variation between runs.

No, you have to do:

```
./contrib/setup
source venv/bin/activate
build-fwupd
fwupdtool get-devices --plugins ccgx
```

See https://github.com/fwupd/fwupd/blob/main/docs/building.md
Your version does not match the one on main branch.

Artim96, 2025-02-19:
Right, the issue was that when executing with sudo it will use the systems version of the program. Here the updated results:

```
fwupdtool --version
Loading…                 [************************************** ]
WARNING: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
WARNING: This package has not been validated, it may not work properly.
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.7
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.fwupd.source  2.0.6-4-gfb7120116
runtime   com.hughsie.libjcat           0.2.0
runtime   org.kernel                    6.12.13-amd64
runtime   org.freedesktop.fwupd         2.0.7
```

```
time fwupdtool get-devices --plugins ccgx
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
Framework Laptop 16 (AMD Ryzen 7040 Series)
│
└─DisplayPort Expansion Card:
      Device ID:          d787669ee4a103fe0b361fe31c10ea037c72f27c
      Current version:    3.0.17.101
      Bootloader Version: 3.0.2.900
      Vendor:             Framework (USB:0x32AC)
      Serial Number:      11AD1D00A49C4018360F0B00
      GUIDs:              7e137108-1fe0-5820-b962-6603fa7b2070 ← USB\VID_32AC&PID_0003
                          f30e2ece-b80f-525d-b651-b95c612cec05 ← USB\VID_32AC&PID_0003&MODE_FW2
      Device Flags:       • Updatable
                          • Device stages updates
                          • Unsigned Payload
                          • Can tag for emulation
    

real	0m0,378s
user	0m0,025s
sys	0m0,020s
```

Still, virtually no difference in real time, though user and sys are about doubled.

JohnAZoidberg, 2025-02-20:
> Still, virtually no difference in real time, though user and sys are about doubled.

This is what I'm seeing.

Now if you try with your old system version of fwupd right after plugging in the card, I assume you'll see it taking a long time.

Artim96, 2025-02-20:
True, it takes increasingly longer. With DP module detached:
```
real	0m0,194s
user	0m0,020s
sys	0m0,003s
```

just after plugging it in:
```
real	0m0,258s
user	0m0,015s
sys	0m0,008s
```

11 sec later:
```
real	0m0,369s
user	0m0,005s
sys	0m0,019s
```

another 3 sec later:
```
real	0m0,386s
user	0m0,010s
sys	0m0,013s
```

Comparing with the newer version:
unplugged:
```
real	0m0,199s
user	0m0,025s
sys	0m0,019s
```

just after plugging in:
```
real	0m0,425s
user	0m0,014s
sys	0m0,031s
```

7 sec later:
```
real	0m0,368s
user	0m0,020s
sys	0m0,025s
```

another 5 sec later:
```
real	0m0,380s
user	0m0,022s
sys	0m0,025s
```

JohnAZoidberg, 2025-02-20:
It's all below half a second. So this is probably not the original issue you saw.

Artim96, 2025-02-20:
I know, hence my theory in https://github.com/fwupd/fwupd/issues/8469#issuecomment-2665936860