fwupd/fwupd #8480: fwupd get-report is not showing the correct history report

CragW, 2025-02-20:
**Describe the bug**
The report in pending.db is okay, but tool is showing a "history" that has never been happened.

History in db:
```
$ sqlite3 ./pending.db "select version_old, version_new, version_format from history;"
01.00.44.01|01.00.44.01|5
01.23|01.23|3
01.59|01.59|3
05.07.07|05.07.07|4
41.81|41.81|3
01.01.00.10|01.01.00.10|5
```

<details>

```
$ fwupdmgr get-history
[sudo] password for crabo: 
WARNING: This package has not been validated, it may not work properly.
Dell Inc. Latitude 7350
│
├─Package level of Dell dock:
│ │   Device ID:          324de9750a6ee3498c614454eaf6e73b8af68cde
│ │   Previous version:   01.00.44.01
│ │   Update State:       Success
│ │   Last modified:      2025-02-20
│ │   GUID:               7fb217aa-2312-5c99-a853-56d865c76815
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Unsigned Payload
│ │ 
│ └─HD22 VMM6221 MST Controller Update:
│       New version:      06.05.06
│       Remote ID:        lvfs
│       Release ID:       102056
│       Summary:          Firmware update for HD22 VMM6221 MST
│       License:          Proprietary
│       Size:             1.0 MB
│       Created:          2024-11-12
│       Urgency:          High
│       Vendor:           Dell
│       Release Flags:    • Trusted payload
│                         • Trusted metadata
│       Description:      
│       -Improved the display compatibility of WD19, WD22, and HD22 series dock between systems and monitors.
│       Checksum:         cf16c3d1c0adda35a41763b4b112bba915a262f396bfe025481fb896dca5be8c
│     
├─RTS5413 in Dell dock:
│ │   Device ID:          6dd818beef03a02cf40f288d056553930866ebf1
│ │   Previous version:   01.23
│ │   Install Duration:   13 seconds
│ │   Update State:       Success
│ │   Last modified:      2025-02-20
│ │   GUID:               b27d25f1-019d-5718-b41a-02ddaefe5577
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │                       • Can tag for emulation
│ │ 
│ └─HD22 VMM6221 MST Controller Update:
│       New version:      06.05.06
│       Remote ID:        lvfs
│       Release ID:       102056
│       Summary:          Firmware update for HD22 VMM6221 MST
│       License:          Proprietary
[... 733 lines cut ...]
│     GUIDs:              74fb1b87-53e6-5115-bd02-a9fefd06ed9a ← THUNDERBOLT\VEN_8087&DEV_0D9C
│                         dac6f95d-7c5b-52b5-8c79-a0436a5ef784 ← TBT-80870d9c-retimer0-0:1.1
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Device stages updates
│                         • Unsigned Payload
│                         • Can tag for emulation
│   
└─YJ3JK:
      Device ID:          aec1a869eb0df71b7cea6b3ac71d39b830faf164
      Vendor:             PNP:AUO
      GUID:               340848af-fb81-553f-a1d4-9b81b5539548 ← DRM\VEN_AUO&DEV_CFA4
      Device Flags:       • Internal device
                          • Can tag for emulation
    

```

</details>

**Additional questions**

- Operating system and version: Ubuntu 24.04
- Have you tried rebooting? Yes
- Is this a regression? Yes

hughsie, 2025-02-20:
Can you attach the pending.db please; I want to try here. You'll have to rename it to something like pending.txt to attach it.

CragW, 2025-02-20:
> Can you attach the pending.db please; I want to try here. You'll have to rename it to something like pending.txt to attach it.

Found out that this issue is related to `polkit` not really sourced the added rules under `/usr/share/polkit-1/actions/org.freedesktop.fwupd.policy` directory.  After added a symlink `/etc/polkit-1/actions -> /usr/share/polkit-1/actions` the issue has gone away.

There is no warning to the frontend `fwupdmgr`, took some time to figure it out. Thanks  @hughsie for taking an eye on this issue too.  Attached `pending.db`:  [pending.txt](https://github.com/user-attachments/files/18883562/pending.txt)

superm1, 2025-02-20:
Is something wrong with helpers or packaging then?

CragW, 2025-02-20:
> Is something wrong with helpers or packaging then?

No idea why polkit is involved. Below are  the reproducible steps:

```
// Delete the mentioned symlink
$ sudo rm /etc/polkit-1/actions
$ sudo reboot

// Check the registered polkit rules, apparently some are missing
$ pkactions | grep fwupd
org.freedesktop.fwupd.device-activate
org.freedesktop.fwupd.device-unlock
org.freedesktop.fwupd.downgrade-hotplug
org.freedesktop.fwupd.downgrade-hotplug-trusted
org.freedesktop.fwupd.downgrade-internal
org.freedesktop.fwupd.downgrade-internal-trusted
org.freedesktop.fwupd.get-bios-settings
org.freedesktop.fwupd.modify-config
org.freedesktop.fwupd.modify-remote
org.freedesktop.fwupd.self-sign
org.freedesktop.fwupd.set-approved-firmware
org.freedesktop.fwupd.set-bios-settings
org.freedesktop.fwupd.update-hotplug
org.freedesktop.fwupd.update-hotplug-trusted
org.freedesktop.fwupd.update-internal
org.freedesktop.fwupd.update-internal-trusted
org.freedesktop.fwupd.verify-update

// Start the daemon
(venv) $ fwupd
(venv) $ fwupdmgr refresh --force

// Get the cab file
(venv) $ fwupdmgr download 

// Install the cab
(venv) $ fwupdmgr install foo.cab --allow-reinstall

// Reproduce the issue
(venv) $ fwupdmgr get-history
```

superm1, 2025-02-20:
In your reproduce steps, can you share the pending.db for both a good and a failure case?  We should be able to reproduce then without actually installing a cab and dig into this.

CragW, 2025-02-20:
> can you share the pending.db for both a good and a failure case? 

Sure, the db has been attached in this comment tho renamed to .txt.
https://github.com/fwupd/fwupd/issues/8480#issuecomment-2670838192

> We should be able to reproduce then without actually installing a cab and dig into this.

Yes, the db in venv/dist path is unchanged and I can repro this issue without needing to install the cab. Just removed the pkrules outside as the steps.

superm1, 2025-02-23:
OK so from Ubuntu 24.04, here is what I'm seeing.

```
❯ fwupdmgr refresh --force
Missing or outdated PolicyKit policy in /usr/share/polkit-1/actions/org.freedesktop.fwupd.policy
Copy into filesystem? [y/N]
y
WARNING: This package has not been validated, it may not work properly.
Updating lvfs
Downloading…             [************************************** ]
Successfully downloaded new metadata: Updates have been published for 4 of 7 local devices
USB4 controller in Dell dock is pending activation; use /home/supermario/src/fwupd/venv/bin/../dist/bin/fwupdmgr activate to complete the update.
WD22TB4 is pending activation; use /home/supermario/src/fwupd/venv/bin/../dist/bin/fwupdmgr activate to complete the update.
```

then here is the `get-history`.  So the issue is that it's showing the MST controller update for the wrong device?  
```
└─WD22TB4:
  │   Device ID:          4f1ef250bc64d22736e67c92e011e1d69a3b718f
  │   Previous version:   01.01.00.10
  │   Install Duration:   2 seconds
  │   Update State:       Success
  │   Update Error:       Device has been removed, Pending activation
  │   Last modified:      2025-02-20
  │   GUID:               cd357cf1-40b2-5d87-b8df-bb2dd82774aa
  │   Device Flags:       • Supported on remote server
  │                       • Device update needs activation
  │                       • Device stages updates
  │                       • Device can recover flash failures
  │                       • Device is usable for the duration of the update
  │                       • Updatable
  │                       • Signed Payload
  │ 
  └─HD22 VMM6221 MST Controller Update:
        New version:      06.05.06
        Remote ID:        lvfs
        Release ID:       102056
        Summary:          Firmware update for HD22 VMM6221 MST
        License:          Proprietary
        Size:             1.0 MB
        Created:          2024-11-12
        Urgency:          High
        Vendor:           Dell
        Release Flags:    • Trusted payload
                          • Trusted metadata
        Description:      
        -Improved the display compatibility of WD19, WD22, and HD22 series dock between systems and monitors.
        Checksum:         cf16c3d1c0adda35a41763b4b112bba915a262f396bfe025481fb896dca5be8c
```

CragW, 2025-02-23:
> then here is the `get-history`. So the issue is that it's showing the MST controller update for the wrong device?

Yes, that's the issue. Weirdly the update history is connecting to a wrong device.

superm1, 2025-02-23:
> 
> > then here is the `get-history`. So the issue is that it's showing the MST controller update for the wrong device?
> 
> Yes, that's the issue. Weirdly the update history is connecting to a wrong device.
> 

I wonder if it's because of a composite cab?
I can't reproduce with fwupdtool as well. This shouldn't be a PK issue.